EHfive/pulseaudio-modules-bt #1: Post your patches to Pulseaudio maillist

ValdikSS, 2018-08-11:
Hi, there's a patch on Pulseaudio maillist for the modular A2DP API: https://lists.freedesktop.org/archives/pulseaudio-discuss/2018-July/030344.html
Support for aptX: https://lists.freedesktop.org/archives/pulseaudio-discuss/2018-July/030344.html
And a discussion regarding using gstreamer or ffmpeg for additional codecs: https://lists.freedesktop.org/archives/pulseaudio-discuss/2018-July/030201.html

It would be great if you can negotiate the best method of LDAC integration with Pulseaudio developers and commit it directly to the main branch.

MabezDev, 2018-08-31:
Hey just chiming in to say thanks for your awesome work, I'm running your modules under a gentoo system and my sony wh-1000xm2's sound great! Would be awesome if we could get this into patch form.

Keruspe, 2018-10-18:
Rather than the mailing list, I think you could post your patches there: https://gitlab.freedesktop.org/pulseaudio/pulseaudio

EHfive, 2018-10-20:
@pasikarkkainen Yes.
Also planning to add codec switching feature. (Not sure if bluez support this not depend on reconnect)

pasikarkkainen, 2018-10-20:
@EHfive awesome! Yeah, being able to set the preferred codec is important. I think it's OK if a reconnect is required to switch the active codec.

G-Ray, 2018-12-03:
Any progress regarding upstreaming these patches ?

EHfive, 2018-12-13:
> @EHfive awesome! Yeah, being able to set the preferred codec is important. I think it's OK if a reconnect is required to switch the active codec.

Sadly, I abandoned this feature.
It's just unstable via dbus. This function should be introduced in bluez.

The good news is I had post patches to pulseaudio-disscuss. 
https://patchwork.freedesktop.org/series/53988/
https://lists.freedesktop.org/archives/pulseaudio-discuss/2018-December/030669.html

EHfive, 2019-02-05:
Close this issue because merge this to upstream is impossible due to license issues.
FDK-AAC using a custom license.
libldac licensed under Apache 2.0, but it is patented by Sony.

closed 2019-02-05 by EHfive

gombosg, 2019-02-05:
@EHfive I completely understand, was kinda expecting this. `ffmpeg` is patent-encumbered, too, impossible to include in e.g. Debian or Fedora.

I'm running into the same license issues when submitting to Fedora. Can you post the patent info for LDAC? It would accelerate the legal review of `libldac` for Fedora.

inducer, 2019-02-05:
Doesn't Sony releasing the code under [Apache 2](https://www.apache.org/licenses/LICENSE-2.0) entail a patent license for all potential users, as long as they don't sue Sony?

EHfive, 2019-02-05:
> @EHfive I completely understand, was kinda expecting this. `ffmpeg` is patent-encumbered, too, impossible to include in e.g. Debian or Fedora.
>
> I'm running into the same license issues when [...]

The only thing I can found is this:
https://android.googlesource.com/platform/external/libldac/+/master/NOTICE
```
---------------
 Certification
---------------
   Taking the certification process is required to use LDAC in your products.
   For the detail of certification process, see the following URL:
      https://www.sony.net/Products/LDAC/aosp/
```

gombosg, 2019-02-07:
I have gotten the following reply from Fedora Legal:
"The LDAC implementation in libldac might be patented by Sony (the inventor of the codec), but because Sony is also the author of libldac and they put that code under Apache 2.0, which includes an explicit patent grant... It is fine for Fedora.
Because there was some odd wording in the NOTICE file accompanying the libldac source, I had to confirm with Red Hat Legal that we were okay to ignore it and include libldac as Apache 2.0 without that requirement, but they gave me the okay."

@inducer so yes, the library itself is not patent-encumbered, and the NOTICE file is probably only for hardware products where you would use some "native implementation" (e.g. a DSP chip) instead of `libldac` on Android, Linux or whatever, where this Apache v2 license is valid.

mhalano, 2019-02-17:
Any news about that? Seems because libldac has a permissive license (APACHE-2.0) seems possible to integrate the code with upstream.

EHfive, 2019-02-17:
> Any news about that? Seems because libldac has a permissive license (APACHE-2.0) seems possible to integrate the code with upstream.

Watch https://patchwork.freedesktop.org/series/55117/.
I am not working on it currently.

Seems APACHE-2.0 is not compatible with LGPL 2.1.
If we want to add ldac support, we need to license PA under newer version of GPL/LGPL license.

12wk34, 2019-04-04:
Since you have re-license project under GPL3, any news about that?

WingofaGriffin, 2019-05-06:
What about making a fork that just enables aptX into pulse audio, with the LDAC branch remaining separate? I feel like it is good stuff to put into the main branch anyway even if it doesn't cover all codecs.

pasikarkkainen, 2019-05-06:
Pali is currently active on linux-bluetooth and pulseaudio mailinglists, he's working on upstreaming bluez A2DP support for multiple audio codecs. Support is already merged to bluez git master, and the pulseaudio patches are partly merged, other part is still work-in-progress.

ValdikSS, 2019-06-21:
>FDK-AAC support would never be added/merged into upstream due to license issues.

https://lists.fedoraproject.org/archives/list/legal@lists.fedoraproject.org/thread/OVW25JRWOKOLVMW3XGUX7E4OXFUR2RCG/

>The Fedora Project is aware that the Free Software Foundation has stated that the Fraunhofer FDK AAC license is GPL incompatible, specifically, because of Clause 3.
>We believe that the fdk-aac software codec implementation that we wish to include in Fedora is no longer encumbered by AAC patents. This fact means that Clause 3 in the FDK AAC license is a "no op", or to put it plainly, if no patents are in play, there are no patent licenses to disclaim. For this (and only this) specific implementation of fdk-aac, we believe that the FDK AAC license is GPL compatible.

@EHfive, please re-open this issue.

EHfive, 2019-06-21:
> https://lists.fedoraproject.org/archives/list/legal@lists.fedoraproject.org/thread/OVW25JRWOKOLVMW3XGUX7E4OXFUR2RCG/

I see, thanks.

> @EHfive, please re-open this issue.

Since a2dp api part of @pali 's patchset is merged, this repo being incompatible with current PulseAudio. Also, the patch merging cycle is a bit long.
Therefore, I would like to make this repo as a temporary workaround for extra codecs support instead of merging it to upstream.

About adding AAC, LDAC codecs support to PulseAudio, I would like to help after the rest of @pali 's patches are merged.

bhack, 2020-04-09:
Something is moving at https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/227