EHfive/pulseaudio-modules-bt #3: Building DEB package

elvenfighter, 2018-09-08:
I've managed to get a working DEB package by building the binary as described (`cmake .. && make`).
Since I really hope this will be merged to the original PulseAudio (and, consequently will end up in repositories sooner or later), I did not bother with fancy stuff and using CPack. 

However, I wanted to share my script for re-building `pulseaudio-module-bluetooth` using binaries build form these sources and the original DEB shenanigans from (Ubuntu/Debian) repository.

```bash
#!/bin/bash

# Builds pulseaudio-module-bluetooth DEB in current directory
# using binaries from CMake build
# Usage:
#
#   # do the cmake build, e.g.
#   git clone ... && mkdir build && cd build && cmake .. && make
#
#   # build a DEB from that binaries on Ubuntu (or maybe Debian)
#   cd ..
#   mkdir deb && cd deb
#   wget $THIS_SCRIPT_URL -O pkg-build.sh
#   ./pkg-build.sh
#
# This would generate a .deb file with -9git~<hash> in name, install it

BUILD_DIR=$1
PKG_NAME=pulseaudio-module-bluetooth
OS_CODENAME=$(lsb_release -sc)
# agli huck
#REPO_VERSION=$(apt policy "${PKG_NAME}" | grep ubuntu | head -n 1 | awk '{print $1}')

[ ! -d "$BUILD_DIR" ] && BUILD_DIR="../build"
if [ ! -d "$BUILD_DIR" ]; then
  echo "## ERROR: specify build directory with binaries as first argument" >&2
  exit 1
fi

rm -rf "$PKG_NAME"
rm -f *.deb

apt download "${PKG_NAME}/${OS_CODENAME}"

ORIG_PKG=$(find . -maxdepth 1 -name "${PKG_NAME}*.deb")

# extract original packages
mkdir "${PKG_NAME}"
dpkg-deb -x "${ORIG_PKG}" "${PKG_NAME}"
dpkg-deb -e "${ORIG_PKG}" "${PKG_NAME}/DEBIAN"

PLUGINS_DIR=$(find "$PKG_NAME" -type d -name 'pulse-*.*')

# copy new binaries
find "$BUILD_DIR" -maxdepth 1 -type f -name '*.so' -print -exec cp {} "${PLUGINS_DIR}" \;

# update version
GIT_HASH=$(git rev-parse --short HEAD)
sed -e "s/Version: \([0-9]\+:[0-9]\+\.[0-9]\).*$/Version: \1-9git~${GIT_HASH}/" -i "${PKG_NAME}/DEBIAN/control"

# update md5sums
find "${PKG_NAME}" -type f ! -path "${PKG_NAME}/DEBIAN*" -exec md5sum {} + | sed -e "s|${PKG_NAME}/||g" > "${PKG_NAME}/DEBIAN/md5sums"

dpkg-deb -b "${PKG_NAME}" "${PKG_NAME}.deb"
dpkg-name "${PKG_NAME}.deb"
```

adisbladis, 2018-09-16:
I didn't want to create a separate issue, I just wanted to mention there is a NixOS package:
https://github.com/NixOS/nixpkgs/pull/46505

EHfive, 2018-10-17:
PPA source here
https://launchpad.net/~eh5/+archive/ubuntu/pulseaudio-a2dp
```
# Ubuntu 18.04, 18.08  
sudo add-apt-repository ppa:eh5/pulseaudio-a2dp
sudo apt-get update
sudo apt-get install libavcodec libavutil libldac pulseaudio-module-bluetooth
```

evilaliv3, 2018-10-21:
Do you know if someone has created a package for Debian Stretch?

hufon, 2018-10-23:
I'm trying to compile on stretch on my raspberry pi... but it seems complicated due to versions

jgoclawski, 2018-10-31:
@EHfive thanks a lot for your work and especially for the PPA! I've just confirmed it works on Ubuntu 18.04. It changes PulseAudio from 11.1 to 12.2, but works :)

Sennheiser HD 4.50 BTNC with aptX @ Ubuntu 18.04:
```
Name: bluez_sink.00_16_94_27_C3_E6.a2dp_sink
Driver: module-bluez5-device.c
	bluetooth.a2dp_codec = "APTX"
	device.api = "bluez"
	bluez.path = "/org/bluez/hci0/dev_00_16_94_27_C3_E6"
	bluez.class = "0x240404"
	bluez.alias = "HD 4.50BTNC"
```

mateff, 2018-11-02:
@EHfive Thank you so much ! Installed it from PPA, worked like a charm.

> AVDTP(s): Set config cmd: transaction 5 nsp 0x00
    ACP SEID 3 - INT SEID 3
    Media Transport
    Media Codec - non-A2DP (aptX)
      44.1kHz 
      Stereo 
    Delay Reporting
< AVDTP(s): Set config rsp: transaction 5 nsp 0x00

plumlis, 2018-12-04:
It seems new version has AAC supported.
Any plan to upgrade PPA's packages version?

EHfive, 2018-12-04:
> It seems new version has AAC supported.
> Any plan to upgrade PPA's packages version?

I didn't update PPA's packages before because AAC is not such important comparing with aptX, LDAC.
But since you are requesting, I would update it.

Check my PPA next day.

plumlis, 2018-12-04:
> > It seems new version has AAC supported.
> > Any plan to upgrade PPA's packages version?
> 
> I didn't update PPA's packages before because AAC is not such important comparing with aptX, LDAC.
> But since you are requesting, I would update it.
> 
> Check my PPA next day.

So kind of you.
I have Pixel Buds which is support AAC.
I'll have a try after PPA upgrade.

medetcan, 2019-01-20:
First of all thank you for your efforts to bring high quality sound to linux.
I am sorry but i am going to ask an unrelated question. How can i view which configuration and driver picked up my headphone ?

orrpan, 2019-01-23:
> I'm trying to compile on stretch on my raspberry pi... but it seems complicated due to versions

I've successfully compile on stretch on rpi, but when I ran it pulse audio crashes at connect.
Steps (from memory maybe forget somethings) to compile, can do a more exact guide if I get it to work:
```
Pulseaudio 12.2 with bluez5
cmake (install not necessary, use the compiled bin)
x264
ffmpeg
pulseaudio-modules-bt
```

orrpan, 2019-01-30:
```
Linux hem 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

pi@hem:~/build/pulseaudio-modules-bt/build$ pactl list sinks | grep -e bluez -e a2dp_codec
	Name: bluez_sink.40_EF_4C_84_21_83.a2dp_sink
	Driver: module-bluez5-device.c
	Monitor Source: bluez_sink.40_EF_4C_84_21_83.a2dp_sink.monitor
		bluetooth.a2dp_codec = "aptX"
		device.api = "bluez"
		bluez.path = "/org/bluez/hci0/dev_40_EF_4C_84_21_83"
		bluez.class = "0x240414"
		bluez.alias = "Stanmore Bluetooth"
```

Success!

oxwivi, 2019-02-13:
@elvenfighter, I'm trying to install this fork on my Debian unstable, do you recommend compiling a `.deb` package is better or just straight up `make install` taking into consideration future Debian updates that will probably interfere with `make install`d files?

If I use your script to make the package, can I just `dpkg -i` that or will there be anymore dependencies to resolve that may or may not break it all?

elvenfighter, 2019-02-13:
@oxvivi

I was building on Ubuntu, and as you know it is basically Debian Unstable (i.e. Sid). So it should be ok in your case (provided version of pulseaudio is matching).

As for the way of installing, I prefer to keep my system well organized, that is why I prefer installing from packages (and not by `make install`). If you feel the same, I highly recommend install from package

oxwivi, 2019-02-13:
The package created using your script appears to have installed without complaint on my container, but what of the codecs? It doesn't seem to express dependencies on packages that provides `aptX` or `ldac` codecs. `dpkg -S` shows that I do have the relevant libraries installed, but I think your `.deb` should explicitly depend on it if it doesn't already (I'm not familiar with the processing of building packages, so forgive me if I'm calling you on something you already took care of).

elvenfighter, 2019-02-13:
libldac is built & bundled with the package (see contents). However,mainstream (PPA) package does not bundle libldac: it depends on it.

As for aptX, the package probably should depend on ffmpeg. But since I have written the script back in the times, when there was no aptX support in this project, it is not reflected in the script. Since then, I have switched to using the PPA from above: https://launchpad.net/~eh5/+archive/ubuntu/pulseaudio-a2dp

w0k3, 2019-04-09:
Good job, building .deb package in Debian 10 Buster (pa v12.2) worked quite well. I've only used aptX though and I've changed package version of pulseaudio-module-bluetooth to 12.2-4+aptx manually.

orrpan, 2019-04-25:
> > maybe forget somethings) to compile, ca
> 
> Any chance you can share a build guide please?

#55

kcajjones, 2019-05-01:
Has anyone had any success getting this to work on a Raspberry Pi 3b(+) stretch?
I'd love to use my Pi as a bluetooth receiver for AptX/LDAC from my phone to my old (dumb) amp/receiver.

w0k3, 2019-05-02:
I recommend using Buster, Stretch has quite old PA, v10.0.

joffeoja, 2019-05-26:
Would it be possible to offer packages for Disco and Eoan at the Ubuntu PPA? Much appreciated work here

Linux-Fan, 2019-06-05:
I join to request for 19.04 packaging.

b8e5n, 2019-06-09:
+1 for 19.04.
I am using the cosmic one, and it works perfectly (with a restart)
:)

Russell-Jones, 2019-07-09:
@kcajjones If you try to get it going on Debian Buster, and want to use pulseaudio, note that you'll need to remove bluealsa or it will prevent the a2dp profile from becoming available.

tborowy, 2019-09-13:
@w0k3 
> Good job, building .deb package in Debian 10 Buster (pa v12.2) worked quite well. I've only used aptX though and I've changed package version of pulseaudio-module-bluetooth to 12.2-4+aptx manually.

Can you share a file for debian?

w0k3, 2019-09-16:
@tborowy this is my current build, it works flawlessly (switches to aptX automatically whenever I turn my headphones on), but I wouldn't trust unknown sources, use with precaution. I am using version suffix so a new upstream update would replace this version, so you loose aptX.

[pulseaudio-module-bluetooth_12.2-4+deb10u1+aptx_amd64.deb.zip](https://github.com/EHfive/pulseaudio-modules-bt/files/3615288/pulseaudio-module-bluetooth_12.2-4%2Bdeb10u1%2Baptx_amd64.zip)

NedkoHristov, 2019-10-14:
>  HD 4.50

I have the same headphones, but still when I change the mic from integrated on my laptop to HD4.50BTNC sound profile switches from High Fidelity Playback (A2DP Sink) to Headset Head Unit (HSP/HFP).

diegoe, 2019-10-29:
> However, I wanted to share my script for re-building `pulseaudio-module-bluetooth` using binaries build form these sources and the original DEB shenanigans from (Ubuntu/Debian) repository.
> 

Thanks for the script @elvenfighter! A small issue I found building this in Debian unstable: You might need to change the `sed` call that changes the Version: in debian/control to something like:

`sed -e "s/^Version: \(.*\)$/Version: \1-9git+${TIMESTAMP}~${GIT_HASH}/" -i "${PKG_NAME}/DEBIAN/control"`
(Note the simplification of the Version: selector)

Also note that in Debian unstable you'll have to build the `pa` submodule in its master version (this is also explained in this project's README).

And remember that if things explode you can just reinstall the original pulseaudio-modules-bluetooth .deb that should be in the same directory as your now modified version. Good luck!

bhack, 2020-03-21:
Can we add this script to the the repo with a PR?

varishtsg, 2020-04-17:
Is there any difference between  pulseaudio-module-bluetooth and pulseaudio-module-bt

Instructions here suggest installing pulseaudio-module-bluetooth whereas on the main page suggest installing pulseaudio-module-bt.

Which one should be installed?

voideater, 2020-05-16:
Anyone who have made aptX work with Raspiban Stretch on a RPI 3B+ that can share a short how-to?

closed 2020-06-07 by EHfive