EHfive/pulseaudio-modules-bt #7: make failed

ofdrykkja, 2018-09-21:
failed if pulseaudio version is lower than 12

[  6%] Linking C shared library libbluez5-util.so
[ 25%] Built target bluez5-util
[ 31%] Linking C shared library libldacBT_enc.so
[ 50%] Built target ldacBT_enc
[ 56%] Building C object CMakeFiles/module-bluetooth-discover.dir/pa/src/modules/bluetooth/module-bluetooth-discover.c.o
<command-line>:0:0: warning: "__FILE__" redefined [-Wbuiltin-macro-redefined]
In file included from /home/user/pulseaudio-modules-bt/pa/src/modules/bluetooth/module-bluetooth-discover.c:24:0:
/home/user/pulseaudio-modules-bt/pa/src/pulsecore/core-util.h:296:42: warning: ‘struct sockaddr’ declared inside parameter list will not be visible outside of this definition or declaration
 int pa_accept_cloexec(int sockfd, struct sockaddr *addr, socklen_t *addrlen);
                                          ^~~~~~~~
/home/user/pulseaudio-modules-bt/pa/src/modules/bluetooth/module-bluetooth-discover.c:28:10: fatal error: module-bluetooth-discover-symdef.h: Нет такого файла или каталога
 #include "module-bluetooth-discover-symdef.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/module-bluetooth-discover.dir/build.make:62: recipe for target 'CMakeFiles/module-bluetooth-discover.dir/pa/src/modules/bluetooth/module-bluetooth-discover.c.o' failed
make[2]: *** [CMakeFiles/module-bluetooth-discover.dir/pa/src/modules/bluetooth/module-bluetooth-discover.c.o] Error 1
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/module-bluetooth-discover.dir/all' failed
make[1]: *** [CMakeFiles/module-bluetooth-discover.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

EHfive, 2018-09-21:
seems you are using  pulseaudio v11.x
```
#include "module-bluetooth-discover-symdef.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

please update your pulseaudio,;
this module acquire pulseaudio>=12.0, run `pkg-config libpulse --modversion` to check your pulaeaudio version;
or just checket out pulseaudio source to v12.2( `git -C pa/ checkout v12.2`) and compile again
the only different between pa v11 bt module and pa v12's is file struct.(module-bluetooth-discover-symdef.h has been removed in pulseaudio v12)

ofdrykkja, 2018-09-21:
got it. thx for explanation, just thought it's possible for pa 11 because red in README something like "check out your pa version" but now i see it's only for 12 version. you can close this issue

closed 2018-09-21 by EHfive

EHfive, 2018-09-21:
you may try branch `pulseaudio_v11.x` 
https://github.com/EHfive/pulseaudio-modules-bt/tree/pulseaudio_v11.x

ofdrykkja, 2018-09-21:
works like a charm! thank you so much!