EHfive/pulseaudio-modules-bt #10: Resume crashes on Manjaro

petr-nehez, 2018-10-08:
I installed this repo on Manjaro via https://aur.archlinux.org/packages/pulseaudio-modules-bt-git/ but I've found out that resuming from suspend basically dies. As soon as the laptop is woken up it is restarted.

There is nothing in the journal and I've been trying 4.18 and also 4.14. 
I had to uninstall it and since then I can resume my laptop again.

EHfive, 2018-10-08:
Can u confirm this when using `pulseaudio-bluetooth` ?
Please post your /etc/pulse/daemon.conf. (via a pastebin service)

petr-nehez, 2018-10-08:
With `pulseaudio-bluetooth` I have no such problem.
You can check my `/etc/pulse/daemon.conf` at https://pastebin.com/X7Xhejq3 but I wouldn't say there is anything suspicious.

EHfive, 2018-10-08:
Manjaro is my daily use OS, but I can't confirm this isuue anyway.
(using `systemctl suspend` command)
Ideally, this issue should never happen.

petr-nehez, 2018-10-08:
Yes, ideally.
But on my laptop suspend does not work anymore.

As you can see here https://pastebin.com/JrVcA0Zv, there is no error at 10:37:41, it just rebooted.
The only package I had to uninstall was `pulseaudio-modules-bt-git`.

Any idea?

petr-nehez, 2018-10-08:
I double-checked `pacman.log` and for sure no other package was installed or uninstalled.

EHfive, 2018-10-08:
Maybe 
https://ubuntuforums.org/showthread.php?t=1556934&p=10190636#post10190636
https://forum.manjaro.org/search?q=acpi_sleep%3Dnonvs
helps u.

petr-nehez, 2018-10-09:
After updating my Manjaro system I do not face this issue anymore so I guess it had to be an issue with these modules and previous kernel version.

closed 2018-10-09 by petr-nehez