EHfive/pulseaudio-modules-bt #11: Coredump with JACK Sink

FKissen, 2018-10-14:
I am using PulseAudios jack sink (pulseaudio-jack 12.2-2) to route pulse audio to jack. This (sort of) works with the official pulseaudio-bluetooth (12.2-2) package on ArchLinux. But the aur/pulseaudio-modules-bt-git-r55.a228264-1 package fails with a [coredump](https://github.com/EHfive/pulseaudio-modules-bt/files/2476598/pulseaudio-modules-bt--coredump.log).

EHfive, 2018-10-14:
It seems the memblock length less than encoder written. It has no concern with jack.

I've pushed some chances to dev branch. Could you please test it?

FKissen, 2018-10-15:
I would do it, but i don’t know how to modify the PKGBUILD so that it uses this branch.

EHfive, 2018-10-15:
```
build() {
    cd "$srcdir/.."
    git checkout dev
    rm -rf test
    mkdir test
    cd test
    cmake -DCMAKE_BUILD_TYPE=Debug ..
    make
}
```

FKissen, 2018-10-15:
This seems to work at least as bad as the original `pulseaudio-bluetooth`. [Here](http://dpaste.com//0AA29GK) is a log where i switch from SBC to aptX HD on my connected Android phone. At first it complains with this `Synchronization error` but then it seems to work. Albeit with lots of hiccups and buffer underruns.

As for LDAC: I am not sure if my phone even supports this. It is selectable in [this](https://i.imgur.com/lsvOLdV.png) menu, but when i do nothing happens. No further log messages and the displayed codec on the phone stays the same.

Thanks in the meantime.

EHfive, 2018-10-15:
No support of LDAC decoding, only encoding.

commit de703bb0 (2018-10-15): Fix write_block_size,read_block_size calculations

Fix #11 (partially)

Clean up & reformat some lines.
closed 2018-10-15 by EHfive