EHfive/pulseaudio-modules-bt #12: Can not connect to  LDAC device

plumlis, 2018-10-27:
Device: Sony Walkman ZX300A With Bluetooth DAC Mode
System: Ubuntu 18.04 with ppa:eh5/pulseaudio-a2dp installed

Pairing is Okay,and I can connect to the device successful (On my walkman's screen: LDAC)

But After few second, Device disconnected and it seems pulseaudio also crashed.

Btw I install an old version deb file from here and it works out of the box
https://gist.github.com/notpushkin/f84fa287762ceb2756e77434dede1bc1
https://files.catbox.moe/i1tsk1.deb

[image]

EHfive, 2018-10-27:
Please post your log.
(edit /etc/pulse/daemon.conf,  set log-level = 4)

plumlis, 2018-10-27:
Thanks for your reply. I checked the log file and solved this problem temporary.

That's because I added the option"default-sample-rate = 192000" to the configure file "/etc/pulse/daemon.conf" 
The reason I set that option is for Hi-Res music support with my USB DAC. This option cause the problem.
I change it to "default-sample-rate = 44100" and everything works okay now.
I tried to set "alternate-sample-rate = 44100" for an alternate option but I doesn't work.

It seems pulseaudio-modules-bt can only works with 44100hz on LDAC and won't effect with "default-sample-rate" option?

by the way, I found this in log file:

> Oct 27 15:31:54 plum-laptop pulseaudio[9558]: E: [pulseaudio] aptx_libs.c: Cannot open libavcodec library: libavcodec.so. libavcodec.so: 无法打开共享对象文件: 没有那个文件或目录
> Oct 27 15:31:54 plum-laptop pulseaudio[9558]: message repeated 3 times: [ E: [pulseaudio] aptx_libs.c: Cannot open libavcodec library: libavcodec.so. libavcodec.so: 无法打开共享对象文件: 没有那个文件或目录]

I installed libavcodec58 from your PPA but it seems APT-X is still broken.

----------------------------

我检查了一下似乎就是default-sample-rate设定数值过高的锅，但是那个alternate-sample-rate设定的内容并没有作用（哪怕我设置到 44100 之后也无效），我只能先把默认的数值改回来了。

还有就是log里有关于 apt-x 的运行库找不到的提示，我已经安装了相关的包，不知道这算不算bug，我手头也没有 apt-x 的设备可以测试。

EHfive, 2018-10-27:
首先, alternate-sample-rate 的确没用

然后, 到96000hz是没问题的,  176.4 khz, 192 khz 由于我手头设备原因没测试过(所以希望你能贴下在192khz下完整的log)

libavcodec58 我看了下包的内容, 它的动态库名是 libavcodec.so.58 ... 
有两个解决方法 
 * 安装libavcodec-dev 
 * 或者 ln -s libavcodec.so.58 libavcodec.so

EHfive, 2018-10-27:
大概知道原因了，是设置的buffer长度不够

_Sent from my OnePlus ONE A2001 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

EHfive, 2018-10-27:
Seems the LDAC encoder provided dose not support 176.4 khz, 192 khz.
I simply disable it.
I will push updated sources to ppa:eh5/pulseaudio-a2dp later.

closed 2018-10-27 by EHfive

plumlis, 2018-10-27:
多谢回复，我试了一下96000的确是可以的。
192000的log抓了一下，从pulseaudio kill掉到重新启动后连接设备又失败的内容都在里面了。
[192000hz.log](https://github.com/EHfive/pulseaudio-modules-bt/files/2521619/192000hz.log)
希望有所帮助，最后感谢你的工作与贡献 :)

EHfive, 2018-10-27:
Now, you can update updated packages from ppa:eh5/pulseaudio-a2dp.
Then, modify default-sample-rate as you like.

plumlis, 2018-10-27:
Thanks, I'll update later

commit 2e73f8fc (2018-10-27): Disable LDAC 176.4 khz, 192 khz capabilities; Fix #12
commit 8077877b (2018-11-17): Disable LDAC 176.4 khz, 192 khz capabilities; Fix #12