EHfive/pulseaudio-modules-bt #14: Autoconnected device won't switch to LDAC

shouya, 2018-11-07:
Thank you for this fantastic project first of all! It works great! However there's a small issue with bluetooth auto connect.

To be more elaborative, if I turn off my headphone while connected to a computer, and then turn the headphone back on again, the headphone will automatically reconnect, but in a low quality mono channel 8kHz mode. However, this can be easily fixed by simply disconnect and reconnect again (from the computer using commands like 'echo disconnect <MAC> | bluetoothctl; echo connect <MAC> | bluetoothctl'). After the reconnection it will be in LDAC mode as expected.

I'm not familiar with details of pulseaudio or bluetooth, so it could be a trivial question. I searched the Internet and find no useful information to this situation.

Here's some basic info:

```
OS: Gentoo x64 with Linux kernel 4.18.6
Wireless adapter: 03:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM4360 802.11ac Wireless Network Adapter [14e4:43a0] (rev 03)
Headphone: MDR-1000X
pulseaudio --version: pulseaudio 12.2
bluetoothctl --version: bluetoothctl: 5.50
```

<details>
  <summary>Here's some outputs from `pactl list sinks` (Click to expand)</summary>

```
# after bluetooth autoconnect
Sink #25
	State: SUSPENDED
	Name: bluez_sink.<MASKED>.headset_head_unit
	Description: MDR-1000X
	Driver: module-bluez5-device.c
	Sample Specification: s16le 1ch 8000Hz
	Channel Map: mono
	Owner Module: 48
	Mute: no
	Volume: mono: 38610 /  59%
	        balance 0.00
	Base Volume: 65536 / 100%
	Monitor Source: bluez_sink.<MASKED>.headset_head_unit.monitor
	Latency: 0 usec, configured 0 usec
	Flags: HARDWARE HW_VOLUME_CTRL LATENCY 
	Properties:
		bluetooth.protocol = "headset_head_unit"
		device.intended_roles = "phone"
		device.description = "MDR-1000X"
		device.string = "<MASKED>"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headset"
		bluez.path = "/org/bluez/hci0/dev_<MASKED>"
		bluez.class = "0x240404"
		bluez.alias = "MDR-1000X"
		device.icon_name = "audio-headset-bluetooth"
	Ports:
		headset-output: Headset (priority: 0)
	Active Port: headset-output
	Formats:
		pcm

# after reconnection
Sink #27
	State: SUSPENDED
	Name: bluez_sink.<MASKED>.a2dp_sink
	Description: MDR-1000X
	Driver: module-bluez5-device.c
	Sample Specification: s32le 2ch 44100Hz
	Channel Map: front-left,front-right
	Owner Module: 50
	Mute: no
	Volume: front-left: 32717 /  50% / -18.10 dB,   front-right: 32717 /  50% / -18.10 dB
	        balance 0.00
	Base Volume: 65536 / 100% / 0.00 dB
	Monitor Source: bluez_sink.<MASKED>.a2dp_sink.monitor
	Latency: 0 usec, configured 0 usec
	Flags: HARDWARE DECIBEL_VOLUME LATENCY 
	Properties:
		bluetooth.protocol = "a2dp_sink"
		bluetooth.a2dp_codec = "LDAC"
		device.description = "MDR-1000X"
		device.string = "<MASKED>"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headset"
		bluez.path = "/org/bluez/hci0/dev_<MASKED>"
		bluez.class = "0x240404"
		bluez.alias = "MDR-1000X"
		device.icon_name = "audio-headset-bluetooth"
		device.intended_roles = "phone"
	Ports:
		headset-output: Headset (priority: 0)
	Active Port: headset-output
	Formats:
		pcm
```

</details>

<details>
  <summary>And my `/etc/pulse/default.pa` config (Click to expand)</summary>

```config
#!/usr/bin/pulseaudio -nF
#
# This file is part of PulseAudio.
#
# PulseAudio is free software; you can redistribute it and/or modify it
# under the terms of the GNU Lesser General Public License as published by
# the Free Software Foundation; either version 2 of the License, or
# (at your option) any later version.
#
# PulseAudio is distributed in the hope that it will be useful, but
# WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU
# General Public License for more details.
#
# You should have received a copy of the GNU Lesser General Public License
# along with PulseAudio; if not, see <http://www.gnu.org/licenses/>.

# This startup script is used only if PulseAudio is started per-user
# (i.e. not in system mode)

.fail

### Automatically restore the volume of streams and devices
load-module module-device-restore
load-module module-stream-restore
load-module module-card-restore

### Automatically augment property information from .desktop files
### stored in /usr/share/application
load-module module-augment-properties

### Should be after module-*-restore but before module-*-detect
load-module module-switch-on-port-available

### Load audio drivers statically
### (it's probably better to not load these drivers manually, but instead
### use module-udev-detect -- see below -- for doing this automatically)
#load-module module-alsa-sink
#load-module module-alsa-source device=hw:1,0
#load-module module-null-sink
#load-module module-pipe-sink

### Automatically load driver modules depending on the hardware available
.ifexists module-udev-detect.so
load-module module-udev-detect
.else
### Use the static hardware detection module (for systems that lack udev support)
load-module module-detect
.endif

### Automatically connect sink and source if JACK server is present
.ifexists module-jackdbus-detect.so
.nofail
load-module module-jackdbus-detect channels=2
.fail
.endif

### Automatically load driver modules for Bluetooth hardware
.ifexists module-bluetooth-policy.so
load-module module-bluetooth-policy
.endif

.ifexists module-bluetooth-discover.so
# load-module module-bluetooth-discover
load-module module-bluetooth-discover a2dp_config="ldac_eqmid=abr ldac_fmt=s32"
.endif

### Load several protocols
.ifexists module-esound-protocol-unix.so
load-module module-esound-protocol-unix
.endif
load-module module-native-protocol-unix

### Network access (may be configured with paprefs, so leave this commented
### here if you plan to use paprefs)
#load-module module-esound-protocol-tcp
#load-module module-native-protocol-tcp

### Load the RTP receiver module (also configured via paprefs, see above)
#load-module module-rtp-recv

### Load the RTP sender module (also configured via paprefs, see above)
#load-module module-null-sink sink_name=rtp format=s16be channels=2 rate=44100 sink_properties="device.description='RTP Multicast Sink'"
#load-module module-rtp-send source=rtp.monitor

### Load additional modules from GSettings. This can be configured with the paprefs tool.
### Please keep in mind that the modules configured by paprefs might conflict with manually
### loaded modules.
.ifexists module-gsettings.so
.nofail
load-module module-gsettings
.fail
.endif

### Automatically restore the default sink/source when changed by the user
### during runtime
### NOTE: This should be loaded as early as possible so that subsequent modules
### that look up the default sink/source get the right value
load-module module-default-device-restore

### Automatically move streams to the default sink if the sink they are
### connected to dies, similar for sources
load-module module-rescue-streams

### Make sure we always have a sink around, even if it is a null sink.
load-module module-always-sink

### Honour intended role device property
load-module module-intended-roles

### Automatically suspend sinks/sources that become idle for too long
load-module module-suspend-on-idle

### If autoexit on idle is enabled we want to make sure we only quit
### when no local session needs us anymore.
.ifexists module-console-kit.so
load-module module-console-kit
.endif
.ifexists module-systemd-login.so
load-module module-systemd-login
.endif

### Enable positioned event sounds
load-module module-position-event-sounds

### Cork music/video streams when a phone stream is active
load-module module-role-cork

### Modules to allow autoloading of filters (such as echo cancellation)
### on demand. module-filter-heuristics tries to determine what filters
### make sense, and module-filter-apply does the heavy-lifting of
### loading modules and rerouting streams.
load-module module-filter-heuristics
load-module module-filter-apply

### Make some devices default
#set-default-sink output
#set-default-source input
```
</details>

EHfive, 2018-11-07:
It's an old issue. https://gitlab.freedesktop.org/pulseaudio/pulseaudio/issues/525
fac42e4 fixes it, I would merge this commit, but not now.

https://github.com/EHfive/pulseaudio-modules-bt#load-modules
> as an alternative, you can fix it with this udev script

EHfive, 2019-02-09:
Seems the issue been fixed in latest bluez git master.

Bluez version
```
$ git describe --long --tags
5.50-264-g750a26cd9
```
My build configurations
```
$ ./bootstrap
......
$ ./configure \
          --prefix=/usr \
          --mandir=/usr/share/man \
          --sysconfdir=/etc \
          --localstatedir=/var \
          --libexecdir=/usr/lib \
          --with-dbusconfdir=/usr/share \
          --enable-btpclient \
          --enable-midi \
          --enable-sixaxis \
          --enable-mesh \
          --enable-external-ell \ # require ell>=0.13
          --enable-experimental \
          --enable-library
```

I did't confirm which commit in bluez git fix the issue.
But the problem would disappear in next bluez release(>5.50).

EHfive, 2019-02-09:
It is https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=477ecca127c529611adbc53f08039cefaf86305d fixes the issue.

23 days later than Bluez 5.50 released, LOL.

closed 2019-02-09 by EHfive

dlbnco, 2019-05-15:
Did all of the above and I still get the issue.

mmstick, 2019-05-21:
Is just that commit required? Thinking about patching bluez and pulseaudio in Pop!_OS so we can have working Bluetooth quality today.

EHfive, 2019-05-22:
> Is just that commit required? Thinking about patching bluez and pulseaudio in Pop!_OS so we can have working Bluetooth quality today.

Yes!

dlbnco, 2019-05-23:
@mmstick just upgraded my Pop!_OS and the problem is fixed! Thank you so much :)

EHfive, 2019-10-03:
> It is https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=477ecca127c529611adbc53f08039cefaf86305d fixes the issue.

Bluez 5.51 has relesaed with those patches fixed this bug.

> http://www.bluez.org/release-of-bluez-5-51/

evilscientress, 2019-10-11:
Since I updated to 5.51 I also can switch between all codecs my WH-1000XM3 supports. Before Pulseaudio would only let me select LDAC and listed all other codecs/sinks as unavailable, even when the connection originated from the PC. Guess that boiled down to the same issue addressed by the patch of bluez not connecting all sinks.

mattia-b89, 2020-06-20:
I have a very similar (if not the same) issue:  
I configured my bluetooth headphone using APTX HD but sometimes it connects using HSP/HFP profile
I can "easily" fix it by using `pavucontrol` utlity and choosing preferred mode in the drop-down menu because other modes are even available!

I am on Arch Linux w/ bluez-5.54-2 pulseaudio 13.0-3 and pulseaudio-modules-bt-git r146.b02f9b6-1 (as of 18 Apr 2020)