EHfive/pulseaudio-modules-bt #16: Include Ubuntu 16.04 in the package

timomulder, 2018-11-14:
I had a tough time building the package on Ubuntu 16.04.5. Any chance this version could be added to the PPA? I'd love to use this!
Thanks

EHfive, 2018-11-14:
Sorry, I wouldn't do that.
There are so many dependencies (ffmpeg 4.0's pulseaudio 12.2's) need to be updated  if we port it to Ubuntu 16. 
I recommend you to update your Ubuntu to 18.04 .

Or, you may try alsa-bluez.

timomulder, 2018-11-14:
Yeah I was afraid of that.. Probably the reason I had such a tough time. Thanks for your suggestion!

closed 2018-11-14 by timomulder

mstarodub, 2018-12-17:
@timomulder Did you manage to successfully replace pulseaudio. libpulse and ffmpeg with the required versions on your 16.04 system?

timomulder, 2018-12-19:
> @timomulder Did you manage to successfully replace pulseaudio. libpulse and ffmpeg with the required versions on your 16.04 system?

Unfortunately I didn't. I was able to compile the latest version of ffmpeg using this guide: https://trac.ffmpeg.org/wiki/CompilationGuide/Ubuntu
but then I wasn't able to properly add all of the elements to my path such that it could be picked up by other programs (I think specifically the new version of libavcodec needed to be recognised by cmake). I must say that I'm not very knowledgeable in compiling programs and making them available system wide, so maybe it's not that hard, but just my lack of knowledge. I might try again at some point in the future, or upgrade to 18.04 as EHfive suggested.
Let me know if you have any luck yourself!

mstarodub, 2019-01-02:
@timomulder I successfully compiled and installed all required packages on my system using a PPA for ffmpeg and compiling pulse like [this](https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/Developer/PulseAudioFromGit/), but It seems that I'm stuck at the `make` stage of the installation (after a successful `cmake ..`).

Maybe @EHfive can help?
```
~/pulseaudio-modules-bt/build$ make
Scanning dependencies of target module-bluetooth-policy
[  4%] Building C object CMakeFiles/module-bluetooth-policy.dir/pa/src/modules/bluetooth/module-bluetooth-policy.c.o
/home/mxs/pulseaudio-modules-bt/pa/src/modules/bluetooth/module-bluetooth-policy.c:34:44: fatal error: module-bluetooth-policy-symdef.h: No such file or directory
compilation terminated.
CMakeFiles/module-bluetooth-policy.dir/build.make:62: recipe for target 'CMakeFiles/module-bluetooth-policy.dir/pa/src/modules/bluetooth/module-bluetooth-policy.c.o' failed
make[2]: *** [CMakeFiles/module-bluetooth-policy.dir/pa/src/modules/bluetooth/module-bluetooth-policy.c.o] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/module-bluetooth-policy.dir/all' failed
make[1]: *** [CMakeFiles/module-bluetooth-policy.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

EHfive, 2019-01-03:
@mstarodub 
master branch require pulseaudio >= 12.0

EHfive, 2019-01-03:
> 
> 
> @EHfive
> 
> ```
> pulseaudio --version
> pulseaudio 12.0-252-g2eb8-dirty
> ```
> 
> I think I compiled and installed the dependencies

Run `pkg-config libpulse --modversion|sed 's/[^0-9.]*\([0-9.]*\).*/\1/'` instead

mstarodub, 2019-01-03:
```
pkg-config libpulse --modversion|sed 's/[^0-9.]*\([0-9.]*\).*/\1/'
12.0
```

EHfive, 2019-01-03:
Have you switched PA source to v12 ?
```
git -C pa/ checkout v`pkg-config libpulse --modversion|sed 's/[^0-9.]*\([0-9.]*\).*/\1/'
```

`module-bluetooth-policy-symdef.h` only appered under PA<=11

mstarodub, 2019-01-03:
Did that, now it's stuck at a different stage:

```
[ 45%] Building C object CMakeFiles/bluez5-util.dir/src/modules/bluetooth/a2dp/a2dp_aac.c.o
/home/mxs/pulseaudio-modules-bt/src/modules/bluetooth/a2dp/a2dp_aac.c: In function ‘pa_aac_config_transport’:
/home/mxs/pulseaudio-modules-bt/src/modules/bluetooth/a2dp/a2dp_aac.c:452:60: error: ‘AACENC_AUDIOMUXVER’ undeclared (first use in this function)
     aac_err = aacEncoder_SetParam(aac_info->aacenc_handle, AACENC_AUDIOMUXVER, 2);
                                                            ^
/home/mxs/pulseaudio-modules-bt/src/modules/bluetooth/a2dp/a2dp_aac.c:452:60: note: each undeclared identifier is reported only once for each function it appears in
/home/mxs/pulseaudio-modules-bt/src/modules/bluetooth/a2dp/a2dp_aac.c: In function ‘pa_aac_setup_stream’:
/home/mxs/pulseaudio-modules-bt/src/modules/bluetooth/a2dp/a2dp_aac.c:534:60: error: ‘AACENC_PEAK_BITRATE’ undeclared (first use in this function)
     aac_err = aacEncoder_SetParam(aac_info->aacenc_handle, AACENC_PEAK_BITRATE, (UINT) max_bitra
```

EHfive, 2019-01-03:
require libfdk-aac >= 0.1.5

mstarodub, 2019-01-03:
Thank you, that was the last missing one :smile_cat: 

Now I have to test it with a bluetooth device

mstarodub, 2019-01-08:
@EHfive 
After playing around with a LDAC capable bluetooth device, I noticed that I didn't even have `module-bluetooth-discover` loaded so I had to do that, to even be able to connect to a bluetooth device in the first place.

However, when I try loading it with arguments, this happens:
```
pactl load-module module-bluetooth-discover a2dp_config="ldac_eqmid=sq"
Failure: Module initialization failed
```
Apending `load-module module-bluetooth-discover a2dp_config="ldac_eqmid=sq"` into `/etc/pulse/default.pa` results in a segfault when starting pulseaudio, and interestingly I can't restart it afterwards, even when removing that line from the config file and trying again - I had to reboot my system.

The same thing happens with `module-bluez5-discover`.

My  `/etc/pulse/default.pa` looks like this:
```
#!/usr/local/bin/pulseaudio -nF
#
# This file is part of PulseAudio.
#
# PulseAudio is free software; you can redistribute it and/or modify it
# under the terms of the GNU Lesser General Public License as published by
# the Free Software Foundation; either version 2 of the License, or
# (at your option) any later version.
#
# PulseAudio is distributed in the hope that it will be useful, but
# WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU
# General Public License for more details.
#
# You should have received a copy of the GNU Lesser General Public License
# along with PulseAudio; if not, see <http://www.gnu.org/licenses/>.

# This startup script is used only if PulseAudio is started per-user
# (i.e. not in system mode)

.fail

### Automatically restore the volume of streams and devices
load-module module-device-restore
load-module module-stream-restore
load-module module-card-restore

### Automatically augment property information from .desktop files
### stored in /usr/share/application
load-module module-augment-properties

### Should be after module-*-restore but before module-*-detect
load-module module-switch-on-port-available

### Load audio drivers statically
### (it's probably better to not load these drivers manually, but instead
### use module-udev-detect -- see below -- for doing this automatically)
#load-module module-alsa-sink
#load-module module-alsa-source device=hw:1,0
#load-module module-oss device="/dev/dsp" sink_name=output source_name=input
#load-module module-oss-mmap device="/dev/dsp" sink_name=output source_name=input
#load-module module-null-sink
#load-module module-pipe-sink

### Automatically load driver modules depending on the hardware available
.ifexists module-udev-detect.so
load-module module-udev-detect
.else
### Use the static hardware detection module (for systems that lack udev support)
load-module module-detect
.endif

### Automatically connect sink and source if JACK server is present
.ifexists module-jackdbus-detect.so
.nofail
load-module module-jackdbus-detect channels=2
.fail
.endif

### Load several protocols
.ifexists module-esound-protocol-unix.so
load-module module-esound-protocol-unix
.endif
load-module module-native-protocol-unix

### Network access (may be configured with paprefs, so leave this commented
### here if you plan to use paprefs)
#load-module module-esound-protocol-tcp
#load-module module-native-protocol-tcp

### Load the RTP receiver module (also configured via paprefs, see above)
#load-module module-rtp-recv

### Load the RTP sender module (also configured via paprefs, see above)
#load-module module-null-sink sink_name=rtp format=s16be channels=2 rate=44100 sink_properties="device.description='RTP Multicast Sink'"
#load-module module-rtp-send source=rtp.monitor

### Load additional modules from GSettings. This can be configured with the paprefs tool.
### Please keep in mind that the modules configured by paprefs might conflict with manually
### loaded modules.
.ifexists module-gsettings.so
.nofail
load-module module-gsettings
.fail
.endif

### Automatically restore the default sink/source when changed by the user
### during runtime
### NOTE: This should be loaded as early as possible so that subsequent modules
### that look up the default sink/source get the right value
load-module module-default-device-restore

### Automatically move streams to the default sink if the sink they are
### connected to dies, similar for sources
load-module module-rescue-streams

### Make sure we always have a sink around, even if it is a null sink.
load-module module-always-sink

### Honour intended role device property
load-module module-intended-roles

### Automatically suspend sinks/sources that become idle for too long
load-module module-suspend-on-idle

### If autoexit on idle is enabled we want to make sure we only quit
### when no local session needs us anymore.
.ifexists module-console-kit.so
load-module module-console-kit
.endif
.ifexists module-systemd-login.so
load-module module-systemd-login
.endif

### Enable positioned event sounds
load-module module-position-event-sounds

### Cork music/video streams when a phone stream is active
load-module module-role-cork

### Modules to allow autoloading of filters (such as echo cancellation)
### on demand. module-filter-heuristics tries to determine what filters
### make sense, and module-filter-apply does the heavy-lifting of
### loading modules and rerouting streams.
load-module module-filter-heuristics
load-module module-filter-apply

### Make some devices default
#set-default-sink output
#set-default-source input

load-module module-bluetooth-discover a2dp_config="ldac_eqmid=sq"
```

Any help would be appreciated.

EHfive, 2019-01-09:
@mstarodub 

Please post your logs of pulseaudio.

mstarodub, 2019-01-09:
After seeing the logs, I noticed the file permissions on the compiled modules in `/usr/local/lib/pulse-12.0/modules` didn't match the other modules, so I changed them. 

But that didn't solve the issue. Here are the logs:
`pulse-success.log` (with `load-module module-bluetooth-discover a2dp_config="ldac_eqmid=sq"` commented out):
```
(   0.000|   0.000) D: [pulseaudio] conf-parser.c: Parsing configuration file '/etc/pulse/client.conf'
(   0.000|   0.000) D: [pulseaudio] conf-parser.c: /etc/pulse/client.conf.d does not exist, ignoring.
(   0.242|   0.242) I: [pulseaudio] main.c: setrlimit(RLIMIT_NICE, (31, 31)) failed: Operation not permitted
(   0.242|   0.000) I: [pulseaudio] main.c: setrlimit(RLIMIT_RTPRIO, (9, 9)) failed: Operation not permitted
(   0.242|   0.000) D: [pulseaudio] core-rtclock.c: Timer slack is set to 50 us.
(   0.242|   0.000) I: [pulseaudio] core-util.c: Failed to acquire high-priority scheduling: Permission denied
(   0.242|   0.000) I: [pulseaudio] main.c: This is PulseAudio 12.0-252-g2eb8-dirty
(   0.242|   0.000) D: [pulseaudio] main.c: Compilation host: x86_64-pc-linux-gnu
(   0.242|   0.000) D: [pulseaudio] main.c: Compilation CFLAGS:  -g -O0 -Wall -W -Wextra -pipe -Wno-long-long -Wno-overlength-strings -Wunsafe-loop-optimizations -Wundef -Wformat=2 -Wlogical-op -Wsign-compare -Wformat-security -Wmissing-include-dirs -Wformat-nonliteral -Wpointer-arith -Winit-self -Wdeclaration-after-statement -Wfloat-equal -Wmissing-prototypes -Wredundant-decls -Wmissing-declarations -Wmissing-noreturn -Wshadow -Wendif-labels -Wcast-align -Wstrict-aliasing -Wwrite-strings -Wno-unused-parameter -ffast-math -fno-common -fdiagnostics-show-option -fdiagnostics-color=auto
(   0.242|   0.000) D: [pulseaudio] main.c: Running on host: Linux x86_64 4.15.0-43-generic #46~16.04.1-Ubuntu SMP Fri Dec 7 13:31:08 UTC 2018
(   0.242|   0.000) D: [pulseaudio] main.c: Found 4 CPUs.
(   0.242|   0.000) I: [pulseaudio] main.c: Page size is 4096 bytes
(   0.242|   0.000) D: [pulseaudio] main.c: Compiled with Valgrind support: no
(   0.242|   0.000) D: [pulseaudio] main.c: Running in valgrind mode: no
(   0.242|   0.000) D: [pulseaudio] main.c: Running in VM: no
(   0.242|   0.000) D: [pulseaudio] main.c: Running from build tree: no
(   0.242|   0.000) D: [pulseaudio] main.c: Optimized build: no
(   0.242|   0.000) D: [pulseaudio] main.c: All asserts enabled.
(   0.243|   0.000) I: [pulseaudio] main.c: Machine ID is 61ee9ff06e004c2db15bd82588da1089.
(   0.243|   0.000) I: [pulseaudio] main.c: Session ID is c2.
(   0.243|   0.000) I: [pulseaudio] main.c: Using runtime directory /run/user/1000/pulse.
(   0.243|   0.000) I: [pulseaudio] main.c: Using state directory /home/mxs/.config/pulse.
(   0.243|   0.000) I: [pulseaudio] main.c: Using modules directory /usr/local/lib/pulse-12.0/modules.
(   0.243|   0.000) I: [pulseaudio] main.c: Running in system mode: no
(   0.243|   0.000) E: [pulseaudio] pid.c: Daemon already running.
(   0.243|   0.243) I: [pulseaudio] main.c: Daemon startup successful.
```

`pulse-fail.log` (with the line commented in)
```
(   0.000|   0.000) D: [pulseaudio] conf-parser.c: Parsing configuration file '/etc/pulse/client.conf'
(   0.000|   0.000) D: [pulseaudio] conf-parser.c: /etc/pulse/client.conf.d does not exist, ignoring.
(   0.003|   0.002) I: [pulseaudio] main.c: setrlimit(RLIMIT_NICE, (31, 31)) failed: Operation not permitted
(   0.003|   0.000) I: [pulseaudio] main.c: setrlimit(RLIMIT_RTPRIO, (9, 9)) failed: Operation not permitted
(   0.003|   0.000) D: [pulseaudio] core-rtclock.c: Timer slack is set to 50 us.
(   0.003|   0.000) I: [pulseaudio] core-util.c: Failed to acquire high-priority scheduling: Permission denied
(   0.003|   0.000) I: [pulseaudio] main.c: This is PulseAudio 12.0-252-g2eb8-dirty
(   0.003|   0.000) D: [pulseaudio] main.c: Compilation host: x86_64-pc-linux-gnu
(   0.003|   0.000) D: [pulseaudio] main.c: Compilation CFLAGS:  -g -O0 -Wall -W -Wextra -pipe -Wno-long-long -Wno-overlength-strings -Wunsafe-loop-optimizations -Wundef -Wformat=2 -Wlogical-op -Wsign-compare -Wformat-security -Wmissing-include-dirs -Wformat-nonliteral -Wpointer-arith -Winit-self -Wdeclaration-after-statement -Wfloat-equal -Wmissing-prototypes -Wredundant-decls -Wmissing-declarations -Wmissing-noreturn -Wshadow -Wendif-labels -Wcast-align -Wstrict-aliasing -Wwrite-strings -Wno-unused-parameter -ffast-math -fno-common -fdiagnostics-show-option -fdiagnostics-color=auto
(   0.003|   0.000) D: [pulseaudio] main.c: Running on host: Linux x86_64 4.15.0-43-generic #46~16.04.1-Ubuntu SMP Fri Dec 7 13:31:08 UTC 2018
(   0.003|   0.000) D: [pulseaudio] main.c: Found 4 CPUs.
(   0.003|   0.000) I: [pulseaudio] main.c: Page size is 4096 bytes
(   0.003|   0.000) D: [pulseaudio] main.c: Compiled with Valgrind support: no
(   0.003|   0.000) D: [pulseaudio] main.c: Running in valgrind mode: no
(   0.003|   0.000) D: [pulseaudio] main.c: Running in VM: no
(   0.003|   0.000) D: [pulseaudio] main.c: Running from build tree: no
(   0.003|   0.000) D: [pulseaudio] main.c: Optimized build: no
(   0.003|   0.000) D: [pulseaudio] main.c: All asserts enabled.
(   0.003|   0.000) I: [pulseaudio] main.c: Machine ID is 61ee9ff06e004c2db15bd82588da1089.
(   0.003|   0.000) I: [pulseaudio] main.c: Session ID is c2.
(   0.004|   0.000) I: [pulseaudio] main.c: Using runtime directory /run/user/1000/pulse.
(   0.004|   0.000) I: [pulseaudio] main.c: Using state directory /home/mxs/.config/pulse.
[... cut ...]
(   0.209|   0.000) D: [pulseaudio] module-alsa-card.c: Jack 'HDMI/DP,pcm=3 Jack' is now unplugged
(   0.209|   0.000) D: [pulseaudio] device-port.c: Setting port hdmi-output-0 to status no
(   0.209|   0.000) D: [pulseaudio] card.c: Setting card alsa_card.pci-0000_00_03.0 profile output:hdmi-stereo to availability status no
(   0.217|   0.217) E: [pulseaudio] main.c: Daemon startup failed.
```

Not sure why there are so many attempts to connect to an HDMI output. There's nothing plugged into the HDMI jack of my machine.

EHfive, 2019-01-09:
@mstarodub 
Return back to your previous comment, I find that you missed "module-bluetooth-policy"

Configure default.pa like this
```
.ifexists module-bluetooth-policy.so
load-module module-bluetooth-policy 
.endif

.ifexists module-bluetooth-discover.so
load-module module-bluetooth-discover
.endif

```

mstarodub, 2019-01-09:
Without ` a2dp_config="ldac_eqmid=sq"`?

EHfive, 2019-01-09:
> Without ` a2dp_config="ldac_eqmid=sq"`?

It doesn't matter. 
module-bluetooth-discover require module-bluetooth-policy

mstarodub, 2019-01-09:
```
pactl list modules short | grep blue
19	module-bluetooth-policy		
20	module-bluetooth-discover		
21	module-bluez5-discover	
```
Trying with the additional parameter now.

EHfive, 2019-01-09:
@mstarodub 
If you still got errors.
I recommend you patch https://github.com/EHfive/pulseaudio-modules-bt/releases/tag/v1.0 instead of build modules using cmake.
Your logs are incomplete, I can't help through it.

mstarodub, 2019-01-09:
Appending `a2dp_config="ldac_eqmid=sq"` causes Pulseaudio to fail on startup.

What part of the log files did I miss? Sorry about that. The way I set up logging was to put `extra-arguments = -vvvv --log-target=newfile:/tmp/pulseverbose.log --log-time=1` at the end of `/etc/pulse/client.conf`, this resulted in lots of similar pulseverbose.log.x files on a failed attempt to load the modules, with x from 1-99. I just uploaded one of those randomly.

mstarodub, 2019-01-09:
Should I revert some changes I already made prior to doing `sudo patch < patch-1.0-pulseaudio-12.0-bluez5-a2dp-extra-codecs.patch`?

mstarodub, 2019-01-09:
Could you please elaborate a bit on how this patch should be applied and installed? I've been trying to figure it out myself, but I'm kind of lost.

mstarodub, 2019-01-12:
@EHfive I'm sorry for constantly bothering you for support with this issue, but I feel like there's only a tiny bit missing for this to successfully work. If I'm not giving you enough information/logs, please tell me what I missed so that I can provide it.

I wasn't able to successfully apply the patch, so I compiled it again, as explained in the readme.
However, I'm still getting 
```
pactl load-module module-bluetooth-discover a2dp_config="ldac_eqmid=sq"
Failure: Module initialization failed
```
when loading with the argument ` a2dp_config="ldac_eqmid=sq`. `pactl load-module module-bluetooth-discover` works fine without arguments. `module-bluetooth-policy` is loaded.

My `default.pa`:
```
#!/usr/local/bin/pulseaudio -nF
#
# This file is part of PulseAudio.
#
# PulseAudio is free software; you can redistribute it and/or modify it
# under the terms of the GNU Lesser General Public License as published by
# the Free Software Foundation; either version 2 of the License, or
# (at your option) any later version.
#
# PulseAudio is distributed in the hope that it will be useful, but
# WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU
# General Public License for more details.
#
# You should have received a copy of the GNU Lesser General Public License
# along with PulseAudio; if not, see <http://www.gnu.org/licenses/>.

# This startup script is used only if PulseAudio is started per-user
# (i.e. not in system mode)

.fail

### Automatically restore the volume of streams and devices
load-module module-device-restore
load-module module-stream-restore
load-module module-card-restore

### Automatically augment property information from .desktop files
### stored in /usr/share/application
load-module module-augment-properties

### Should be after module-*-restore but before module-*-detect
load-module module-switch-on-port-available

### Load audio drivers statically
### (it's probably better to not load these drivers manually, but instead
### use module-udev-detect -- see below -- for doing this automatically)
#load-module module-alsa-sink
#load-module module-alsa-source device=hw:1,0
#load-module module-oss device="/dev/dsp" sink_name=output source_name=input
#load-module module-oss-mmap device="/dev/dsp" sink_name=output source_name=input
#load-module module-null-sink
#load-module module-pipe-sink

### Automatically load driver modules depending on the hardware available
.ifexists module-udev-detect.so
load-module module-udev-detect
.else
### Use the static hardware detection module (for systems that lack udev support)
load-module module-detect
.endif

### Automatically connect sink and source if JACK server is present
.ifexists module-jackdbus-detect.so
.nofail
load-module module-jackdbus-detect channels=2
.fail
.endif

### Load several protocols
.ifexists module-esound-protocol-unix.so
load-module module-esound-protocol-unix
.endif
load-module module-native-protocol-unix

### Network access (may be configured with paprefs, so leave this commented
### here if you plan to use paprefs)
#load-module module-esound-protocol-tcp
load-module module-native-protocol-tcp auth-ip-acl=127.0.0.1

### Load the RTP receiver module (also configured via paprefs, see above)
#load-module module-rtp-recv

### Load the RTP sender module (also configured via paprefs, see above)
#load-module module-null-sink sink_name=rtp format=s16be channels=2 rate=44100 sink_properties="device.description='RTP Multicast Sink'"
#load-module module-rtp-send source=rtp.monitor

### Load additional modules from GSettings. This can be configured with the paprefs tool.
### Please keep in mind that the modules configured by paprefs might conflict with manually
### loaded modules.
.ifexists module-gsettings.so
.nofail
load-module module-gsettings
.fail
.endif

### Automatically restore the default sink/source when changed by the user
### during runtime
### NOTE: This should be loaded as early as possible so that subsequent modules
### that look up the default sink/source get the right value
load-module module-default-device-restore

### Automatically move streams to the default sink if the sink they are
### connected to dies, similar for sources
load-module module-rescue-streams

### Make sure we always have a sink around, even if it is a null sink.
load-module module-always-sink

### Honour intended role device property
load-module module-intended-roles

### Automatically suspend sinks/sources that become idle for too long
load-module module-suspend-on-idle

### If autoexit on idle is enabled we want to make sure we only quit
### when no local session needs us anymore.
.ifexists module-console-kit.so
load-module module-console-kit
.endif
.ifexists module-systemd-login.so
load-module module-systemd-login
.endif

### Enable positioned event sounds
load-module module-position-event-sounds

### Cork music/video streams when a phone stream is active
load-module module-role-cork

### Modules to allow autoloading of filters (such as echo cancellation)
### on demand. module-filter-heuristics tries to determine what filters
### make sense, and module-filter-apply does the heavy-lifting of
### loading modules and rerouting streams.
load-module module-filter-heuristics
load-module module-filter-apply

### Make some devices default
#set-default-sink output
#set-default-source input

### load bluetooth modules with ldac support
.ifexists module-bluetooth-policy.so
load-module module-bluetooth-policy
.endif

.ifexists module-bluetooth-discover.so
load-module module-bluetooth-discover a2dp_config="ldac_eqmid=sq"
.endif
```

Here is the relevant pulseaudio log:
```
(1433.817|   0.018) E: [pulseaudio] module-bluez5-discover.c: failed to parse module arguments.
(1433.817|   0.000) E: [pulseaudio] module.c: Failed to load module "module-bluez5-discover" (argument: "a2dp_config=ldac_eqmid=sq"): initialization failed.
```