EHfive/pulseaudio-modules-bt #19: Trying to package for Fedora 29

gombosg, 2018-12-02:
Hi, I'd like to make a COPR repo for this in Fedora since aptX is working great when building with your instructions.
I only want to upload the `pulseaudio-module-bluetooth` to the repo and not the whole PA to avoid issues - PA is an important package.
I tried the patch the [current PA source SRPM package](https://koji.fedoraproject.org/koji/buildinfo?buildID=1127890) with your patches for the best system compatibility.

I applied them in order:
```
%patch901 -p1 -b _v1_1.a2dp
%patch902 -p1 -b _v1_2.a2dp_property
%patch903 -p1 -b _v1_3.ldac_header
%patch905 -p1 -b _v1_5.fix_ldac_frequencies
%patch906 -p1 -b _v2.1_1.aptx_aptx_hd_fix
%patch907 -p1 -F2 -b _v2.1_2.buildfile
```
They applied fine, but build fails with
```
modules/bluetooth/a2dp/a2dp_aptx.c:5:10: fatal error: libavcodec/avcodec.h: No such file or directory
 #include <libavcodec/avcodec.h>
```

It finds libavcodec during the configure phase:
```
checking for FF_AVCODEC... yes
checking for FF_AVUTIL... yes
```
And yet it fails.
I know I should contact the packager maybe but maybe you know why this fails or at least point me to the right direction why make doesn't find this folder (e.g. by adding the correct `-I` parameter).

Here's the complete build log with the error at the end: 
[build.log](https://github.com/EHfive/pulseaudio-modules-bt/files/2636550/build.log)

Here's my modified RPM spec file:
[pulseaudio.spec.txt](https://github.com/EHfive/pulseaudio-modules-bt/files/2636551/pulseaudio.spec.txt)
At line 335 you can see the configure parameters, maybe those are the culprit.

It would be my first package ever so I don't have much experience so I appreciate the help.

EHfive, 2018-12-02:
I forgot adding -I <path to  ff headers directory>.

EHfive, 2018-12-02:
Please modify buildfile_fix.patch to
```
+libbluez5_util_la_CFLAGS = $(AM_CFLAGS) $(SBC_CFLAGS) $(DBUS_CFLAGS) $(FF_AVCODEC_CFLAGS) $(FF_AVUTIL_CFLAGS)
```
Try again.

gombosg, 2018-12-02:
Ha! This works. I'm letting you know that `volume-test.c` fails at
```
    fail_unless(mdn <= 253);
```
because it's 283. This test passes without your patches. I don't know whether this is an issue.

EHfive, 2018-12-02:
> Ha! This works. I'm letting you know that `volume-test.c` fails at
> 
> ```
>     fail_unless(mdn <= 253);
> ```
> 
> because it's 283. This test passes without your patches. I don't know whether this is an issue.

Passed on PA master branch.
Failed on PA v12.2.
Both without patching.

I think it's not our issue.

closed 2018-12-02 by EHfive

G-Ray, 2018-12-03:
@gombosg Any progress ? I may help you to setup build on copr

gombosg, 2018-12-03:
Yep, it's here: https://src.fedoraproject.org/fork/gombosg/rpms/pulseaudio
Now, I'm not sure whether I should bump the version so first I just bumped the release. Now it's waiting for the COPR build at https://copr.fedorainfracloud.org/coprs/gombosg/pulseaudio-modules-bt-aptx/

gombosg, 2018-12-04:
@G-Ray let me know if you tested it; also let me know if you know a better way to build & package it. Problem is that if I bump the version, I need to provide all pulseaudio packages for both i686 and x86_64 (because of the bumped dependencies). And I don't want to replace core PA packages for anyone from this COPR repo.

It's a COPR repo so this manual method is IMHO better and if the original source repo is updated, the COPR can be easily upgraded, too.

If it works for you, too, we can add the repo to the wiki. :)

commit 43c8f526 (2018-12-02): [Makefile.am] Add missing $(FF_AVCODEC_CFLAGS) $(FF_AVUTIL_CFLAGS) to libbluez5_util_la_CFLAGS

Close #19