EHfive/pulseaudio-modules-bt #20: Add Fedora COPR repo to wiki

gombosg, 2018-12-06:
Hi, I created this COPR (user) repo for the package:
https://copr.fedorainfracloud.org/coprs/gombosg/pulseaudio-modules-bt-aptx/

I repackaged the original Fedora Pulseaudio source with your patches. It works for me™, hopefully people can test it this way. Please add it to the wiki.

EHfive, 2018-12-06:
Add libldac please.
https://github.com/EHfive/ldacBT

EHfive, 2018-12-06:
https://github.com/EHfive/pulseaudio-modules-bt/wiki/Packages/2422a6518a0598f073c1d756794da070ae1a4469

wiki updated

next time just mail me ( eh5 at sokka.cn )

closed 2018-12-06 by EHfive

gombosg, 2018-12-18:
I updated it, please test Fedora users: https://copr.fedorainfracloud.org/coprs/gombosg/pulseaudio-modules-bt-aptx/

inffy, 2019-01-22:
@gombosg seems this copr repo has been removed. It worked fine this morning when I installed. Did something happen?

inffy, 2019-01-23:
I guess RPMFusion would be the best bet, or some other repo. I have no clue how to submit packages/get packages to rpmfusion though or if there would be other "public" repos that would take it.

Shame really, just got my F29 set up the way i like it with this, never could get this to work with my manjaro install.

inffy, 2019-01-23:
@gombosg i found this from rpmfusion: https://rpmfusion.org/Contributors#Submitting_a_new_package

gombosg, 2019-01-23:
Thanks!
In the meantime, here's the F29 RPM file.
[pulseaudio-module-bluetooth-aptx-12.2-1.fc29.x86_64.zip](https://github.com/EHfive/pulseaudio-modules-bt/files/2788469/pulseaudio-module-bluetooth-aptx-12.2-1.fc29.x86_64.zip)

gombosg, 2019-01-24:
@EHfive What is the status for merging into upstream? If not expected, I'll post to RPMFusion since getting kicked out from the Fedora COPR repo.

inffy, 2019-01-24:
@gombosg thanks for the rpm, but it can't be installed because no libldac can be found (as its not in any repo)

EHfive, 2019-01-25:
> 
> 
> @EHfive What is the status for merging into upstream? If not expected, I'll post to RPMFusion since getting kicked out from the Fedora COPR repo.

Currently it won't be merged.

gombosg, 2019-01-25:
@inffy I added it... seems that going to RPMFusion is the way currently. I'll look into it in the upcoming days.

mumrau, 2019-02-07:
Maybe we could re-open the issue until a user-friendly (i.e repository) solution is found and added into the wiki? So people can subscribe to it and get a decent overview on the RPM situation.

inffy, 2019-02-07:
No need. It's not hard to install two rpm files.

mumrau, 2019-02-07:
But they won't be kept up to date eventually.
Plus for some users, yes it is, you even need to search the issue, the doc doesn't state it (probably because it isn't the intended way to serve the packages in the (near) future).

gombosg, 2019-02-07:
Hi guys, I have submitted libldac to [Fedora](https://bugzilla.redhat.com/show_bug.cgi?id=1671064) and the aptx module to [RPMFusion](https://bugzilla.rpmfusion.org/show_bug.cgi?id=5155). They will find their home there... review just takes time. :)

In the meantime you can indeed install the [RPMs](https://github.com/EHfive/pulseaudio-modules-bt/issues/20#issuecomment-456897792) directly.

davidlt, 2019-02-11:
Just discovered this package a day ago. I am happy to help with any SPEC/RPM related questions/issues. Looks like libldac is stuck on `FE-Legal`, but comments are positive here. 

There is a comment regarding "fdk-aac" package, which basically means that this targets nonfree repo in RPMFusion.

Nice work!

gombosg, 2019-02-11:
@davidlt Thanks, the package would be approved soon, they just need to lift the FE-LEGAL bug.
Are you a Fedora packager yourself? I could use some guidance getting into reviewing packages. :)

davidlt, 2019-02-12:
@gombosg I am not a packager, but I maintain RISC-V 64-bit Fedora port thus I can help with technical bits.

Btw, people are writing about the project and your work:
- https://eischmann.wordpress.com/2019/02/11/better-bluetooth-sound-quality-on-linux/
- https://www.phoronix.com/scan.php?page=news_item&px=Better-Bluetooth-Linux-Audio

Sesivany, 2019-02-12:
> Hi guys, I have submitted libldac to [Fedora](https://bugzilla.redhat.com/show_bug.cgi?id=1671064) and the aptx module to [RPMFusion](https://bugzilla.rpmfusion.org/show_bug.cgi?id=5155). They will [...]

Can't it be package the way that it could get into Fedora and at least enable LDAC there and then it could be extended by some package from RPMFusion with support for AptX, AptX-HD etc.? It's a pity to rely on RPMFusion even for LDAC if it's accepted in Fedora.

gombosg, 2019-02-12:
@Sesivany Would be great but according to the Fedora legal & packaging guidelines:

- [Patent-encumbered](https://fedoraproject.org/wiki/Forbidden_items) stuff is not allowed (ffmpeg)
- All build dependencies must be within the Fedora repos

I'm just the packager, @EHfive can tell whether it's possible to create or build the package without ffmpeg and AAC present, with only LDAC. (I guess that version won't have aptX and AAC support even if the codec packages are installed.)

It's not impossible but not sure either. Essentially it wouldn't be a new package then, just a patch to be applied against the Fedora version of PA and a PR could be submitted [here](https://src.fedoraproject.org/rpms/pulseaudio).

In RPMFusion we already [started the discussion](https://bugzilla.rpmfusion.org/show_bug.cgi?id=5155) and it will simply be a `pulseaudio-module-bluetooth-nonfree` package replacing the original, that would contain all codecs.

EHfive, 2019-02-13:
> I'm just the packager, @EHfive can tell whether it's possible to create or build the package without ffmpeg and AAC present, with only LDAC. (I guess that version won't have aptX and AAC support even if the codec packages are installed.)

Right. In the next release, I  would probably make FDK-AAC/ffmpeg/LDAC be optional at building stage.

Sesivany, 2019-02-13:
That would really help because then we could have it at least with LDAC support in Fedora/RHEL (and other patent-concerned distributions) by default although some claim there is also a problem with LGPL 2.1->Apache License compatibility. But this project has had 3 contributors so far, so it shouldn't be such a problem to grant an exception if necessary.

lukehutch, 2019-02-14:
The soname version number for fdk-aac in rpmfusion is 2, not 1:

```
[root@localhost ~]# rpm -Uhv /tmp/libldacbt-1.1-1.fc29.x86_64.rpm /tmp/pulseaudio-module-bluetooth-aptx-12.2-1.fc29.x86_64.rpm 
error: Failed dependencies:
	libfdk-aac.so.1()(64bit) is needed by pulseaudio-module-bluetooth-aptx-12.2-1.fc29.x86_64
[root@localhost ~]# rpm -ql fdk-aac | grep libfdk-aac
/usr/lib64/libfdk-aac.so.2
/usr/lib64/libfdk-aac.so.2.0.0
```

EHfive, 2019-02-28:
Actually, you can use [fdk-aac-free](https://cgit.freedesktop.org/~wtay/fdk-aac/log/?h=fedora) licensed under [FDK-AAC](https://fedoraproject.org/wiki/Licensing/FDK-AAC)
fdk-aac-free don't support AAC-HE, but AAC format in A2DP transmission is AAC-LC.
fdk-aac-free and fdk-aac(non-free) share most ABI in AAC-LC case, so I don't need to change anything in pulseaudio-modules-bt to support fdk-aac-free.

Sesivany, 2019-03-07:
Just reporting: the packages provided here don't seem to survive upgrade to Fedora 30. They stay installed, but I'm back from LDAC to SBC after the upgrade.

gombosg, 2019-03-07:
Hopefully the [RPMFusion package](https://bugzilla.rpmfusion.org/show_bug.cgi?id=5155) gets reviewed soon and then we should be good...

gombosg, 2019-03-13:
The review process is crazy slow, so I'm giving you some new RPMs in the meantime. Sorry. Been waiting for a review for months now.
The below zip contains F29 and rawhide RPMs.
[pulseaudio-modules-bt-fedora.tar.gz](https://github.com/EHfive/pulseaudio-modules-bt/files/2962304/pulseaudio-modules-bt-fedora.tar.gz)

makruiten, 2019-03-14:
Thanks! It doesn't replace the existing packages from the COPR repo. I gues I can remove them with `sudo rpm -e --nodeps libldacbt pulseaudio-module-bluetooth-aptx`? Removing them with dnf also removes a whole lot of bogus dependencies.

makruiten, 2019-03-14:
Something else: the packages you just provided don't support AAC, is that right?

[image]

gombosg, 2019-03-14:
Yes, you can remove the old COPR stuff, the repo has been disabled for quite some time now.

I checked, it's because `fdk-aac-devel` was not installed (I specified `fdk-aac`, which was a mistake). Corrected it and updated the ZIP. @makruiten please tell me if it works. It doesn't build on Rawhide (fdk-aac 2.0) though so I need to raise an issue for that.

makruiten, 2019-03-14:
It works now, thanks.

Yes, I knew it was disabled and that those could be removed, it was more about the method of removing, since dnf wanted to remove a lot of dependent packages. For people browsing this issue: use my command from https://github.com/EHfive/pulseaudio-modules-bt/issues/20#issuecomment-472826703 instead of dnf to remove the old COPR packages.

gombosg, 2019-03-14:
#43 is the 2.0 `fdk-aac` build issue.

The RPMFusion package got approved by a reviewer, then another person jumped in that "it shouldn't be approved like this", so I'm confused (and a bit upset) now. We'll see about that.

yassineim, 2019-03-14:
Fuck that bureaucratic dude, whoever he is

gombosg, 2019-03-14:
He sponsored me as a packager in Fedora, so I wouldn't say that. :)
I understand his stance, I'm just upset because he didn't take the review (the package has been sitting there for some time), and after someone reviewed and approved, he is making a [rather vague comment](https://bugzilla.rpmfusion.org/show_bug.cgi?id=5155#c15) about what should or shouldn't be done about the package.

davidlt, 2019-03-15:
On Fedora 29:
```
% sudo dnf install pulseaudio-module-bluetooth-nonfree-1.1-3.fc29.x86_64.rpm
Last metadata expiration check: 0:31:33 ago on Fri 15 Mar 2019 09:36:11 CET
Error: 
 Problem: problem with installed package pulseaudio-module-bluetooth-12.2-1.fc29.x86_64
  - package pulseaudio-module-bluetooth-nonfree-1.1-3.fc29.x86_64 conflicts with pulseaudio-module-bluetooth <= 12.2-1.fc29 provided by pulseaudio-module-bluetooth-12.2-1.fc29.x86_64
  - conflicting requests
(try to add '--allowerasing' to command line to replace conflicting packages or '--skip-broken' to skip uninstallable packages)
```

yassineim, 2019-03-15:
Add --alow-erasing like the output says

Sesivany, 2019-03-20:
> He sponsored me as a packager in Fedora, so I wouldn't say that. :)
> I understand his stance, I'm just upset because he didn't take the review (the package has been sitting there for some time), and [...]

If the package doesn't hard-depend on ffmpeg and doesn't contain any codecs why can't it be directly in Fedora? I'm not involved in RPMFusion, but I can certainly help with Fedora (I'm a packager, know package sponsors etc.).

Sesivany, 2019-03-20:
BTW someone is trying to implement the same directly in the PulseAudio project and is concerned about PA-libldac license compatibility: https://eischmann.wordpress.com/2019/02/11/better-bluetooth-sound-quality-on-linux/#comment-10714
It may have the same problem as fdk-aac-free discussed in the RPMFusion review.

gombosg, 2019-03-20:
@Sesivany : it has to be built with these nonfree libraries present and that counts as a dependency. I became a Fedora packager, too because of this. :slightly_smiling_face: See [this](https://docs.fedoraproject.org/en-US/packaging-guidelines/#_package_dependencies):

> All package dependencies (build-time or runtime, regular, weak or otherwise) MUST ALWAYS be satisfiable within the official Fedora repositories.

So this has to go to RPMFusion.

I [requested](https://bugzilla.rpmfusion.org/show_bug.cgi?id=5155) the package in RPMFusion, it was approved by a reviewer, but another reviewer disapproved (?) - it's technically approved (the `fedora-review +` flag is there) but they don't give me the repo to upload the package to, and don't give me any exact rejection reasons or guidelines. 

So I'm pretty much stuck at this point despite putting so much work into the project. Poor thing.

The conversation you linked is interesting. Yes, Pali Rohár's patches are on the way to upstream PA but we don't know how much time it takes to integrate them. There was indeed discussion about @EHfive 's patches on the PA mailing list, but in the end they weren't integrated due to multiple reasons.

There are a lot of licensing issues, yes.

- Libldac is ApacheV2
- ffmpeg is LGPL 2.1
- FDK-AAC has its own license which [can be free](https://bugzilla.redhat.com/show_bug.cgi?id=1501522), but is [GPL-incompatible](https://www.gnu.org/licenses/license-list.html#fdk)
- Upstream PA is LGPL 2.1
- This project is now GPLv3 which is [compatible](https://fedoraproject.org/wiki/Licensing:Main?rd=Licensing#SoftwareLicenses) with ApacheV2 and LGPL2.1, and with extra distribution permission, with FDK-AAC-free.

Problem is that while @EHfive can relicense this project under GPLv3, the Pulseaudio project may not want to do the same.

Sesivany, 2019-03-20:
> All package dependencies (build-time or runtime, regular, weak or otherwise) MUST ALWAYS be satisfiable within the official Fedora repositories.

The trick is that it doesn't have to be a package dependency. Of course, if the software doesn't build or run without it it's a problem, but if it's done the way that it runs well and only doesn't provide certain codecs if ffmpeg is not present in the system, it's fine.
E.g. Firefox relies on ffmpeg for multimedia support, yet it's part of Fedora. If ffmpeg is not present, it simply doesn't playback e.g. H264 videos, that's it.

gombosg, 2019-03-20:
AFAIK in C you can't link against a library without its devel package being present. Looking at the Firefox [specfile](https://src.fedoraproject.org/rpms/firefox/blob/master/f/firefox.spec), `libffi` might be used for this purpose?

pali, 2019-03-20:
It does not matter if you **link** to library or not. Important is if you **use** that library (either statically bundled or dynamically loaded, or whatever...) and **how** you call functions of that library. GPL license does not allows you to call GPL-incompatible functions and pass memory from GPL software.

So trying to figure out some tricks how to load GPL-incompatible library at runtime via plugins or whatever would lead just to GPL-violations.

See:
https://www.gnu.org/licenses/gpl-faq.html#GPLStaticVsDynamic
https://www.gnu.org/licenses/gpl-faq.html#GPLPlugins

gombosg, 2019-04-03:
Okay, the package has been accepted **without AAC support**. Sorry guys but the FDK-AAC license is really messy. It currently supports LDAC and aptX. You have to build your own for AAC or wait for ffmpeg AAC support.
**Testing is appreciated!**
The package should be in the updates-testing repo of RPM Fusion or you can download it [here](http://koji.rpmfusion.org/koji/packageinfo?packageID=588).

makruiten, 2019-04-04:
Congrats! My headphones are SBC/AAC only, so no feedback from my side. Hopefully AAC support will be added somewhere down the road in one way or another.

pali, 2019-04-04:
@makruiten look, authors of FDK-AAC library decided that they do not want their library to be used in pulseaudio and also any other GPL software. So this is not problem of pulseaudio, but in authors of FDK-AAC library licensing options. They decided to not allow such thing and you need to respect their will. It is same like if you create something on your own and do not want that somebody else take it from you and would do something with that your own thing against you.

makruiten, 2019-04-04:
I'm not suggesting to do anything against anyone's will :) Maybe down the road there will be an alternative library, some sort of agreement, or whatever the solution might be. Just hoping.

vlastoun, 2019-04-04:
Hello @gombosg  thanks for creating rpm package.
I am not able to install it dirrectly from from repository instead I downloaded that rpm package locally.
I have problem with switching to LDAC codec. I used your rpm package before and everything worked.
```
Sink #2
	State: SUSPENDED
	Name: bluez_sink.CC_98_8B_4A_C1_1B.a2dp_sink
	Description: WH-1000XM3
	Driver: module-bluez5-device.c
	Sample Specification: s16le 2ch 44100Hz
	Channel Map: front-left,front-right
	Owner Module: 26
	Mute: no
	Volume: front-left: 57672 /  88% / -3,33 dB,   front-right: 57672 /  88% / -3,33 dB
	        balance 0,00
	Base Volume: 65536 / 100% / 0,00 dB
	Monitor Source: bluez_sink.CC_98_8B_4A_C1_1B.a2dp_sink.monitor
	Latency: 0 usec, configured 0 usec
	Flags: HARDWARE DECIBEL_VOLUME LATENCY 
	Properties:
		bluetooth.protocol = "a2dp_sink"
		bluetooth.a2dp_codec = "SBC"
		device.description = "WH-1000XM3"
		device.string = "CC:98:8B:4A:C1:1B"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headset"
		bluez.path = "/org/bluez/hci0/dev_CC_98_8B_4A_C1_1B"
		bluez.class = "0x240404"
		bluez.alias = "WH-1000XM3"
		device.icon_name = "audio-headset-bluetooth"
		device.intended_roles = "phone"
	Ports:
		headset-output: Headset (priority: 0)
	Active Port: headset-output
	Formats:
		pcm
```
Before I had bluetooth.a2dp_codec = "LDAC"

I am using fedora 29 with kernel 5 and i have installed libldac.

gombosg, 2019-04-04:
@vlastoun : Thanks for the test results! 
Is `libldac` installed? (It is a recommended dependency but I can still switch to required if needed) Did you try selecting the codec in the GNOME control panel?
The package was built with LDAC support enabled for sure, and the BT module itself may be a newer version that what you had tried before.
The best would be if you could build it on your own then compare the results - this is to learn whether it's a packaging or a module issue.

vlastoun, 2019-04-04:
`libldac` is installed.
I tried switching from gnome panel. It changes when I switch from low quality plugin to SBC. Switching to LDAC does not do anything. I hope that package is OK. I will try to build it.

gombosg, 2019-04-04:
@vlastoun also looking at `journalctl` helps after trying to connect/switch codecs. Let's see if there's some strange output or error messages.

vlastoun, 2019-04-07:
From journalctl
```
dub 07 15:29:53 ntbvsa rtkit-daemon[1011]: Successfully made thread 2326 of process 2326 (/usr/bin/pulseaudio) owned by '1152' high priority at nice level -11.
dub 07 15:29:53 ntbvsa pulseaudio[2326]: E: [pulseaudio] a2dp_aptx.c: Cannot open libavcodec library: libavcodec.so. libavcodec.so: cannot open shared object file: No such file or directory
dub 07 15:29:53 ntbvsa pulseaudio[2326]: E: [pulseaudio] a2dp_aptx.c: Cannot open libavcodec library: libavcodec.so. libavcodec.so: cannot open shared object file: No such file or directory
dub 07 15:29:53 ntbvsa pulseaudio[2326]: E: [pulseaudio] a2dp_ldac.c: Cannot open LDAC encoder library: libldacBT_enc.so. libldacBT_enc.so: cannot open shared object file: No such file or directory
dub 07 15:29:53 ntbvsa pulseaudio[2326]: E: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files
dub 07 15:29:53 ntbvsa rtkit-daemon[1011]: Successfully made thread 2377 of process 2326 (/usr/bin/pulseaudio) owned by '1152' RT at priority 5.
dub 07 15:29:53 ntbvsa rtkit-daemon[1011]: Successfully made thread 2385 of process 2326 (/usr/bin/pulseaudio) owned by '1152' RT at priority 5.
dub 07 15:30:24 ntbvsa rtkit-daemon[1011]: Successfully made thread 3553 of process 2326 (/usr/bin/pulseaudio) owned by '1152' RT at priority 5.
dub 07 15:31:41 ntbvsa pulseaudio[2326]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_ldac: Unavailable profile
dub 07 15:33:12 ntbvsa pulseaudio[2326]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_ldac: Unavailable profile
```

vlastoun, 2019-04-07:
@gombosg i solved my problem... i added link `ln -s libldacBT_enc.so.2.0.2.2 libldacBT_enc.so` and now it works as before...
```
~ » pactl list | grep a2dp_codec                                                                                                 vsadilek@ntbvsa
		bluetooth.a2dp_codec = "LDAC"
```

gombosg, 2019-04-07:
@vlastoun : thanks for the test, now we see what's happening.

It's a libldac packaging error, the  `libldacBT_(enc|abr).so` symlink was missing indeed.
But in the long term, maybe using the soname is better, @EHfive , I can correct libldac package, but can you make that this package also finds `libldacBT_(enc|abr).so.2`?

EHfive, 2019-04-08:
> @vlastoun : thanks for the test, now we see what's happening.
> 
> The `libldacBT_(enc|abr).so` symlink was missing indeed, it's in the devel package.
> @EHfive , can you make that this package also finds `libldacBT_(enc|abr).so.2`?
> 
> EDIT: I made this because the [Fedora guidelines](https://docs.fedoraproject.org/en-US/packaging-guidelines/#_devel_packages) tell to move unversioned libraries (which is a symlink in this case) to the devel package.
> 
> > The versioned shared library files (/usr/lib/libfoo.so.3.2.0 and /usr/lib/libfoo.so.3) are necessary for users to run programs linked against libfoo, so they belong in the **base package**. The other, unversioned, shared library file (/usr/lib/libfoo.so) is only used to actually link libfoo to code being compiled, and is not necessary to be installed on a users system. This means that it belongs in a -**devel package**. Please note that in most cases, only the fully versioned shared library file (/usr/lib/libfoo.so.3.2.0) is an actual file, all of the other files are symbolic links to it. When a shared library file is only provided in an unversioned format, the packager should ask upstream to consider providing a properly versioned library file. **However, in such cases, if the shared library file is necessary for users to run programs linked against it, it must go into the base package.**
> 
> So I can move it to the base package but if @EHfive can make this to use the libldac2, then it should be OK.

Dynamic loading simply requires devel package (no soversion).

vlastoun, 2019-04-08:
@EHfive installing `libldac-devel` hellped - no symlink needed

gombosg, 2019-04-08:
@vlastoun Yes, I'll move that `libldacBT_(enc|abr).so` link from the devel to the base package. Not a nice move but doable.

devurandom, 2019-04-08:
> @vlastoun Yes, I'll move that `libldacBT_(enc|abr).so` link from the devel to the base package. Not a nice move but doable.

That is most certainly not what you want.

`DT_SONAME` (and a symlink of that name) is generally used to ensure ABI-compatibility of the library and its users. If an application is linked against a library, it is linked against a specific interface version. This interface version is part of that library's `SONAME`. Trying to run that program against a library with a different interface version (and hence different `SONAME`) is outright dangerous and may fail in unexpected ways -- the library authors specifically state that this will break, by giving their library a different `SONAME` (and thus interface version) than a previous version of their library, and usually they are the most knowledgeable in that area.

Hence when loading a library into a program (be it by adding the library to `DT_NEEDED` at compile time or using `dlopen`), one should always load the version of the library the program was compiled against, never any other version. Thus the load should happen using the full `SONAME`, not some part of it that happens to match an existing (but potentially different) library by accident. If the program happens to support multiple versions of the library, it could for example try to `dlopen` the `SONAME`s it supports (and no other names) in some defined order, until it finds a supported library.

If in this case the pulseaudio module happens to `dlopen` the library with something other than its `SONAME`, the pulseaudio module should be fixed, not another Fedora package that is not actually broken.

See-Also: https://en.wikipedia.org/wiki/Soname
See-Also: https://www.gnu.org/software/libtool/manual/libtool.html#Versioning

gombosg, 2019-04-08:
@devurandom : thanks for the explanation!

If @EHfive  is willing to fix this, I'm going to cancel the Bodhi update.

gombosg, 2019-04-12:
@vlastoun can you test [this](http://koji.rpmfusion.org/koji/packageinfo?packageID=588) please after removing the devel packages?

vlastoun, 2019-04-12:
@gombosg its working, i have instaled only libldac and the latest package of pulseaudio
```
/usr/lib64 » ll | grep libldacBT_enc.so                                                                                                                                                            vsadilek@ntbvsa
lrwxrwxrwx   1 root root    24 15. úno 15.51 libldacBT_enc.so.2 -> libldacBT_enc.so.2.0.2.2
-rwxr-xr-x   1 root root   52K 15. úno 15.51 libldacBT_enc.so.2.0.2.2
```

[24 more replies]