EHfive/pulseaudio-modules-bt #21: patches don't apply

mirlang, 2018-12-09:
could you please release just one single (and complete) patch against pulsaudio-12.2 or so?
or even better: fork the pulseaudio repo and apply your changes to an extra branch, having a own build system and overwriting files on a users system isn't a good way to go, anyway

besides from that, love your work :)

gombosg, 2018-12-10:
That would be appreciated, since packaging a single patch is easy and simple. Currently the patches only work with some extra "fuzz factor" (`-F`) added to `patch` and a single patch can help this.

EHfive, 2018-12-10:
The patches set generated from pulseaudio@cdeac178019ed3e73840702cb1223df5edd9df43.

I also delete 2 line in buildfile.patch for the compatibility.

But patches should works on PA >= 11.59 with fuzz and offset.

You should regenerate a single patch file after patching .

Then using the new patch file for packaging.