EHfive/pulseaudio-modules-bt #22: AAC is not automatically selected on my AAC/SBS headphone

makruiten, 2018-12-12:
My system if Fedora 29 with pulseaudio-modules-bt installed from the COPR repo, fdk-aac and ffmpeg installed manually. I already reloaded pulseaudio. Is there something else that I should do to get this to work?

makruiten, 2018-12-12:
To be clear: it uses the SBC codec now.

```
[12:59 martijn ~]$ pactl list sinks | grep -e bluez -e a2dp_codec
	Name: bluez_sink.28_11_A5_33_E2_C6.a2dp_sink
	Driver: module-bluez5-device.c
	Monitor Source: bluez_sink.28_11_A5_33_E2_C6.a2dp_sink.monitor
		bluetooth.a2dp_codec = "SBC"
		device.api = "bluez"
		bluez.path = "/org/bluez/hci0/dev_28_11_A5_33_E2_C6"
		bluez.class = "0x240418"
		bluez.alias = "QC35 II Martijn"
```

EHfive, 2018-12-12:
#20
https://copr.fedorainfracloud.org/coprs/gombosg/pulseaudio-modules-bt-aptx/builds/

Seems @gombosg didn't patch https://github.com/EHfive/pulseaudio-modules-bt/releases/tag/patch_v3,
so the package has no AAC support.

closed 2018-12-12 by EHfive

EHfive, 2018-12-12:
You could try https://github.com/EHfive/pulseaudio-modules-bt#general-installation

digititus, 2018-12-12:
I have similar issue with same copr repo, but it is LDAC / Apt X wich also do not work (as well as AAC).

digititus, 2018-12-12:
> You could try https://github.com/EHfive/pulseaudio-modules-bt#general-installation

I get an error:

> cmake ..
-- The C compiler identification is GNU 8.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.5.3") 
-- Checking for module 'libpulse'
--   Found libpulse, version 12.2-rebootstrapped
-- Checking for module 'dbus-1'
--   Found dbus-1, version 1.12.10
-- Checking for module 'sbc'
--   Package 'sbc', required by 'virtual:world', not found
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:436 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPkgConfig.cmake:602 (_pkg_check_modules_internal)
  CMakeLists.txt:23 (pkg_check_modules)

gombosg, 2018-12-12:
Sorry guys I can do this during the weekend. I'll update here when I added the latest patch and LDAC support!

EHfive, 2018-12-12:
> > You could try https://github.com/EHfive/pulseaudio-modules-bt#general-installation
> 
> I get an error:
> 
> > cmake ..
> > -- The C compiler identification is GNU 8.2.1
> > -- Check for working C compiler: /usr/bin/cc
> > -- Check for working C compiler: /usr/bin/cc -- works
> > -- Detecting C compiler ABI info
> > -- Detecting C compiler ABI info - done
> > -- Detecting C compile features
> > -- Detecting C compile features - done
> > -- Found PkgConfig: /usr/bin/pkg-config (found version "1.5.3")
> > -- Checking for module 'libpulse'
> > --   Found libpulse, version 12.2-rebootstrapped
> > -- Checking for module 'dbus-1'
> > --   Found dbus-1, version 1.12.10
> > -- Checking for module 'sbc'
> > --   Package 'sbc', required by 'virtual:world', not found
> > CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:436 (message):
> > A required package was not found
> > Call Stack (most recent call first):
> > /usr/share/cmake/Modules/FindPkgConfig.cmake:602 (_pkg_check_modules_internal)
> > CMakeLists.txt:23 (pkg_check_modules)
> 
> SBC is installed:
> 
> > pactl list sinks | grep -e bluez -e a2dp_codec
> > Name: bluez_sink.40_ED_98_19_86_CA.a2dp_sink
> > Driver: module-bluez5-device.c
> > Monitor Source: bluez_sink.40_ED_98_19_86_CA.a2dp_sink.monitor
> > bluetooth.a2dp_codec = "SBC"
> > device.api = "bluez"
> > bluez.path = "/org/bluez/hci0/dev_40_ED_98_19_86_CA"
> > bluez.class = "0x240418"
> > bluez.alias = "FiiO BTR3"

Maybe a development package of sbc need to be installed.
So does fdk-aac, ffmpeg(libavcodec, libavutil)

gombosg, 2018-12-18:
I updated it, please test: https://copr.fedorainfracloud.org/coprs/gombosg/pulseaudio-modules-bt-aptx/
Feedback is welcome!

digititus, 2018-12-18:
It works! Thanks

gombosg, 2018-12-18:
@makruiten can you test AAC support please? (Enable RPMFusion free & nonfree before installing)

makruiten, 2018-12-18:
I’ll test it tomorrow!

makruiten, 2018-12-19:
It works perfectly, thanks! 

```
bluetooth.a2dp_codec = "AAC"
```

I don't have any feedback right now, but I'll tell you if I do.