EHfive/pulseaudio-modules-bt #24: sony wh-1000xm2 using AptX HD instead of LDAC

WingofaGriffin, 2019-01-02:
`pactl list sinks | grep -e bluez -e a2dp_codec
	Name: bluez_sink.04_5D_4B_DE_B9_93.a2dp_sink
	Driver: module-bluez5-device.c
	Monitor Source: bluez_sink.04_5D_4B_DE_B9_93.a2dp_sink.monitor
		bluetooth.a2dp_codec = "APTX_HD"
		device.api = "bluez"
		bluez.path = "/org/bluez/hci0/dev_04_5D_4B_DE_B9_93"
		bluez.class = "0x240404"
		bluez.alias = "WH-1000XM2"
`

Seems weird to me as it should automatically detect LDAC is enabled properly. I have libldac installed through the AUR, and it works on my phone.

EHfive, 2019-01-02:
Check your outputs of `ldconfig -p | grep ldac`

It should like this if your libldac installed correctly.
```
        libldacBT_enc.so (libc6,x86-64) => /usr/lib/libldacBT_enc.so
        libldacBT_abr.so (libc6,x86-64) => /usr/lib/libldacBT_abr.so
```
Then restart your pulseaudio (`pulseaudio -k`), ldac should work.

WingofaGriffin, 2019-01-02:
Hm, that fixed it. Looks like libldac was not properly installing as a dependency with yay despite having to confirm it. installing it explicitly fixed the issue. Thanks!

closed 2019-01-02 by WingofaGriffin