EHfive/pulseaudio-modules-bt #25: crackling sound / bandwith problem?

yannick, 2019-01-09:
hi! first of all thank you, its awesome to have ldac on my notebook! 

not 100% sure if this is an actual issue or my hardware is just crap ( i tried with usb bluetooth chips too though): 

especially when using ldac_eqmid=hq  i get a lot of crackling, probably a bandwith problem. 
even if i rfkill block the wifi ( Intel Dual Band Wireless-AC 8265 ).

my questions:

- is there a way to get more debug information, i.e. what quality is actually being used if `ldac_eqmid=abr` is set ?  
- is there other options to tune the bt stack e.g. adding more buffer ? (on os-x you can tune the buffers)
- do the pulseaudio settings such as `default-sample-format` `default-sample-rate` and `resample-method` influence LDAC playback ?

EHfive, 2019-01-09:
> not 100% sure if this is an actual issue or my hardware is just crap ( i tried with usb bluetooth chips too though):

It's not an issue.
LDAC HQ require 990kbps/909kbps(audio part only).
The bluetooth bandwidth depends on your radio environment;
So if your radio environment is noisy, the transmission bandwidth just can not reach what LDAC HQ required, then cracks appear.

See https://github.com/EHfive/pulseaudio-modules-bt/wiki#bitrate-of-ldac.
 

> my questions:
> * is there a way to get more debug information, i.e. what quality is actually being used if `ldac_eqmid=abr` is set ?

No in modules. But you can modify ldac abr 's source and append a specific definition to cflags to achieve it.

> * is there other options to tune the bt stack e.g. adding more buffer ? (on os-x you can tune the buffers)

Current, no.

> * do the pulseaudio settings such as `default-sample-format` `default-sample-rate` and `re

Question is no sense. The encoding process would not cause lagging.

Close this issue because i have nothing to do.

closed 2019-01-09 by EHfive

smirnov-vs, 2019-01-15:
Hi. I have same band and aptX headset.
bt_coex_active=0 driver option helps a bit.
And if wifi turned off there is almost no crackling.