EHfive/pulseaudio-modules-bt #26: Sound artifacts/cracking with aptX after some time after installation

pszkotak, 2019-01-17:
Hello,
Thank you for your great work, really appreciate it.
My setup is Dell E7440 with Intel 7260, Kubuntu 18.04, Sennheiser M2 AEBT with aptX.
The modules were installed from the deb package.

At the beginning everything was great, I had no issues.
Some weeks later heavy cracking while listening started.
Calling `pulseaudio -k` fixes it, but I think it also messes up with the sound quality. Is it possible that after the aptX is no longer used?

```
piter@piter:~$ pactl list sinks | grep -e bluez -e a2dp_codec
	Name: bluez_sink.00_1B_66_84_1D_AF.a2dp_sink
	Driver: module-bluez5-device.c
	Monitor Source: bluez_sink.00_1B_66_84_1D_AF.a2dp_sink.monitor
		bluetooth.a2dp_codec = "aptX"
		device.api = "bluez"
		bluez.path = "/org/bluez/hci0/dev_00_1B_66_84_1D_AF"
		bluez.class = "0x240404"
		bluez.alias = "MOMENTUM M2 AEBT"
piter@piter:~$ pulseaudio -k
piter@piter:~$ pactl list sinks | grep -e bluez -e a2dp_codec
	Name: bluez_sink.00_1B_66_84_1D_AF.headset_head_unit
	Driver: module-bluez5-device.c
	Monitor Source: bluez_sink.00_1B_66_84_1D_AF.headset_head_unit.monitor
		device.api = "bluez"
		bluez.path = "/org/bluez/hci0/dev_00_1B_66_84_1D_AF"
		bluez.class = "0x240404"
		bluez.alias = "MOMENTUM M2 AEBT"
```

At the first glance it may seem like a bandwidth or link quality issue, but there are some points that make me think this is not the case:
1. Everything is fine when I use the mobile phone (also with aptX) in the same environment, I can even be at a significant distance from the phone and everything is fine.
2. Headphones are really close to the PC.
3. Nothing in the environment changed, at the beginning it was fine.

After the issue has been observed I tried things like:
1. Installing the package again, 
2. Compiling it from the source. 
3. Setting an option `MultiProfile = multiple` in `/etc/bluetooth/main.conf` seemed to work for a day, but the issue reoccurred.
4. Manually switching output source back and forth sometimes used to fix it for a short time, but now it does not do the job.
5.  After `pulseaudio -k` there are no cracks, but executing it every headphones connection is a pain and I think that sound quality decreases as well.

As I have no idea what additional information I could share please find some random logs below ;)

```
piter@piter:~$ uname -a
Linux piter 4.19.8-041908-generic #201812080831 SMP Sat Dec 8 13:34:18 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
piter@piter:~$ blue
bluedevil-sendfile  bluemoon            bluetoothctl
bluedevil-wizard    bluetooth           bluetoothd
piter@piter:~$ blue
bluedevil-sendfile  bluemoon            bluetoothctl
bluedevil-wizard    bluetooth           bluetoothd
piter@piter:~$ bluetoothctl -v
bluetoothctl: 5.48
piter@piter:~$ pulseaudio -v
I: [pulseaudio] main.c: setrlimit(RLIMIT_NICE, (31, 31)) failed: Operation not permitted
I: [pulseaudio] main.c: setrlimit(RLIMIT_RTPRIO, (9, 9)) failed: Operation not permitted
I: [pulseaudio] core-util.c: Successfully gained nice level -11.
I: [pulseaudio] main.c: This is PulseAudio 12.2
I: [pulseaudio] main.c: Page size is 4096 bytes
I: [pulseaudio] main.c: Machine ID is 4cb345eb2c51419bad365be3c01b80e8.
I: [pulseaudio] main.c: Session ID is 3.
I: [pulseaudio] main.c: Using runtime directory /run/user/1000/pulse.
I: [pulseaudio] main.c: Using state directory /home/piter/.config/pulse.
I: [pulseaudio] main.c: Using modules directory /usr/lib/pulse-12.2/modules.
I: [pulseaudio] main.c: Running in system mode: no
E: [pulseaudio] pid.c: Daemon already running.
E: [pulseaudio] main.c: pa_pid_file_create() failed.
```

I messed a lot with the bt/modules maybe I should clean the BT and pulseaudio modules installation?
How can I do it to be sure that the setup is as good as it can be?

Any hints are very welcome!

Kind regards,
Piotr

EHfive, 2019-01-18:
Try https://github.com/EHfive/pulseaudio-modules-bt/issues/25#issuecomment-454213128
https://wiki.archlinux.org/index.php/Wireless_network_configuration#Bluetooth_coexistence

WingofaGriffin, 2019-01-19:
I had this issue with LDAC and the fix seems to work. there's a small crack at the start of a new song, but it is not super noticeable at all compared to what it was before the fix. This issue did not happen without the fix on vanilla pulseaudio-bluetooth, so I think there is a bug with this fork

pszkotak, 2019-01-25:
@EHfive @WingofaGriffin Thank you for responses.
Unfortunately this is not the case, with the coex disabled I hear even more cracks than before.

```
piter@piter:~$ cat  /sys/module/iwlwifi/parameters/bt_coex_active
N
piter@piter:~$ pactl list sinks | grep -e bluez -e a2dp_codec
	Name: bluez_sink.00_1B_66_84_1D_AF.a2dp_sink
	Driver: module-bluez5-device.c
	Monitor Source: bluez_sink.00_1B_66_84_1D_AF.a2dp_sink.monitor
		bluetooth.a2dp_codec = "aptX"
		device.api = "bluez"
		bluez.path = "/org/bluez/hci0/dev_00_1B_66_84_1D_AF"
		bluez.class = "0x240404"
		bluez.alias = "MOMENTUM M2 AEBT"
```

I have even tried to disable the Wi-Fi at all as I use mostly wired connection. That did not change anything.

smirnov-vs, 2019-01-28:
It seems like intel driver bug.
I've installed this drivers for pulseaudio because after sometime sound was getting worse in my OnePlus Bullets. Only `pulseaudio -k` helped. Now I have cracklings, and it seems only intel chips are affected. Correct me, if anyone has another chip.

Another thing is that a lot of cracks happening when I listen music at work - openspace place.
So there can be an electromagnetic interference. 

Also sometimes linux for some reason uses wifi connection if wired connection is established, so I had to turn my WiFi off because of network speed and bluetooth problems.
Unfortunately I can't compare quality of Windows driver in same conditions because I need Linux, but Intel states that my 8265 model supports aptX.

So there is no solution now, but when I disable WiFi and set bt_coex_active=0 I experienced much less cracks. Also I've set latency of output to 100ms.

Maybe somebody can explain or give a link to some theory base of these bluetooth drivers. Maybe we can deal with this problem

smirnov-vs, 2019-03-06:
Updated to Linux 5.0, no crackling for 2 days. Maybe new kernel drivers fixed this.

pszkotak, 2019-03-07:
Updated to the kernel 5.0 as to test if it solves the issue in my case.
It did not.

I performed one more test - I opened the laptop lid (normally it is closed as laptop is connected to the dock). It seems that it has fixed the issue. I hear occasional cracks, but it may be connected with the noisy environment.

Maybe my mobile phone simply has much higher radio output power and this is why I do not observe the issue in that case. 

It seems that the issue was noisy environment and weak radio.
All that is left in my case to test with a BT dongle and during the time at most 2.4GHz Wi-Fi networks are not in heavy use.

dejanr, 2019-03-15:
@pszkotak what dongle are you using, i am experiencing the same issues?

Linux-Fan, 2019-06-20:
I experience the same issue - sound glitches/cracks over my Sennheiser HD4.50 BTNC. My rig - MSI GT75 Titan 8RG with:
1. Processor: [code]achilleus@Linux:~$ cat /proc/cpuinfo | grep 'model name' | uniq
model name      : Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz[/code]
2. Sound: [code]achilleus@Linux:~$ cat /proc/asound/cards
 0 [PCH            ]: HDA-Intel - HDA Intel PCH
                      HDA Intel PCH at 0x9a210000 irq 172[/code]
3. WiFi: [code]achilleus@Linux:~$ lspci -vv -s 04:00.0
04:00.0 Network controller: Intel Corporation Wireless-AC 9260 (rev 29)
        Subsystem: Bigfoot Networks, Inc. Wireless-AC 9260
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 18
        Region 0: Memory at 9cc00000 (64-bit, non-prefetchable) [size=16K]
        Capabilities: <access denied>
        Kernel driver in use: iwlwifi
        Kernel modules: iwlwifi[/code]

I use WiFi 5Ghz WLAN, so there shouldn't be any interference with my WiFi (card: Killer Wireless-AC 1550).
Not sure if it's relevant, but MSI states, that my laptop has ESS SABRE HIFI AUDIO DAC.
Everything works fine on Windows 10 on the same system and I've tried plenty of tips from the Internet, among others: editing some options in /etc/pulse/daemon.conf, adding "options ath9k btcoex_enable = 1" to the /etc/modprobe.d/ath9k.conf (or /etc/modprobe.d/bluetooth-audio.conf), creating a file named /etc/modprobe.d/bluetooth-audio.conf with the following content: "options ath9k btcoex_enable=1 bt_ant_diversity=1" and other, but none seem to work.

Can anyone give me a tip or is it a bug?

WingofaGriffin, 2019-11-19:
I'd put in a note somewhere that because of commit 683e89f making AAC constant bitrate, it is encouraged to turn on afterburner and set bitrate mode to 5 if you have a modern computer. Especially with Airpods and Galaxy Buds being so popular these days that only support AAC (and samsung scalable codec which I do not believe is open source), this will solve static issues for many users (me included)

closed 2020-12-20 by EHfive