EHfive/pulseaudio-modules-bt #28: COPR repo seems to be down or unmaintained

martijnkruiten, 2019-01-29:
The COPR repo (https://copr.fedorainfracloud.org/coprs/gombosg/pulseaudio-modules-bt-aptx/) doesn't have any packages anymore. @gombosg, what is the status? Is it not maintained anymore?

```
martijn ~ $ sudo dnf update --refresh
Failed to synchronize cache for repo 'gombosg-pulseaudio-modules-bt-aptx', ignoring this repo.
```

martijnkruiten, 2019-01-29:
I didn't properly read:

> This repo has been ̶c̶e̶n̶s̶o̶r̶e̶d̶ deleted by Fedora Legal due to an optional dependency on ffmpeg, which is patent encumbered. (It was not included in the final package, but used during build time for linking.)

closed 2019-01-29 by martijnkruiten

gombosg, 2019-01-30:
@EHfive  I have submitted the packages to RPMFusion, they are awaiting review. Correcting errors etc. might take some time but I think that sooner or later it will work.

[#5154](https://bugzilla.rpmfusion.org/show_bug.cgi?id=5154)
[#5155](https://bugzilla.rpmfusion.org/show_bug.cgi?id=5155)