EHfive/pulseaudio-modules-bt #31: Check if bluetooth device supports LDAC/AAC

oxwivi, 2019-02-12:
I'm on Debian testing and before I mess around trying to get Ubuntu PPA working on Debian, is there anyway to verify that my device supports those codecs in the first place?

```
$ pactl list | grep codec
$ pulseaudio --version
pulseaudio 12.2
```

There doesn't seem to be any codec entry in the output of `pactl list`. Is there any prods I can use to poke at my headset and find out what codecs it supports?

**PS** Idol animes are trash.

EHfive, 2019-02-12:
1.Install package `bluez-tools` aka `bluez-utils` which include command `avinfo`
2. Get your bluetooth device address
3. Make sure device is disconnected
4. Run `avinfo <bluetooth device address>` 
```
$ avinfo 70:26:05:00:00:00
Connecting ... 
Stream End-Point #1: Audio Sink 
        Media Codec: SBC
                Channel Modes: Mono DualChannel Stereo JointStereo
                Frequencies: 44.1Khz 48Khz 
                Subbands: 4 8
                Blocks: 4 8 12 16 
                Bitpool Range: 2-53
        Content Protection: SCMS-T
Stream End-Point #5: Audio Sink 
        Media Codec: Vendor Specific A2DP Codec
                Vendor ID 0x0000012d
                Vendor Specific Codec ID 0x00aa
                Vendor Specific Data: 0x3c 0x07
                Vendor Specific Value (LDAC)
                        Frequencies: 44.1kHz 48kHz 88.2kHz 96kHz 
                        Channel modes: Mono Dual Stereo 
        Content Protection: SCMS-T
Stream End-Point #6: Audio Sink 
        Media Codec: Vendor Specific A2DP Codec
                Vendor ID 0x000000d7
                Vendor Specific Codec ID 0x0024
                Vendor Specific Data: 0x32 0x00 0x00 0x00 0x00
                Vendor Specific Value (aptX HD)
                        Frequencies: 44.1kHz 48kHz 
                        Channel modes: Stereo 
        Content Protection: SCMS-T
Stream End-Point #3: Audio Sink 
        Media Codec: Vendor Specific A2DP Codec
                Vendor ID 0x0000004f
                Vendor Specific Codec ID 0x0001
                Vendor Specific Data: 0x32
                Vendor Specific Value (aptX)
                        Frequencies: 44.1kHz 48kHz 
                        Channel modes: Stereo 
        Content Protection: SCMS-T
Stream End-Point #2: Audio Sink 
        Media Codec: MPEG24
                Object Types: MPEG-2 AAC LC MPEG-4 AAC LC 
                Frequencies: 44.1kHz 48kHz 
                Channels: 1 2 
                Bitrate: 320000
                VBR: Yes
        Content Protection: SCMS-T

```

> **PS** Idol animes are trash.

Excuse me??? Aji people ?

zylxpl, 2019-02-12:
I installed bluez-tools, but there is no avinfo command. Ubuntu 18.04

EHfive, 2019-02-12:
> I installed bluez-tools, but there is no avinfo command. Ubuntu 18.04

That's why i wrote "aka `bluez-utils`"
>     1. Install package `bluez-tools` aka `bluez-utils` which include command `avinfo`

zylxpl, 2019-02-12:
But there is no bluez-utils either. In fact, there is no bluez package with avinfo.

EHfive, 2019-02-12:
> But there is no bluez-utils either. In fact, there is no bluez package with avinfo.

Seems those files are removed by Ubuntu packgers.

You can build bluez from source, than you find `avinfo` in `tools/` subfolder after `make`

zylxpl, 2019-02-12:
It a lot of effort just to check what codec are supported on my headphones. It is much quicker with dev options in android phone. 

Anyway, thx for your great work. My headphones sounds much better on aptX vs SBC. Hope it will be merged soon.

EHfive, 2019-02-12:
@zylxpl 
The options in Android dev mode show what codec are supported on your Android phone not your headphone

zylxpl, 2019-02-12:
Yeah, thats right, but it's posible to force use of specific codec, and when it won't work it simply mean that it is not supported by headphones.

closed 2019-02-12 by EHfive

oxwivi, 2019-02-12:
> Seems those files are removed by Ubuntu packgers.

Nah, the problems start with [Debian](https://packages.debian.org/search?suite=testing&section=all&arch=any&searchon=contents&keywords=avinfo) land:
[image]

Now to see how to get `bluez` compiled.

oxwivi, 2019-02-12:
... Good thing I didn't go forward with installing the PPA and fucking everything up before making this post...
```
$  /.#machine.4a2cc4a92cf51f35/home/$USER/bluez-4.101/tools/avinfo 93:05:01:A0:01:C0
Connecting ... 
Stream End-Point #5: Audio Sink 
        Media Codec: Unknown
Stream End-Point #1: Audio Sink 
        Media Codec: SBC
                Channel Modes: Mono DualChannel Stereo JointStereo
                Frequencies: 16Khz 32Khz 44.1Khz 48Khz 
                Subbands: 4 8
                Blocks: 4 8 12 16 
                Bitpool Range: 2-53
```

EHfive, 2019-02-12:
@oxwivi 
Seems your device also support a unknown codec, it might be aptX or one of other a2dp codecs.
You could build bluez from blueZ 5 or bluez git master to see more infomations.
In bluz 4, avinfo prints unknown in other codecs case(non SBC).

oxwivi, 2019-02-13:
Hell ye! IDK why I ignored `bluez` current release and instead looked at the older release in their download page.

```
~/bluez-5.50$  tools/avinfo 93:05:01:A0:01:C0
Connecting ... 
Stream End-Point #5: Audio Sink 
        Media Codec: Vendor Specific A2DP Codec
                Vendor ID 0x0000004f
                Vendor Specific Codec ID 0x0001
                Vendor Specific Value (aptX)
                        Frequencies: 16kHz 32kHz 44.1kHz 48kHz 
                        Channel modes: Stereo 
Stream End-Point #1: Audio Sink 
        Media Codec: SBC
                Channel Modes: Mono DualChannel Stereo JointStereo
                Frequencies: 16Khz 32Khz 44.1Khz 48Khz 
                Subbands: 4 8
                Blocks: 4 8 12 16 
                Bitpool Range: 2-53
        Content Protection: SCMS-T
```

Many thanks, amigo. Now then, should I fuck up my Debian with PPA or just compile it from source... None of the compiled files put into place by `make install` replaces existing `bluez` files from Debian packages right? And if it one day does gets mainlined in Debian (which I doubt, because patents) the `make install` files will get replaced, yeah?

EHfive, 2019-02-13:
> Many thanks, amigo. Now then, should I fuck up my Debian with PPA or just compile it from source... None of the compiled files put into place by `make install` replaces existing `bluez` files from Debian packages right? And if it one day does gets mainlined in Debian (which I doubt, because patents) the `make install` files will get replaced and not conflict, yeah?

The modules only replace the original pulseaudio bt modules.
```
$ MODDIR=`pkg-config --variable=modlibexecdir libpulse`
$ find $MODDIR -regex ".*\(bluez5\|bluetooth\).*\.so"
/usr/lib/pulse-12.2/modules/module-bluetooth-discover.so
/usr/lib/pulse-12.2/modules/module-bluetooth-policy.so
/usr/lib/pulse-12.2/modules/libbluez5-util.so
/usr/lib/pulse-12.2/modules/module-bluez5-device.so
/usr/lib/pulse-12.2/modules/module-bluez5-discover.so
```
I can't give you advise because I don't familiar with using PPA in Debian.

oxwivi, 2019-02-13:
@EHfive, I installed the package generated using @elvenfighter's script, and now I can't connect to my headset. Android phone can still connect. I just get the following on `bluetoothctl`:
```
bluetooth]# connect 93:05:01:A0:01:C0 
Attempting to connect to 93:05:01:A0:01:C0
Failed to connect: org.bluez.Error.Failed
[CHG] Device 93:05:01:A0:01:C0 Connected: yes
[CHG] Device 93:05:01:A0:01:C0 Connected: no
bluetooth]# connect 93:05:01:A0:01:C0 
Attempting to connect to 93:05:01:A0:01:C0
```

ValdikSS, 2019-06-09:
Just FYI, the following codec IDs is by far the most complete one:

```
Mandatory:
0x00 - SBC

Optional:
0x01 - MPEG-1,2 (aka MP3)
0x02 - MPEG-2,4 (aka AAC)
0x04 - ATRAC

Vendor specific:
0xFF 0x004F 0x01   - aptX
0xFF 0x00D7 0x24   - aptX HD
0xFF 0x000A 0x02   - aptX Low Latency
0xFF 0x00D7 0x02   - aptX Low Latency
0xFF 0x000A 0x01   - FastStream
0xFF 0x012D 0xAA   - LDAC
0xFF 0x0075 0x0102 - Samsung HD
0xFF 0x0075 0x0103 - Samsung Scalable Codec
0xFF 0x053A 0x484C - Savitech LHDC

0xFF 0x000A 0x0104 - The CSR True Wireless Stereo v3 Codec ID for AAC
0xFF 0x000A 0x0105 - The CSR True Wireless Stereo v3 Codec ID for MP3
0xFF 0x000A 0x0106 - The CSR True Wireless Stereo v3 Codec ID for aptX
```

Lastique, 2019-06-12:
Another way to discover which codecs are supported by your Bluetooth device is [this](https://askubuntu.com/a/425334):

- Install `bluez-hcidump`.
- Make sure your device/headset is *not* connected.
- Run `sudo hcidump avdtp`.
- Connect your device.

In the initial output of `hcidump` there will be message exchange listing the sinks supported by the device as well as capabilities of the sinks, including codecs. Then there will be a "Set config" command saying which sink and codec is actually selected.

takatakatakatak, 2019-06-30:
@Lastique thank you for this, the hcidump was able to verify that my 1000xm2 are only connecting via SBC codec even after successful setup and install of EHfive pulseuadio modules. 

Not sure where to go from here. Back to playing with /etc/pulse/default.pa to see if I can get one of the better codecs to load.

Lastique, 2019-07-01:
If you're using Ubuntu packages, make sure you installed `pulseaudio-modules-bt`, not `pulseaudio-module-bluetooth`, and rebooted.

takatakatakatak, 2019-07-01:
> If you're using Ubuntu packages, make sure you installed `pulseaudio-modules-bt`, not `pulseaudio-module-bluetooth`, and rebooted.

That did it! I couldn't work out why it didn't work on first compile given that all of the dependencies were met, but having done as you suggested and run the make file again it has picked up aptX and aptX-HD ! Sounds fantastic. Been using it near 24 hours now and absolutely zero problems.

Still not sure why LDAC wasn't picked up given libldac was available and bluez command inserted in default.pa but to be honest I'm quite happy with the quality and battery life running aptxHD! This really does deserve to find its way into all of the repos, absolutely amazing job by EHfive and no reason it shouldn't be supported out of the box. Absolute game changer. I'm not going back to windows now!!

raffaem, 2020-12-20:
> Another way to discover which codecs are supported by your Bluetooth device is [this](https://askubuntu.com/a/425334):
>
> * Install `bluez-hcidump`.
> * Make sure your device/headset is _not_ connected. [...]

Hi,

I tried your method and got:

```
$ sudo hcidump avdtp
HCI sniffer - Bluetooth packet analyzer ver 5.55
device: hci0 snap_len: 1500 filter: 0x400
< AVDTP(s): Discover cmd: transaction 5 nsp 0x00
> AVDTP(s): Discover rsp: transaction 5 nsp 0x00
    ACP SEID 1 - Audio Sink
    ACP SEID 5 - Audio Sink
    ACP SEID 6 - Audio Sink
    ACP SEID 3 - Audio Sink
    ACP SEID 2 - Audio Sink
< AVDTP(s): Set config cmd: transaction 6 nsp 0x00
    ACP SEID 5 - INT SEID 1
    Media Transport
    Media Codec - non-A2DP (LDAC)
      20 01 
> AVDTP(s): Set config rsp: transaction 6 nsp 0x00
< AVDTP(s): Open cmd: transaction 7 nsp 0x00
    ACP SEID 5
> AVDTP(s): Open rsp: transaction 7 nsp 0x00
< AVDTP(s): Start cmd: transaction 8 nsp 0x00
    ACP SEID 5
> AVDTP(s): Start rsp: transaction 8 nsp 0x00
< AVDTP(m): ver 2 cc 0 pt 96 seqn 0 time 0 ssrc 1
< AVDTP(m): ver 2 cc 0 pt 96 seqn 1 time 256 ssrc 1
```

Where can I find the codecs supported by my headset?

Lastique, 2020-12-20:
For some reason in your case there is no "All Capabilities" command and response, where the capabilities should be listed. But "Set config" says LDAC was chosen.

raffaem, 2020-12-23:
> For some reason in your case there is no "All Capabilities" command and response, where the capabilities should be listed. But "Set config" says LDAC was chosen.

Yes, LDAC it's what I had selected in blueman as my codec. So it is not showing all the capabilities but only the one I have selected.

> > But there is no bluez-utils either. In fact, there is no bluez package with avinfo.
> 
> Seems those files are removed by Ubuntu packgers.
> 
> You can build bluez from source, than you can find `avinfo` in `tools/` subfolder after `make`

I [opened a bug report](https://bugs.launchpad.net/ubuntu/+source/bluez/+bug/1907886) asking Ubuntu maintainers to include a package that mirrors bluez/tools.

In the meantime, I compiled it from source:

```
git clone https://git.kernel.org/pub/scm/bluetooth/bluez.git/
cd bluez
autoreconf -i
./configure
make
```

This is the output for my WH-1000XM3

```
./avinfo CC:98:8B:B0:C4:6C
Connecting ... 
Stream End-Point #1: Audio Sink 
        Media Codec: SBC
                Channel Modes: Mono DualChannel Stereo JointStereo
                Frequencies: 44.1Khz 48Khz 
                Subbands: 4 8
                Blocks: 4 8 12 16 
                Bitpool Range: 2-53
        Content Protection: SCMS-T
Stream End-Point #5: Audio Sink 
        Media Codec: Vendor Specific A2DP Codec
                Vendor ID 0x0000012d
                Vendor Specific Codec ID 0x00aa
                Vendor Specific Data: 0x3c 0x07
                Vendor Specific Value (LDAC)
                        Frequencies: 44.1kHz 48kHz 88.2kHz 96kHz 
                        Channel modes: Mono Dual Stereo 
        Content Protection: SCMS-T
Stream End-Point #6: Audio Sink 
        Media Codec: Vendor Specific A2DP Codec
                Vendor ID 0x000000d7
                Vendor Specific Codec ID 0x0024
                Vendor Specific Data: 0x32 0x00 0x00 0x00 0x00
                Vendor Specific Value (aptX HD)
                        Frequencies: 44.1kHz 48kHz 
                        Channel modes: Stereo 
        Content Protection: SCMS-T
Stream End-Point #3: Audio Sink 
        Media Codec: Vendor Specific A2DP Codec
                Vendor ID 0x0000004f
                Vendor Specific Codec ID 0x0001
                Vendor Specific Data: 0x32
                Vendor Specific Value (aptX)
                        Frequencies: 44.1kHz 48kHz 
                        Channel modes: Stereo 
        Content Protection: SCMS-T
Stream End-Point #2: Audio Sink 
        Media Codec: MPEG24
                Object Types: MPEG-2 AAC LC MPEG-4 AAC LC 
                Frequencies: 44.1kHz 48kHz 
                Channels: 1 2 
                Bitrate: 320000
                VBR: Yes
        Content Protection: SCMS-T
```

How can I get the address of my bluetooth device from the command line? I used blueman to do that actually.