EHfive/pulseaudio-modules-bt #32: A2DP Sink dont support microphone

brauliobo, 2019-02-12:
How can I use the microphone in a good quality as well? When I switch to A2DP Sink it doesn't show any recording device for my Mi Sports BT Earphones

EHfive, 2019-02-12:
>  When I switch to A2DP Sink it doesn't show any recording device for my Mi Sports BT Earphones

It's not a bug or something,
It's a feature.

> How can I use the microphone in a good quality as well?

You can't. The use of HSP/HSF is phone calls, not music.

brauliobo, 2019-02-12:
Can good codecs be used with HSP/HSF?

EHfive, 2019-02-12:
> Can good codecs be used with HSP/HSF?

Can't... 

https://en.wikipedia.org/wiki/List_of_Bluetooth_profiles#Advanced_Audio_Distribution_Profile_(A2DP)

brauliobo, 2019-02-12:
Do you know why Android have a better bluetooth call quality?

EHfive, 2019-02-12:
> Do you know why Android have a better bluetooth call quality?

I'm not familiar with Android 's implements. Maybe Android has a higher audio sample frequency setting in HSP/HSF and a better I/O.

closed 2019-02-12 by EHfive

venatiodecorus, 2019-02-20:
I just built this and got it installed to use my microphone but I guess I'm out of luck! Thanks for your work anyways I'm sure I'll enjoy the higher quality codecs. Is there any way to adjust HSP/HFP profile to sound better at all? Or any way to use a microphone while listening to high quality audio on Linux over Bluetooth? Or is my only option to go wired or with a different wireless protocol?

Thanks again for your work and I appreciate if you take any time to answer my questions.

gombosg, 2019-02-20:
No, HSP is always bad quality, like a phone. It's an old codec.
For my Sennheiser HD4.50BTNC, it switches to HSP when using the microphone. On Linux and on my phone, too. So I think this is true for all headsets, since they can only have 1 BT connection to the same computer (phone etc.) at a time and A2DP is [unidirectional](https://en.wikipedia.org/wiki/List_of_Bluetooth_profiles#Advanced_Audio_Distribution_Profile_(A2DP)).

brauliobo, 2019-02-20:
@gombosg android HSP quality is surely better! there must be a setting/limitation to make linux/pulseaudio as good as android

EHfive, 2019-02-20:
The sample rate of HSP on pulseaudio bluetooth modules is 8000Hz.
You could try alsa-bluez which support HSP 1.6 "wideband speech" mSBC codec at 16 kHz.

brauliobo, 2019-02-20:
@EHfive just asked them at https://github.com/Arkq/bluez-alsa/issues/187 on how to configure, have you ever done it?

dceee, 2019-02-22:
> You could try alsa-bluez which support HSP 1.6 "wideband speech" mSBC codec at 16 kHz.

This is exactly what I was missing on my Linux desktops. Is this a pulseaudio, alsa or a hardware limitation? My Sennheiser PXC-550 sound way better on my Android devices when using in headset mode (full duplex, not A2DP). I guess they are using the HSP 1.6 mSBC profile.
I tried the bluez-alsa approach. But since it states that it cannot run alongside pulseaudio, it's not a solution for me.

brauliobo, 2019-02-22:
@dceee thanks for confirming!

bhack, 2020-04-01:
Please upvote the feature at https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/254

bhack, 2020-04-01:
See also https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues/776