EHfive/pulseaudio-modules-bt #33: bluetoothd[11111]: a2dp-sink profile connect failed for XX:XX:XX:XX:XX:XX: Protocol not available

okias, 2019-02-12:
After pulseaudio-modules-bt instalation I getting this from EOZ Air bluetooth headphones.

pactl 12.2
Přeloženo s libpulse 12.2.0
Propojeno s libpulse 12.2.0

BlueZ: 5.50-r1 + some patches for a2dp codecs (without them it doesn't work either)
Linux: Gentoo ~amd64

Any ideas?
Thanks

okias, 2019-02-12:
kinda looks like that bluez missing some patch, but no idea which one.

EHfive, 2019-02-12:
Why not rebuild your bluez with those patches?

EHfive, 2019-02-12:
Run `pactl list short modules | grep blue` to see if your bluetooth modules are loaded.

okias, 2019-02-12:
no module loaded :(

EHfive, 2019-02-12:
Append following lines to /etc/pulse/default.pa
```

.ifexists module-bluetooth-policy.so
load-module module-bluetooth-policy
.endif

.ifexists module-bluetooth-discover.so
load-module module-bluetooth-discover
.endif

```
And restart your pulseaudio, then check again.

EHfive, 2019-02-13:
Any response?

okias, 2019-02-13:
@EHfive cool, after running load module it's fine :) Thanks (I'll change config later)

closed 2019-02-13 by okias

pacho2, 2019-06-10:
The problem is that, when building pulseaudio without its bluetooth support (as it collides with this one), that lines in default.pa config file are removed. Isn't there a way for pulseaudio-modules-bt to provide the proper config file on its side to prevent this?

Thanks!

sriv, 2020-05-20:
I see something similar in popOS

```
bluetoothd[8257]: src/device.c:connect_profiles() /org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX (all), client :1.175
bluetoothd[8257]: profiles/audio/a2dp.c:a2dp_sink_connect() path /org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX
bluetoothd[8257]: a2dp-sink profile connect failed for XX:XX:XX:XX:XX:XX: Protocol not available
```

`/etc/pulse/default.pa` has the following entries:

```
### Automatically load driver modules for Bluetooth hardware
.ifexists module-bluetooth-policy.so
load-module module-bluetooth-policy
.endif

.ifexists module-bluetooth-discover.so
load-module module-bluetooth-discover
.endif
```
I tried to replace `load-module module-bluetooth-discover` with `load-module module-bluetooth-discover a2dp_config="ldac_eqmid=sq"` but it doesn't work.

The device I'm trying to pair is FiiO uBtr - https://www.fiio.com/%CE%BCbtr 

This device supports aptX (which I use in my android phone)

Please help!