EHfive/pulseaudio-modules-bt #34: AAC encoding error

prurigro, 2019-02-13:
First of all, thank you for this project! I didn't think I'd ever be able to use bluetooth headphones on Linux before finding this.

Everything seems to be working correctly with pactl showing the AAC codec being used, but after a certain amount of time (sometimes minutes, sometimes seconds), the pulseaudio output device automatically switches back to the default (line out) and journalctl shows the following error:

```
pulseaudio[22152]: E: [bluetooth] a2dp_aac.c: AAC encoding error, 0x60
```

I can then switch the output device back to the bluetooth headphones and it'll work again for a few more seconds or minutes until the error appears again. I tried manually switching the output away from the bluetooth headphones to see if the error was triggered in the other direction, but no error messages appear in that case. The bluetooth logs don't show any brief disconnects or anything either.

Let me know if there's any additional output or tests I can perform to help!

EHfive, 2019-02-13:
Please build from https://github.com/EHfive/pulseaudio-modules-bt/tree/aac_fix_dev .

If issue not fixed, show me the AAC error log

prurigro, 2019-02-13:
Thanks for the quick response and attempt at a fix! Unfortunately the error hit again, but I see you've added additional debugging output which will hopefully help:

```
pulseaudio[13772]: E: [bluetooth] a2dp_aac.c: AAC encoding error, 0x60; frame_count:0, in_bufSizes:4096, out_bufSizes 1332, to_read:4096, to_write:1332, encoded:0
```

prurigro, 2019-02-13:
I'm not sure if this will tell you anything, but I tested a few more times and the error is identical every time:

```
Feb 13 17:15:34 hostname pulseaudio[13772]: E: [bluetooth] a2dp_aac.c: AAC encoding error, 0x60; frame_count:0, in_bufSizes:4096, out_bufSizes 1332, to_read:4096, to_write:1332, encoded:0
Feb 13 17:19:42 hostname pulseaudio[13772]: E: [bluetooth] a2dp_aac.c: AAC encoding error, 0x60; frame_count:0, in_bufSizes:4096, out_bufSizes 1332, to_read:4096, to_write:1332, encoded:0
Feb 13 17:23:53 hostname pulseaudio[13772]: E: [bluetooth] a2dp_aac.c: AAC encoding error, 0x60; frame_count:0, in_bufSizes:4096, out_bufSizes 1332, to_read:4096, to_write:1332, encoded:0
```

EHfive, 2019-02-14:
> pulseaudio[13772]: E: [bluetooth] a2dp_aac.c: AAC encoding error, 0x60; frame_count:0, in_bufSizes:4096, out_bufSizes 1332, to_read:4096, to_write:1332, encoded:0
The log shows nothing wrong in buffers.
It probably is a AAC encoder's issue.

I pushed a commit to `[aac_fix_dev](https://github.com/EHfive/pulseaudio-modules-bt/tree/aac_fix_dev)` which make it initialize necessary AAC encoder modules only.
Also, you might consider upgrading you fdk-aac library.

prurigro, 2019-02-14:
It seems like I'm running the latest version of fdk-aac (1.0.0), but I'll try git master after giving your update to aac_fix_dev a shot.

prurigro, 2019-02-14:
I tested your update and still got the error. I then updated to the master branch of fdk-aac and got the error again. Downgrading to v0.1.6 of fdk-aac however seems to have resolved the issue (it's been running successfully for an hour now).

If newer versions of fdk-aac (2.0.0 and above so far) are going to break pulseaudio-modules-bt, would it be worth statically compiling v0.1.6 until things are resolved upstream so other people don't have this issue as well?

Either way, thank you for your help!

closed 2019-02-14 by prurigro

EHfive, 2019-02-14:
> If newer versions of fdk-aac (2.0.0 and above so far) are going to break pulseaudio-modules-bt, would it be worth statically compiling v0.1.6 until things are resolved upstream so other people don't have this issue as well?

Have you re-compiled pulseaudio-modules-bt after fdk-aac >= 2.0.0 installed?
If not, pulseaudio-modules-bt would still using the break one.
But because pulseaudio-modules-bt still link to `libfdk-aac.so.1`, after you got v0.1.6 installed the issue been resolved.
```bash
$ ldd libbluez5-util.so | grep aac
        libfdk-aac.so.2 => /usr/lib/libfdk-aac.so.2 (0x00007fa39da42000)
```

EHfive, 2019-02-14:
Still OK with fdk-aac git latest (2.0.0.r100.g95858d7)

prurigro, 2019-02-14:
Strange, maybe there's something unique about my setup-- I suppose if others have the same configuration quirks and run into this problem they can read this issue and see that they need to downgrade.

Thanks again for your help and this project!

prurigro, 2019-02-14:
Just noticed your other comment-- for comparison, I'm running:

pulseaudio v12.2 s16le 2ch 48000Hz
bluez 5.50

tobiaspredel, 2019-02-14:
With fix applied still E: [bluetooth] a2dp_aac.c: AAC encoding error, 0x60; frame_count:0, in_bufSizes:4096, out_bufSizes 1332, to_read:4096, to_write:1332, encoded:0
Now recompiled with downgraded libfdk-aac v0.2.6, also bluez 5.50, pulseaudio 12.2-2
So far no aac encoding errors.

CypherZealot, 2019-07-23:
I was having the same issue with libfdk--aac v2.0.0. Downgrading to v0.1.6 seems to have resolved the issue.

pulseaudio v12.2
bluez-git v5.50.r470.g79bf51dbb-1

Schlaefer, 2019-08-22:
My headphones were disconnecting with same error every few minutes too. I toyed with the settings and switching to CBR with `aac_bitrate_mode=0` seems to avoid the issue.

pulseaudio-modules-bt-git r141.3ecf1d7-1
bluez 5.50
libfdk-aac 2.2.0
pulseaudio 12.2

2by3, 2019-08-23:
That's a good tip. Changed the /etc/pulse/default.pa accordingly and
encountered no problems so far. 😀👍

lkraav, 2019-12-11:
Hmm, I'm not sure this issue should be closed. It's definitely constantly happening with `pulseaudio-modules-bt`, whereas it never happened with upstream `bluetooth` module implementation.

In my case, this error seems to consistently trigger within 30 min of Spotify playtime:

```
dets  11 15:42:42 papaya pulseaudio[89657]: [bluetooth] a2dp_aac.c: AAC encoding error, 0x60; frame_count:0, in_bufSizes:4096, out_bufSizes 1778, to_read:4096, to_write:1778, encoded:0
```

pulseaudio-13.0
bluez-5.51
fdk-aac-2.0.0

I have bluez-5.52 and fdk-aac-2.0.1 available in unstable, I guess I could try those out.

xhpohanka, 2020-05-11:
Hello, I'm also affected by this issue 
```
pulseaudio[2057]: E: [bluetooth] a2dp_aac.c: AAC encoding error, 0x60; frame_count:0, in_bufSizes:4096, out_bufSizes 1332, to_read:4096, to_write:1332, encoded:0
```
My system is Arch linux, pulseaudio 13.0-3, pulseaudio-modules-bt 1.3-6, bluez 5.54-1, libfdk-aac 2.0.1-1