EHfive/pulseaudio-modules-bt #35: a2dp_config not being passed

Coffee2CodeNL, 2019-02-15:
```
pactl list short modules | grep blue
7	module-bluetooth-policy		
8	module-bluetooth-discover		
9	module-bluez5-discover		
23	module-bluez5-device	path=/org/bluez/hci0/dev_70_26_05_57_B1_60 autodetect_mtu=0 a2dp_config=""
```

```
pacmd list-sinks
    index: 2
	name: <bluez_sink.70_26_05_57_B1_60.a2dp_sink>
	driver: <module-bluez5-device.c>
	flags: HARDWARE DECIBEL_VOLUME LATENCY 
	state: SUSPENDED
	suspend cause: IDLE
	priority: 9050
	volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
	        balance 0.00
	base volume: 65536 / 100% / 0.00 dB
	volume steps: 65537
	muted: no
	current latency: 0.00 ms
	max request: 2 KiB
	max rewind: 0 KiB
	monitor source: 3
	sample spec: s16le 2ch 44100Hz
	channel map: front-left,front-right
	             Stereo
	used by: 0
	linked by: 0
	fixed latency: 36.61 ms
	card: 1 <bluez_card.70_26_05_57_B1_60>
	module: 23
	properties:
		bluetooth.protocol = "a2dp_sink"
		bluetooth.a2dp_codec = "LDAC"
		device.description = "WH-1000XM2"
		device.string = "70:26:05:57:B1:60"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headset"
		bluez.path = "/org/bluez/hci0/dev_70_26_05_57_B1_60"
		bluez.class = "0x240404"
		bluez.alias = "WH-1000XM2"
		device.icon_name = "audio-headset-bluetooth"
		device.intended_roles = "phone"
	ports:
		headset-output: Headset (priority 0, latency offset 0 usec, available: unknown)
			properties:
				
	active port: <headset-output>
```

```
### Automatically load driver modules for Bluetooth hardware
.ifexists module-bluetooth-policy.so
load-module module-bluetooth-policy
.endif

.ifexists module-bluetooth-discover.so
#load-module module-bluetooth-discover
load-module module-bluetooth-discover a2dp_config="ldac_eqmid=hq ldac_fmt=f32"
.endif
```

EHfive, 2019-02-15:
Check the path of daemon.conf which pulseaudio is using.
```
$ pulseaudio --dump-conf
### Read from configuration file: /home/eh5/.config/pulse//daemon.conf ###
daemonize = no
fail = yes
high-priority = yes
...
```
And make sure you are editing the default.pa which at the same folder with daemon.conf.

Coffee2CodeNL, 2019-02-16:
Okay, so, had to reinstall my laptop anyway.

After a clean Fedora minimal installation, it passes the options, but it still isn't up to expectations: `sample spec: float32le 2ch 44100Hz`

Really hoping to get that 96KHz.

EHfive, 2019-02-16:
You need to edit the value of default-sample-frequency in daemon.conf.

closed 2019-02-16 by EHfive

easyrider, 2020-04-26:
> default-sample-frequency

in my case it was `default-sample-rate = 96000` as there is was no `default-sample-frequency`
which gives:
```
$ pacmd list-sinks
sample spec: s16le 2ch 96000Hz
```