EHfive/pulseaudio-modules-bt #37: Switching to an unsupported PA profile breaks all A2DP profiles until bluetooth device is reconnected

intelfx, 2019-02-17:
1. Connect to a device
   ```
   $ bluetoothctl

   [bluetooth]# connect 30:C0:1B:8D:9C:F1
   Attempting to connect to 30:C0:1B:8D:9C:F1
   [CHG] Device 30:C0:1B:8D:9C:F1 ServicesResolved: yes
   Connection successful
   [JBL Clip 3]#
   ```

2. Observe the card and sink
   ```
   $ pactl list cards
   <...>
   Card #4
          Name: bluez_card.30_C0_1B_8D_9C_F1
          Driver: module-bluez5-device.c
          Owner Module: 28
          Properties:
                  device.description = "JBL Clip 3"
                  device.string = "30:C0:1B:8D:9C:F1"
                  device.api = "bluez"
                  device.class = "sound"
                  device.bus = "bluetooth"
                  device.form_factor = "speaker"
                  bluez.path = "/org/bluez/hci0/dev_30_C0_1B_8D_9C_F1"
                  bluez.class = "0x200414"
                  bluez.alias = "JBL Clip 3"
                  device.icon_name = "audio-speakers-bluetooth"
          Profiles:
                  headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: yes)
                  a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: yes)
                  a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: no)
                  a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: no)
                  a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (sinks: 1, sources: 0, priority: 40, available: no)
                  a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (sinks: 1, sources: 0, priority: 40, available: no)
                  off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
          Active Profile: a2dp_sink_sbc
          Ports:
                  speaker-output: Speaker (priority: 0, latency offset: 0 usec, available)
                          Part of profile(s): headset_head_unit, a2dp_sink_sbc, a2dp_sink_aac, a2dp_sink_aptx, a2dp_sink_aptx_hd, a2dp_sink_ldac
                  speaker-input: Bluetooth Input (priority: 0, latency offset: 0 usec)
                          Part of profile(s): headset_head_unit

   $ pactl list sinks
   <...>
   Sink #9
          State: IDLE
          Name: bluez_sink.30_C0_1B_8D_9C_F1.a2dp_sink
          Description: JBL Clip 3
          Driver: module-bluez5-device.c
          Sample Specification: s16le 2ch 44100Hz
          Channel Map: front-left,front-right
          Owner Module: 28
          Mute: no
          Volume: front-left: 30148 /  46% / -20,23 dB,   front-right: 30148 /  46% / -20,23 dB
                  balance 0,00
          Base Volume: 65536 / 100% / 0,00 dB
          Monitor Source: bluez_sink.30_C0_1B_8D_9C_F1.a2dp_sink.monitor
          Latency: 46258 usec, configured 39512 usec
          Flags: HARDWARE DECIBEL_VOLUME LATENCY
          Properties:
                  bluetooth.protocol = "a2dp_sink"
                  bluetooth.a2dp_codec = "SBC"
                  device.description = "JBL Clip 3"
                  device.string = "30:C0:1B:8D:9C:F1"
                  device.api = "bluez"
                  device.class = "sound"
                  device.bus = "bluetooth"
                  device.form_factor = "speaker"
                  bluez.path = "/org/bluez/hci0/dev_30_C0_1B_8D_9C_F1"
                  bluez.class = "0x200414"
                  bluez.alias = "JBL Clip 3"
                  device.icon_name = "audio-speakers-bluetooth"
          Ports:
                  speaker-output: Speaker (priority: 0, available)
          Active Port: speaker-output
          Formats:
                  pcm
   ```

3. Try to switch to an unsupported codec (it looks like it worked)
   ```
   $ pactl set-card-profile bluez_card.30_C0_1B_8D_9C_F1 a2dp_sink_aac
   ```

4. Observe the sink didn't change
   ```
   $ pactl list sinks
   Sink #12
           State: SUSPENDED
           Name: bluez_sink.30_C0_1B_8D_9C_F1.a2dp_sink
           Description: JBL Clip 3
           Driver: module-bluez5-device.c
           Sample Specification: s16le 2ch 44100Hz
           Channel Map: front-left,front-right
           Owner Module: 29
           Mute: no
           Volume: front-left: 30148 /  46% / -20,23 dB,   front-right: 30148 /  46% / -20,23 dB
                   balance 0,00
           Base Volume: 65536 / 100% / 0,00 dB
           Monitor Source: bluez_sink.30_C0_1B_8D_9C_F1.a2dp_sink.monitor
           Latency: 0 usec, configured 0 usec
           Flags: HARDWARE DECIBEL_VOLUME LATENCY
           Properties:
                   bluetooth.protocol = "a2dp_sink"
                   bluetooth.a2dp_codec = "SBC"
                   device.description = "JBL Clip 3"
                   device.string = "30:C0:1B:8D:9C:F1"
                   device.api = "bluez"
                   device.class = "sound"
                   device.bus = "bluetooth"
                   device.form_factor = "speaker"
                   bluez.path = "/org/bluez/hci0/dev_30_C0_1B_8D_9C_F1"
                   bluez.class = "0x200414"
                   bluez.alias = "JBL Clip 3"
                   device.icon_name = "audio-speakers-bluetooth"
           Ports:
                   speaker-output: Speaker (priority: 0, available)
           Active Port: speaker-output
           Formats:
                   pcm
   ```

5. **At this point, if I try to play something via this sink, it hangs.**

6. Try to switch the profile back (it fails)
   ```
   $ pactl set-card-profile bluez_card.30_C0_1B_8D_9C_F1 a2dp_sink_sbc
   Failure: Input/Output error
   ```

7. This is where things go wrong. The sink disappears and does not reappear until the device is reconnected.
   ```
   $ pactl list sinks
   <...> (only irrelevant sinks here now)
   ```

8. Try to switch to HFP/HSP profile (it actually works)
   ```
   $ pactl set-card-profile bluez_card.30_C0_1B_8D_9C_F1 headset_head_unit
   ```

9. Observe the sink again
   ```
   Sink #13
           State: IDLE
           Name: bluez_sink.30_C0_1B_8D_9C_F1.headset_head_unit
           Description: JBL Clip 3
           Driver: module-bluez5-device.c
           Sample Specification: s16le 1ch 8000Hz
           Channel Map: mono
           Owner Module: 29
           Mute: no
           Volume: mono: 43691 /  67%
                   balance 0,00
           Base Volume: 65536 / 100%
           Monitor Source: bluez_sink.30_C0_1B_8D_9C_F1.headset_head_unit.monitor
           Latency: 29926 usec, configured 28000 usec
           Flags: HARDWARE HW_VOLUME_CTRL LATENCY
           Properties:
                   bluetooth.protocol = "headset_head_unit"
                   device.intended_roles = "phone"
                   device.description = "JBL Clip 3"
                   device.string = "30:C0:1B:8D:9C:F1"
                   device.api = "bluez"
                   device.class = "sound"
                   device.bus = "bluetooth"
                   device.form_factor = "speaker"
                   bluez.path = "/org/bluez/hci0/dev_30_C0_1B_8D_9C_F1"
                   bluez.class = "0x200414"
                   bluez.alias = "JBL Clip 3"
                   device.icon_name = "audio-speakers-bluetooth"
           Ports:
                   speaker-output: Speaker (priority: 0, available)
           Active Port: speaker-output
           Formats:
                   pcm
   ```

intelfx, 2019-02-17:
Note: the problem still exists with bluez git and experimental interfaces enabled.

EHfive, 2019-02-17:
> Note: the problem still exists with bluez git and experimental interfaces enabled.

Yes, I working on fix it.

commit 8a58314e (2019-02-17): Return PA_ERR_INVALID if trying to set unavailable card profile; Fix #37
closed 2019-02-17 by EHfive