EHfive/pulseaudio-modules-bt #38: SBC codec is marked "unavailable" on AirPods

intelfx, 2019-02-17:
```
$ bluetoothctl
[bluetooth]# connect 24:F6:77:DD:EE:00
Attempting to connect to 24:F6:77:DD:EE:00
[CHG] Device 24:F6:77:DD:EE:00 Connected: yes
Connection successful
[CHG] Device 24:F6:77:DD:EE:00 ServicesResolved: yes
[intelfx-pods]#
```

```
$ pactl list cards
<...>
Card #6
        Name: bluez_card.24_F6_77_DD_EE_00
        Driver: module-bluez5-device.c
        Owner Module: 30
        Properties:
                device.description = "intelfx-pods"
                device.string = "24:F6:77:DD:EE:00"
                device.api = "bluez"
                device.class = "sound"
                device.bus = "bluetooth"
                device.form_factor = "headphone"
                bluez.path = "/org/bluez/hci0/dev_24_F6_77_DD_EE_00"
                bluez.class = "0x240418"
                bluez.alias = "intelfx-pods"
                device.icon_name = "audio-headphones-bluetooth"
        Profiles:
                a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: no)
                a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: yes)
                a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: no)
                a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (sinks: 1, sources: 0, priority: 40, available: no)
                a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (sinks: 1, sources: 0, priority: 40, available: no)
                headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: no)
                off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
        Active Profile: a2dp_sink_aac
        Ports:
                headphone-output: Headphone (priority: 0, latency offset: 0 usec, available)
                        Part of profile(s): a2dp_sink_sbc, a2dp_sink_aac, a2dp_sink_aptx, a2dp_sink_aptx_hd, a2dp_sink_ldac, headset_head_unit
                headphone-input: Bluetooth Input (priority: 0, latency offset: 0 usec, not available)
                        Part of profile(s): headset_head_unit
```

Note that the SBC profile is "available: no":

> a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: no)

I think this is wrong because these headphones work with stock PulseAudio (which presumably does not support any of the advanced codecs).

EHfive, 2019-02-17:
Because you can't switching codec under bluez 5.50
https://github.com/EHfive/pulseaudio-modules-bt/wiki#can-i-choose-codec-

closed 2019-02-17 by EHfive

intelfx, 2019-02-17:
Yeah, I forgot to daemon-reload after I appended that `-E` flag to bluetoothd cmdline.