EHfive/pulseaudio-modules-bt #39: Enable AAC afterburner by default ?

vdemonchy, 2019-02-20:
Hello,

Since the AAC support is done through the Fraunhofer FDK AAC library, I've had a look at their wiki and they recommend to always enable the afterburner feature since it brings better quality (at the expense of more processing power required). However, the current implementation here disables it by defaut, so my question is shouldn't it be enabled instead ?

My source is: https://wiki.hydrogenaud.io/index.php?title=Fraunhofer_FDK_AAC#Afterburner

Thanks ! :)

EHfive, 2019-02-20:
I set that option to false because "afterburner" sucks more CPU resources.

Ref `aacenc_libs.h`
```
  AACENC_AFTERBURNER =
      0x0200, /*!< This parameter controls the use of the afterburner feature.
                   The afterburner is a type of analysis by synthesis algorithm
                 which increases the audio quality but also the required
                 processing power. It is recommended to always activate this if
                 additional memory consumption and processing power consumption
                   is not a problem. If increased MHz and memory consumption are
                 an issue then the MHz and memory cost of this optional module
                 need to be evaluated against the improvement in audio quality
                 on a case by case basis.
                   - 0: Disable afterburner (default).
                   - 1: Enable afterburner. */
```
------------------
Comparing results of `top` on my laptop

**afterburner on**
```
  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                           
 1366 eh5       21   1 2576836  22208  16320 S   3.3   0.1   0:13.64 /usr/bin/pulseaudio --daemonize=no 
```
**afterburner off**
```
  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                           
 3542 eh5       21   1 2568252  21380  15684 S   2.8   0.1   0:01.58 /usr/bin/pulseaudio --daemonize=no 
```

vdemonchy, 2019-02-20:
Yes I agree, however as it is said on the ref you've given: `It is recommended to always activate this if additional memory consumption and processing power consumption is not a problem`, and I don't think such a small increase in CPU and memory resource consumption is a problem with modern hardware :)

EHfive, 2019-02-20:
It's about 0.5/3.3 ~= 15% increment on pulseaudio process.
And the total CPU usage is about 15% on my laptop when playing.
15% * 15% ~= 2.3%.
Of course, it's depends.
But we should also consider about old slow machine running Linux or bad battery life on mobile device.

I think the default option should always be low resource required and compatible.

vdemonchy, 2019-02-20:
You make a point, conservative defaults are generally preferable indeed. 
It's configurable anyway, so I'm definitely okay with that ! I'll close this issue.

closed 2019-02-20 by vdemonchy