EHfive/pulseaudio-modules-bt #40: AAC codec not recognized

anthonytex, 2019-03-02:
Good morning,
i've some problem to make [these](https://item.mi.com/product/9209.html) [(english version)](https://www.xiaomitoday.com/xiaomi-airdots-pro/) earbuds working with AAC audio codec.
According to their specs, AAC is supported: in fact i'm able to make them working with some devices (e.g. Nvidia shield / Oneplus 3... the funniest thing is that they are not working with Xiaomi phones but this is another story).
*avinfo* doesn't recognize any 2nd sink profile (no unknown profile is listed)
I've attached some logs
Any hints? 
Thanks in advance 
Anthony
[bmon.log](https://github.com/EHfive/pulseaudio-modules-bt/files/2921951/bmon.log)
[bt.log](https://github.com/EHfive/pulseaudio-modules-bt/files/2921952/bt.log)

EHfive, 2019-03-03:
That's more like a upstream issue.
If you has tried run bluez git but the issue still exist, submit your issue to [linux-bluetooth mailing list](http://www.bluez.org/development/lists/).

EHfive, 2019-03-03:
> the funniest thing is that they are not working with Xiaomi phones but this is another story).

The newest qualcomm specific bluetooth stack on Android has a device white-list in AAC use case which may not contains your device.

closed 2019-03-15 by EHfive