EHfive/pulseaudio-modules-bt #41: LDAC sample rate won't go up to 96000hz

Provissy, 2019-03-06:
`pacmd list-sinks`   gives 
 `sample spec: float32le 2ch 44100Hz  `

According to a previous closed issue, default sample rate was set to 96000hz in this case 
`; default-sample-format = s16le
; default-sample-rate = 96000
; alternate-sample-rate = 44100`

In addition, default.pa was modified for
`load-module module-bluetooth-discover a2dp_config="ldac_eqmid=hq ldac_fmt=f32"`

Clementine music player was used and output sample rate was also set to 96000hz.
Why? So confused!

EHfive, 2019-03-06:
Run https://github.com/EHfive/pulseaudio-modules-bt/issues/31#issuecomment-462717049 to see if your device support LDAC 96000hz

EHfive, 2019-03-06:
This issue might duplicate https://github.com/EHfive/pulseaudio-modules-bt/issues/35#issuecomment-464238015

Provissy, 2019-03-06:
My headset is Sony's MDR-1000X and should support 96000hz, here is the result

Connecting ... 
Stream End-Point #1: Audio Sink 
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 44.1Khz 48Khz 
		Subbands: 4 8
		Blocks: 4 8 12 16 
		Bitpool Range: 2-53
	Content Protection: SCMS-T
Stream End-Point #5: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000012d
		Vendor Specific Codec ID 0x00aa
		Vendor Specific Value (LDAC)
			Unknown: 3c 07
	Content Protection: SCMS-T
Stream End-Point #3: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000004f
		Vendor Specific Codec ID 0x0001
		Vendor Specific Value (aptX)
			Frequencies: 44.1kHz 48kHz 
			Channel modes: Stereo 
	Content Protection: SCMS-T
Stream End-Point #2: Audio Sink 
	Media Codec: MPEG24
		Object Types: MPEG-2 AAC LC MPEG-4 AAC LC 
		Frequencies: 44.1kHz 48kHz 
		Channels: 1 2 
		Bitrate: 320000
		VBR: Yes
	Content Protection: SCMS-T

EHfive, 2019-03-06:
> Vendor Specific Value (LDAC)
> Unknown: 3c 07

Your headset's LDAC vendor data is "3c 07" which is equal with mine, suggest that your headset do has LDAC 96000hz support.

```
                Vendor Specific Data: 0x3c 0x07
                Vendor Specific Value (LDAC)
                        Frequencies: 44.1kHz 48kHz 88.2kHz 96kHz 
                        Channel modes: Mono Dual Stereo
```

So the reason is clear, you configured `daemon.conf` at wrong place.
Just follow https://github.com/EHfive/pulseaudio-modules-bt/issues/41#issuecomment-470074409 but ignore default.pa.

closed 2019-03-06 by EHfive