EHfive/pulseaudio-modules-bt #42: segfault / assertion error in module-bluez5-discover

bistromath, 2019-03-08:
Ubuntu 18.10, pulseaudio 12.2, bluez 5.50. Trying to get a Sony WH-1000XM3 connected.

from pulseaudio log:
(  18.287|   0.815) D: [pulseaudio] bluez5-util.c: Properties changed in device /org/bluez/hci0/dev_CC_98_8B_93_C3_3E
(  18.471|   0.184) D: [alsa-sink-USB Audio] alsa-sink.c: Wakeup from ALSA!
(  19.082|   0.611) D: [pulseaudio] backend-native.c: dbus: path=/Profile/HSPAGProfile, interface=org.bluez.Profile1, member=NewConnection
(  19.082|   0.000) D: [pulseaudio] backend-native.c: dbus: NewConnection path=/org/bluez/hci0/dev_CC_98_8B_93_C3_3E, fd=27, profile headset_head_unit
(  19.082|   0.000) I: [pulseaudio] backend-native.c: doing listen
(  19.082|   0.000) D: [pulseaudio] bluez5-util.c: Transport /org/bluez/hci0/dev_CC_98_8B_93_C3_3E/fd27 state: disconnected -> idle
(  19.082|   0.000) D: [pulseaudio] backend-native.c: Transport /org/bluez/hci0/dev_CC_98_8B_93_C3_3E/fd27 available for profile headset_head_unit
(  19.471|   0.388) D: [alsa-sink-USB Audio] alsa-sink.c: Wakeup from ALSA!
(  20.471|   1.000) D: [alsa-sink-USB Audio] alsa-sink.c: Wakeup from ALSA!
(  21.471|   0.999) D: [alsa-sink-USB Audio] alsa-sink.c: Wakeup from ALSA!
(  22.083|   0.611) D: [pulseaudio] bluez5-util.c: Timeout expired, and device /org/bluez/hci0/dev_CC_98_8B_93_C3_3E still has disconnected profiles: a2dp_sink
(  22.083|   0.000) D: [pulseaudio] module-bluez5-discover.c: Loading module-bluez5-device path=/org/bluez/hci0/dev_CC_98_8B_93_C3_3E autodetect_mtu=0 a2dp_config="ldac_eqmid=hq ldac_fmt=f32"
Segmentation fault (core dumped)

Occasionally I get the following at the end of the PA log:
(  16.053|   0.000) E: [pulseaudio] module-bluez5-device.c: Assertion 'pa_a2dp_codec_index_is_source(codec_index)' failed at modules/bluetooth/module-bluez5-device.c:1867, function create_card_profile(). Aborting.
Aborted (core dumped)

...And sometimes the following as well:
(  16.132|   0.000) E: [pulseaudio] module-bluez5-device.c: Assertion 'profiles[i].profile_string' failed at modules/bluetooth/module-bluez5-device.c:2117, function add_card(). Aborting.
Aborted (core dumped)

EHfive, 2019-03-08:
Try a clean build and a totally restart for pulseaudio, bluz

EHfive, 2019-03-08:
Temporaril, you can install all of optional dependencies to fix this

EHfive, 2019-03-08:
Try a clean build or a totally restart for pulseaudio, bluez.

EHfive, 2019-03-09:
What a terrible mistake!
Thank you so much for reporting, otherwise me would never found that.

commit d5fb3027 (2019-03-09): Fix #42
closed 2019-03-09 by EHfive