EHfive/pulseaudio-modules-bt #43: Doesn't build with fdk-aac 2.0

gombosg, 2019-03-14:
Latest 'Rawhide' version of Fedora comes with [fdk-aac 2.0](http://koji.rpmfusion.org/koji/packageinfo?packageID=477). Build fails with that since the build dependency is simply fdk-aac>=0.1.5 and 2.0 is accepted this way. Probably other distros will upgrade soon, as well.

Is it possible to add [2.0 support](https://github.com/mstorsjo/fdk-aac/releases/tag/v2.0.0)? (I know it's probably a breaking release.)

I have attached a build log.
[build.log](https://github.com/EHfive/pulseaudio-modules-bt/files/2967262/build.log)

Basically it's:
```
-- Checking for module 'fdk-aac>=0.1.5'
--   Found fdk-aac, version 2.0.0
```
but
```
BUILDSTDERR: /usr/bin/ld: cannot find -lfdk-aac
```

Here's the [fdk-aac-devel-2.0](http://koji.rpmfusion.org/koji/rpminfo?rpmID=417542) package info and here are [all the new fdk-aac packages](http://koji.rpmfusion.org/koji/buildinfo?buildID=10515).

Let me know if this is a Fedora packaging issue, but the [2.0 update](https://pkgs.rpmfusion.org/cgit/nonfree/fdk-aac.git/commit/fdk-aac.spec?id=7d34707898c7505a1425e654eb4eccf3e4bb1719) was as simple as a version bump. (Which only modified the source URL, with the same build config)

Thanks!

davidlt, 2019-03-14:
I think, this is a small bug in CMake file. It does not use LDFLAGS / set library directory for fdk-acc:

```
  3 libdir=/usr/lib64/fdk-aac                                                       
  4 includedir=/usr/include   
[..]
  9 Libs: -L${libdir} -lfdk-aac                                                     
 10 Libs.private: -lm                                                                                                                                                                                                               
 11 Cflags: -I${includedir}  
```

gombosg, 2019-03-14:
Switching to fdk-aac-free solves the problem, though. Found it dug deep in the repos.

EHfive, 2019-03-14:
It's there possible a ldconfig cache issue?

gombosg, 2019-03-14:
It fails in cleanly installed mock environments and on my computer, too.

davidlt, 2019-03-14:
https://koji.fedoraproject.org/koji/buildinfo?buildID=1186469

This is installed in standard system location (`/usr/lib64`), but fdk-aac is not (`/usr/lib64/fdk-aac`).

You need one line fix in CMake to use directory during linking provided from pkg-config. Then it will work for both.

gombosg, 2019-03-14:
See [this](https://bugzilla.rpmfusion.org/show_bug.cgi?id=5155#c24):

> rpmfusion fdk-aac had to be altered to avoid collision with the fedora package, see
> https://pkgs.rpmfusion.org/cgit/nonfree/fdk-aac.git/tree/fdk-aac.spec#n36
> The non-standard path will cause issues if your package doesn't use pkconfig

I'll use fdk-aac-free from now on but of course if you can fix this later, it's better.

EHfive, 2019-03-14:
Hey, I just pushed some changes (https://github.com/EHfive/pulseaudio-modules-bt/commit/f584da90177161c2e58c0df660b4e51c48f58c18) to branch ld-fix; Could you plz verify it?

EHfive, 2019-03-14:
Just to notice that I would bump version to v1.1.99 which contains commits (mostly fixes) after v1.1.

gombosg, 2019-03-14:
Confirmed, works fine with fdk-aac 2.0. In Fedora we'll use fdk-aac-free which is still the old version, so it's not urgent to push a new version, but this is useful for the future.

EHfive, 2019-03-14:
> Confirmed, works fine with fdk-aac 2.0. In Fedora we'll use fdk-aac-free which is still the old version, so it's not urgent to push a new version, but this is useful for the future.

Yep, It's just a version contains fixes for several tiny misses in v1.1

commit b252c7a2 (2019-03-14): Fix issues in linking library; Fix #43
closed 2019-03-14 by EHfive