EHfive/pulseaudio-modules-bt #44: GPL is incompatible with FDK-AAC - redistribution permission needed

gombosg, 2019-03-14:
See [this](https://bugzilla.rpmfusion.org/show_bug.cgi?id=5155#c20):

> FSF issued a clear statement that the fdk-aac is GPL incompatible
> https://www.gnu.org/licenses/license-list.en.html#fdk
> Can you clarify with upstream about this ? 
> Please ask upstream to **send an email to explicitly allow us to redistribute
> the software when built with fdk-aac.**
> Please send the email at root@rpmfusion.org from an upstream maintainer (using public maillist list).
**The email will be added to the scm if allowed**.
> Thx in advances.

2 issues:

- GPL incompatibility, see [here](https://github.com/negativo17/ffmpeg/issues/1), and for fdk-aac-free, [here](https://bugzilla.redhat.com/show_bug.cgi?id=1501522)
- Please send an email to the above address, cc me, too (gombosg at gmail dotcom)

e.g. 

> I, xx, upstream developer of the pulseaudio-modules-bluetooth-freeworld package explicitly allow RPMFusion to redistribute the software when built with fdk-aac libraries.

Thanks!

EHfive, 2019-03-14:
That's fine. 
I just sent the mail.

gombosg, 2019-03-14:
Thanks! If this is not good for them, I'll just disable AAC support. Have you thought about using `ffmpeg` AAC? Or is it not possible?
AAC license seems to be quite problematic.

EHfive, 2019-03-14:
It's totally possible because modules require only LC-AAC.
But we might loss some features like vbr, AAC afterburner .

EHfive, 2019-03-14:
I didn't implement dynamic loading on fdk-aac because of it's complex structure in header files;
By using ffmpeg for AAC support, we could share the current dynamic loading codes on ffmpeg.
I'll look into it when i have free time.

Closing the issue.

closed 2019-03-14 by EHfive