EHfive/pulseaudio-modules-bt #46: Can't get aptx or aptx_hd to be available

mumrau, 2019-03-18:
Hello.

I am under Fedora and I updated to the 1.1.3 rpm packages you provided (#20) which provides handy looking profiles.

The only sad part is that... I can't get my device (which is a Plantronics Backeat Pro 2) to get detected as a aptx enabled device as shown on the output below:
```
Card #3
	Name: bluez_card.E4_22_A5_BD_10_A9
	Driver: module-bluez5-device.c
	Owner Module: 25
	Properties:
		device.description = "PLT BB PRO 2"
		device.string = "E4:22:A5:BD:10:A9"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headset"
		bluez.path = "/org/bluez/hci0/dev_E4_22_A5_BD_10_A9"
		bluez.class = "0x240404"
		bluez.alias = "PLT BB PRO 2"
		device.icon_name = "audio-headset-bluetooth"
		device.intended_roles = "phone"
	Profiles:
		headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: yes)
		a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: yes)
		a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (sinks: 1, sources: 0, priority: 40, available: no)
		a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (sinks: 1, sources: 0, priority: 40, available: no)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: a2dp_sink_aac
	Ports:
		headset-output: Headset (priority: 0, latency offset: 0 usec)
			Part of profile(s): headset_head_unit, a2dp_sink_sbc, a2dp_sink_aac, a2dp_sink_aptx, a2dp_sink_aptx_hd, a2dp_sink_ldac
		headset-input: Headset (priority: 0, latency offset: 0 usec)
			Part of profile(s): headset_head_unit
```
I am nonetheless sure that it worked at some point with the previous package under Aptx.

Some audio package updated afterwards, and I had huge audio artifacts, removed the older version and tried the new version install (which obviously worked with the sink separations.

I tried un-pairing, repairing, unconneting, reconneting and so on, couldn't get the availability to pop up.

Any hint or known issue?

EHfive, 2019-03-18:
Have you got ffmpeg devel package intsalled?

mumrau, 2019-03-18:
Oh that's right, it looks like it solved the issue.

I probably missed it during my packages reinstallation and clean-up. I noticed that it is not specified as runtime dependencies on the README which probably lead me to my mistake.

closed 2019-03-18 by mumrau

EHfive, 2019-03-18:
> Oh that's right, it looks like it solved the issue.
> 
> I probably missed it during my packages reinstallation and clean-up. I noticed that it is not specified as runtime dependencies on the README which probably lead me to my mistake.

Dynamic loading requires library has no version suffix which usually is a soft link contained in devel package.

Those runtime dependencies in README not stand for distributed packages but the thing itself.