EHfive/pulseaudio-modules-bt #50: Loading soversion-less libraries

gombosg, 2019-04-08:
See #20 and by @devurandom [this comment](https://github.com/EHfive/pulseaudio-modules-bt/issues/20#issuecomment-480866908) for LDAC, plus see [this bug report](https://bugzilla.rpmfusion.org/show_bug.cgi?id=5226) for libavcodec.

Essentially, at runtime the package should require the same soname version that it was compiled with, since soversion-less files are only in -devel packages, at least this is the standard for Fedora.

closed 2019-04-09 by EHfive

rathann, 2019-04-09:
I'm not sure it makes sense to use `dlopen()` vs just linking dynamically at build time, at least from distribution point of view.

gombosg, 2019-04-09:
This works just great @EHfive !
Do you specifically tie to `libavcodec.so.58` or does it allow anything higher than that?
Any plans to do a release or should I pull this specific commit for the package update?

leigh123linux, 2019-04-09:
@EHfive So you don't support libavcodec.so.57?

http://ffmpeg.org/download.html

EHfive, 2019-04-10:
> Do you specifically tie to `libavcodec.so.58` or does it allow anything higher than that?
It will search `libavcodec.so.58`,  `libavcodec.so`, until one successfully loaded
You could specify that as 58.
So does libldac
> Any plans to do a release or should I pull this specific commit for the package update?
OK, I will release v1.1.99 in few days.

> I'm not sure it makes sense to use `dlopen()` vs just linking dynamically at build time, at least from distribution point of view.
 Use `dlopen()` makes ffmpeg or libldac be runtime optional.

> @EHfive So you don't support libavcodec.so.57?

Only above libavcodec.so.58 has aptX support.

devurandom, 2019-04-10:
>  Do you specifically tie to `libavcodec.so.58` or does it allow anything higher than that?

It makes sense, because ffmpeg 4.1 / libavcodec.so.58 is the latest version, and you cannot yet know what ABI changes the next version will bring. Once the next version is released, the imported symbols / functions and their usage within the pulseaudio module will be adjusted. That's when it makes sense to add another `SONAME` to the list of `dlopen`ed libraries.

EHfive, 2019-04-12:
@gombosg  https://github.com/EHfive/pulseaudio-modules-bt/releases/tag/v1.1.99 has released.

gombosg, 2019-04-12:
@EHfive Super, thanks! I pushed [this update](http://koji.rpmfusion.org/koji/packageinfo?packageID=588).

gombosg, 2019-04-12:
@EHfive It works for me, and works for 1 tester, but another got [this error](https://bugzilla.rpmfusion.org/show_bug.cgi?id=5232#c0):
```
Apr 12 21:36:11 pulseaudio[2318]: E: [pulseaudio] ltdl-bind-now.c: Failed to open module /usr/lib64/pulse-12.2/modules/module-bluez5-device.so: /usr/lib64/pulse-12.2/modules/module-bluez5-device.so: undefined symbol: pa_a2dp_codec_index_is_source
Apr 12 21:36:11 pulseaudio[2318]: E: [pulseaudio] module.c: Failed to open module "module-bluez5-device".
```

EHfive, 2019-04-13:
You should mention that guy it requires a restart of pulseaudio to reload all of  those modules.

`pa_a2dp_codec_index_is_source` is a function in `libbluez5-util.so`, which is previously loaded by `module-bluez5-discover.so`

```
$ ldd module-bluez5-discover.so | grep blue
libbluez5-util.so => /usr/lib/pulse-12.2/modules/libbluez5-util.so (0x00007f99b69c9000)
```

`module-bluez5-discover.so` is a loader which dynamically loads `module-bluez5-device.so` when a new Audio BT device connected, but `module-bluez5-device.so` also links to `libbluez5-util.so`

```
$ ldd module-bluez5-device.so | grep blue
libbluez5-util.so => /usr/lib/pulse-12.2/modules/libbluez5-util.so (0x00007f99b69c9000)
```

If installing `pulseaudio-modules-bt` without reloading, when a device conneted, `module-bluez5-device.so` would be new, but `module-bluez5-discover.so` ,`libbluez5-util.so` are still the stock version which have no `pa_a2dp_codec_index_is_source`... , then error happens.

gombosg, 2019-04-13:
@EHfive thanks!
@rathann hm, maybe just `pulseaudio -k` restarts? Or a system restart simply?

EHfive, 2019-04-13:
> @EHfive thanks!
> @rathann hm, maybe just `pulseaudio -k` restarts? Or a system restart simply?

Yes, or `systemctl --user restart pulseaudio`, or `pactl unload-module`, `pactl load-module`

rathann, 2019-04-14:
It works after restarting pulseaudio, thank you.

gombosg, 2019-04-19:
Does anyone know how to do this module loading/unloading from a `%post` scriptlet? It's root, but `pulseaudio` is a user instance.

For `pulseaudio -k` I get
```
E: [pulseaudio] main.c: Failed to kill daemon: No such process
```

For `pactl unload-module` I get errors, too.

Basically, how would you restart pulseaudio, or load-unload the module from a root console?

eclipseo, 2019-04-19:
Try with:

```
%postun
if [ $1 -ge 1 ] ; then 
        # Package upgrade, not uninstall 
        systemctl --user try-restart pulseaudio >/dev/null 2>&1 || : 
fi
```

commit 765766c4 (2019-04-09): Fix #50