EHfive/pulseaudio-modules-bt #51: Add HFP support ?

joakim-tjernlund, 2019-04-09:
Over at https://git.kernel.org/pub/scm/linux/kernel/git/jejb/pulseaudio.git/log/?h=jejb-v12
there are patches for HFP support so that headset with HFP only profile will
work with PA. It would be nice if you could pull in the first 4 patches(sans mSBC)
into here so one can use such headsets with these new codes

EHfive, 2019-04-12:
So headset with HFP only profile will can't work with ofono?

joakim-tjernlund, 2019-04-12:
Theory has it that ofono could work but then you have setup some sort of dummy modem and configure
stuff manually. Consider if HSP were the same, would you accept that?
HSP/HFP should work the same I think in PA. ofono is for special needs.
The first 4 patches just adds a working HFP profile which I think applies on top of your repo.
The mSBC stuff does not apply and is extra ATM

EHfive, 2019-04-12:
Merged to https://github.com/EHfive/pulseaudio-modules-bt/commits/hfp,
could you please make a test?
My headset support both HFP and HSP, but can't get it connected with HFP.

joakim-tjernlund, 2019-04-12:
Thanks, will test but I am leaving for one a week vacation tmw so I will test when I get back(headset at home)
That said, I don't think it will work with your patch on top(to prefer HSP) as my headset a "false"
HSF UUID that does work at all. I think that adding this was to PA was a mistake, it didn't
fix HFP headsets at all. I am referring to PA_BLUETOOTH_UUID_HSP_HS_ALT

EHfive, 2019-04-12:
https://github.com/EHfive/pulseaudio-modules-bt/commit/9109668effa97baf09965d471dbb8f37b9341d6a is not for prefer HSP, but leave HSP,HFP choosing to bluez.

The concernd thing is my headset can't get conneted in bluez layer without that commit.

So there might be some conflicts with HSP/HFP in bluez or PA behaivor, we could add an option making module prefer HFP to solve it.

joakim-tjernlund, 2019-04-12:
Oh, I was too quick reading that commit, sorry. I will check my headset  at home then.

joakim-tjernlund, 2019-04-12:
I should mention that those HFP patches on top of PA 12.2 does work for me.

redg3ar, 2019-05-05:
This branch works as well for me. Thank you very much for your work on this stuff!

dceee, 2019-06-14:
> This branch works as well for me. Thank you very much for your work on this stuff!

Does that enable wideband speech out of the box for conference calls or do I still need to setup ofono for this?

joakim-tjernlund, 2019-08-28:
I don't know, easy to test though.

ailin-nemui, 2019-08-30:
Thanks for this branch/patch, I can use it to get my Logitech H800 Bluetooth head set working; however, I cannot switch to a2dp_sink_sbc ("hq audio") after I apply this patch, any idea?

joakim-tjernlund, 2019-08-30:
Apply patch on top of master ?
This branch is old

ailin-nemui, 2019-08-30:
thanks for the idea, that's what I did!

 (running on Pulseaudio 12.2 as provided by openSUSE)

joakim-tjernlund, 2019-08-30:
These patches should not affect a2dp so not sure what is wrong.
Try rebuild you src without them and see if it works, then add a  patch at a time

joakim-tjernlund, 2019-09-06:
@EHfive , any chance you can add these to your patch set ?

joakim-tjernlund, 2019-09-21:
James has upped his patch set to PA 13:
https://git.kernel.org/pub/scm/linux/kernel/git/jejb/pulseaudio.git/log/?h=jejb-v13

jmfernandez, 2019-10-26:
> @EHfive , any chance you can add these to your patch set ?

As @EHfive told on https://github.com/EHfive/pulseaudio-modules-bt/issues/51#issuecomment-482541443 , the patches were added to [hfp branch](https://github.com/EHfive/pulseaudio-modules-bt/tree/hfp) . BTW, the patches do their work. The only issues I have found are that the A2DP profile remains unavailable when applied, and the volume of the device can only be adjusted either at the application level or with its embedded buttons.

jmfernandez, 2019-10-27:
@ailin-nemui 
> Thanks for this branch/patch, I can use it to get my Logitech H800 Bluetooth head set working; however, I cannot switch to a2dp_sink_sbc ("hq audio") after I apply this patch, any idea?

@joakim-tjernlund 
> These patches should not affect a2dp so not sure what is wrong.
> Try rebuild you src without them and see if it works, then add a patch at a time

From my experience with my headphones, I guess the problem here is on PulseAudio - BlueZ interaction. PulseAudio backend first identifies all the profiles, and then it tries activating them in order to mark them as available. But the HFP profile identification does some kind of activation first, so, A2DP profile fails.

My trick to get both A2DP and HFP profiles has been:

1. Connecting the headphones: the A2DP profile should appear *unavailable* in pavucontrol.

2. Restarting the PulseAudio daemon (for instance, with pulseaudio -k) which is going to disconnect from the headphones.

3. Last, using some tool like bluetoothctl (blueman could also work), explicitly reconnecting to the headphones. Both HFP and A2DP profiles should appear available in pavucontrol.

joakim-tjernlund, 2019-10-30:
@jmfernandez , are you using latest bluez(5.51)? There is a patch on top of 5.51:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=52faac4648500dd45c06b6aa41f1be426c43a125
I do not know if that fixes you case, just an idea.

joakim-tjernlund, 2019-11-02:
bluez 5.52 is out

joakim-tjernlund, 2019-11-02:
Just played som with my HFP headset and i too have problems to switch to A2DP, it just fails.
But then I rembered reading somewhere that some headsets needs some fiddling.
I press Play button on headset and after that I switch between A2DP and HFP
until I turn headset off

joakim-tjernlund, 2019-11-02:
@jmfernandez There are some fixes on top of the original HFP patches over at
https://github.com/paravoid/pulseaudio/commits/hfp-v7

This stands out
bluetooth: support soft volume control for HFP

jmfernandez, 2019-11-04:
> @jmfernandez , are you using latest bluez(5.51)? There is a patch on top of 5.51:
> https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=52faac4648500dd45c06b6aa41f1be426c43a125
> I do not know if that fixes you case, just an idea.

> bluez 5.52 is out

@joakim-tjernlund  Yes, I'm currenly using 5.51 . I'm testing either the patch or 5.52 this evening, thanks!

jmfernandez, 2019-11-04:
> @jmfernandez There are some fixes on top of the original HFP patches over at
> https://github.com/paravoid/pulseaudio/commits/hfp-v7
> 
> This stands out
> bluetooth: support soft volume control for HFP

That patch seem promising, I will test it

eddyb, 2020-03-18:
Any updates on this? I patched the base `pulseaudio` package in NixOS before realizing that `pulseaudio-modules-bt` replaces the relevant modules and the `hfp-v7` patches don't apply cleanly.

With just base `pulseadio` + `hfp-v7` I lose the ability to enable A2DP as well, so that's not an option for me.

joakim-tjernlund, 2020-03-18:
> Any updates on this? I patched the base `pulseaudio` package in NixOS before realizing that `pulseaudio-modules-bt` replaces the relevant modules and the `hfp-v7` patches don't apply cleanly.
>
> With [...]

Some headsets needs some fiddling, like pressing Play button once to allow switching.
You can find more somewhere on PA home pages.

That said, when I upgraded to bluez-5.53/5.54 my need for pressing Play went away.

herrzinter, 2020-04-16:
Tried to compile https://github.com/paravoid/pulseaudio/commits/hfp-v7 under arch linux and succeeded, but if I used it I can not switch to hfp profile anymore :/

bhack, 2020-04-22:
See also upstream PR status at https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/254

mirh, 2020-06-11:
This is probably more important
https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/288

Ph0rk0z, 2020-06-13:
Yes please, I'd like to see battery level finally.

EHfive, 2020-06-14:
I would not merge this. But the idea of https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/288 is great!

I am reading/learning HSP/HFP/A2DP/... specs and bluez sources currently. 
I would like to implement a daemon which handles both HSP/HFP and A2DP.

joakim-tjernlund, 2020-06-15:
> I would not merge any of those patches. But the idea of https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/288 is great!
> 
> I am reading/learning HSP/HFP/A2DP/... specs and bluez sources currently.
> I would like to implement a daemon which handles both HSP/HFP and A2DP.

There is an perl prototype daemon at https://github.com/pali/hsphfpd-prototype
It works fine for my and a few more people

SpineEyE, 2020-07-16:
Has there been any progress? Optimally I would like to use your repository merged with https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/288

Because I can't switch my headset to HSP/HFP modeo with your repo, I created [this AUR package](https://aur.archlinux.org/packages/pulseaudio-hsphfpd/) together with [hsphfpd](https://aur.archlinux.org/packages/hsphfpd-git/) which gives me a working headset with mic but unfortunately a worse audio quality than with your repo.

mirh, 2020-07-17:
Of course high quality A2DP is gonna be better than HFP.. But you can't really compare the two things. 
https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/254
This is the PR you are waiting for if any.

SpineEyE, 2020-07-17:
Well yeah. But with your repo, as well as with current Pulseaudio stable release I can not even use HFP, that's my problem. Sometimes it works, sometimes it shows as unavailable, sometimes it's available but when I want to select it, it automatically jumps to "off".

I know, this repo is not a solution for high quality HFP, but with the merge request 288 I can't listen to music as the quality is too bad.

domenkozar, 2020-08-03:
With Freebuds 3 I have the same problem. I'm fine switching profiles as long as I can have one with quality audio and one with a working mic.

Going to try to package it for NixOS.

@SpineEyE can you switch between the two profiles? I'm also confused why this needs to be merged here and not into pulseaudio.

SpineEyE, 2020-08-03:
Yes, using pali's hsphfpd branch I can switch between high quality audio without mic, and HSP with mic. In fact there are multiple audio profiles (HFP can't be used, but isn't useful anyway):

[image]

It would preferrably be merged into Pulseaudio, but there is a discussion that pali's changeset would remove some features used by an old component that a minority of users somehow still need, i.e. the outdated HFP profile. pali commented that it would be too much work to modify his changeset and the maintainers can't do it themselves.

Seeing that this repository here exists to me it seems like the Pulseaudio maintainers have problems to keep up with users' needs even though solutions exist.

domenkozar, 2020-08-03:
Seems like @georgewhewell packaged it in https://github.com/georgewhewell/nixos-host/blob/98c6ccfb30072d1bcc207a74f223590d906737ff/modules/hsphfpd.nix

gombosg, 2020-10-08:
> Seeing that this repository here exists to me it seems like the Pulseaudio maintainers have problems to keep up with users' needs even though solutions exist.

Off:
When I packaged this for Fedora last year, I was aware that [pali's changes](https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/227) into PA could make this package obsolete quick... but it didn't. Which is actually a bit sad.

mirh, 2020-10-08:
You can help them: https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues/962
https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues/733

domenkozar, 2020-12-08:
Note that @pali has deleted that fork on gitlab.

mirh, 2020-12-08:
See https://github.com/EHfive/pulseaudio-modules-bt/issues/115#issuecomment-739032966

closed 2020-12-20 by EHfive