EHfive/pulseaudio-modules-bt #52: PPA Ubuntu Disco Support

v00d00, 2019-04-10:
Ubuntu 19.04 (disco) is released next week, any chance the PPA can be updated?

EHfive, 2019-04-12:
Since patching support is removed, I am planning to add this as a sperate package as a replacement of pulseaudio-bluetooth.

mhalano, 2019-04-12:
I think is always a bummer to replace an official package to add a feature, but seems that's the only option.

satmandu, 2019-04-19:
Would you please also enable arm64 builds in the PPA when you enable the disco builds please?

plumlis, 2019-05-24:
Also it's time to update PPA to latest version.(include 18.04 / 18.10)
Before that I'll try to compile myself, Hope it works well

polesapart, 2019-05-29:
> Since patching support is removed, I am planning to add this as a separate package as a replacement of pulseaudio-bluetooth.

could you clarify that please?

francescortiz, 2019-06-01:
I believe that it means that now the aptx support will have to be a replacement of official pulseadio-bluetooth rather than an additional package.

lelecanfora, 2019-06-04:
Any updates on this? Currently the ppa can't be installed on 19.04 at all

yassineim, 2019-06-05:
@Lele you can replace “disco” by “cosmic” in the PPA’s .lst as a temporary
workaround. I can’t guarantee it won’t break anything, although that’s what
I have done and it’s working alright.

Linux-Fan, 2019-06-05:
So any chance for update (Kubuntu 19.04)?

v6277, 2019-06-08:
> @Lele you can replace “disco” by “cosmic” in the PPA’s .lst as a temporary workaround. I can’t guarantee it won’t break anything, although that’s what I have done and it’s working alright.
> […](#)

How do you go about doing this? Sorry for the noob question.

v6277, 2019-06-08:
> @Lele you can replace “disco” by “cosmic” in the PPA’s .lst as a temporary workaround. I can’t guarantee it won’t break anything, although that’s what I have done and it’s working alright.
> […](#)

@yassineim How do you go about doing this? Sorry for the noob question, I'm a but new at this.

yassineim, 2019-06-08:
First try to add the repo using add-apt-repository. Ignore any errors. Edit
the file /etc/apt/sources.list.d/_pulseaudio-something_.list with nano or
anything, look for the word "disco" and replace it with "bionic". Then do
another apt update, and install the package.

If you need any more details let me know!

EHfive, 2019-06-11:
Done! 
Ubuntu 18.04, 18.10 also got updates.
> https://launchpad.net/~eh5/+archive/ubuntu/pulseaudio-a2dp/+sourcepub/10215653/+listing-archive-extra
> https://launchpad.net/~eh5/+archive/ubuntu/pulseaudio-a2dp/+sourcepub/10215650/+listing-archive-extra
> https://launchpad.net/~eh5/+archive/ubuntu/pulseaudio-a2dp/+sourcepub/10215649/+listing-archive-extra

```
sudo add-apt-repository ppa:eh5/pulseaudio-a2dp
sudo apt-get update
sudo apt-get install libavcodec libldac pulseaudio-modules-bt
```

closed 2019-06-11 by EHfive