EHfive/pulseaudio-modules-bt #53: a2dp_sink_*: Unavailable profile (except SBC)

e-minguez, 2019-04-11:
Fedora 29 with latest pulseaudio-module-bluetooth-freeworld package:

```
Name        : pulseaudio-module-bluetooth-freeworld
Version     : 1.1
Release     : 3.fc29
Architecture: x86_64
Install Date: mar 09 abr 2019 10:52:52 CEST
Group       : Unspecified
Size        : 257920
License     : GPLv3
Signature   : RSA/SHA1, jue 04 abr 2019 16:54:04 CEST, Key ID 1191a7c442f19ed0
Source RPM  : pulseaudio-module-bluetooth-freeworld-1.1-3.fc29.src.rpm
Build Date  : mié 03 abr 2019 19:35:01 CEST
Build Host  : buildvm-06.virt.rpmfusion.net
Relocations : (not relocatable)
Packager    : RPM Fusion
Vendor      : RPM Fusion
URL         : https://github.com/EHfive/pulseaudio-modules-bt/
Summary     : Bluetooth support for the PulseAudio sound server, supports aptX, LDAC codecs
Description :
Contains Bluetooth audio (A2DP/HSP/HFP) support for the PulseAudio sound server.
Includes support for LDAC, aptX and aptX-HD codecs.
```

I can select aptX, aptX-HD and LDAP codecs with pavucontrol or gnome settings but it doesn't work.

Journalctl says:

```
abr 11 14:21:55 galactica.minwi.lan pulseaudio[2276]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_ldac: Unavailable profile
abr 11 14:21:58 galactica.minwi.lan pulseaudio[2276]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_aptx_hd: Unavailable profile
abr 11 14:22:00 galactica.minwi.lan pulseaudio[2276]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_aptx: Unavailable profile
```

bwars, 2019-04-11:
I'm having the same issue. Also on F29, also with pulseaudio-module-bluetooth-freeworld-1.1-3.fc29.src.rpm.

I definitely have ffmpeg-devel and ffmpeg-libs installed, with libavcodec 58 and libavutil 56.

EHfive, 2019-04-12:
Check https://github.com/EHfive/pulseaudio-modules-bt/issues/31

EHfive, 2019-04-12:
Merged to https://github.com/EHfive/pulseaudio-modules-bt/commits/hfp,
could you please make a test?
My headset support both HFP and HSP, but can't get it conneted with HFP.

rathann, 2019-04-14:
I have a similar issue.

Unfortunately, Fedora's `bluez` package comes without the `avinfo` tool included: [bug #1699680](https://bugzilla.redhat.com/show_bug.cgi?id=1699680), so I rebuilt it myself with the tool included. My headset seems to support SBC only:
```
Connecting ... 
Stream End-Point #1: Audio Sink 
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 16Khz 32Khz 44.1Khz 48Khz 
		Subbands: 4 8
		Blocks: 4 8 12 16 
		Bitpool Range: 2-53
	Content Protection: SCMS-T
```

bwars, 2019-04-15:
I also have a similar issue on Fedora. I am connecting to a BTR3 device (https://www.fiio.com/btr3) which supports SBC, aptX, aptX-HD, and LDAC. However, I can only connect using aptX-HD. If I try to switch to any other audio profile with either gnome settings or blueman, I get the same error in journalctl.

My avinfo output is as follows:

> Connecting ... 
Stream End-Point #6: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0002
	Content Protection: SCMS-T
Stream End-Point #7: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x000000d7
		Vendor Specific Codec ID 0x0024
	Content Protection: SCMS-T
Stream End-Point #5: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000004f
		Vendor Specific Codec ID 0x0001
		Vendor Specific Value (aptX)
			Frequencies: 44.1kHz 48kHz 
			Channel modes: Stereo 
	Content Protection: SCMS-T
Stream End-Point #4: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0001
Stream End-Point #3: Audio Sink 
	Media Codec: MPEG24
		Object Types: MPEG-2 AAC LC MPEG-4 AAC LC 
		Frequencies: 8kHz 11.025kHz 12kHz 16kHz 22.05kHz 24kHz 32kHz 44.1kHz 48kHz 
		Channels: 1 2 
		Bitrate: 320000
		VBR: Yes
	Content Protection: SCMS-T
Stream End-Point #2: Audio Sink 
	Media Codec: MPEG12
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 16Khz 22.05Khz 24Khz 32Khz 44.1Khz 48Khz 
		CRC: Yes
		Layer: 3 
		Bit Rate: Free format
		VBR: Yes
		Payload Format: RFC-2250
	Content Protection: SCMS-T
Stream End-Point #9: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000012d
		Vendor Specific Codec ID 0x00aa
		Vendor Specific Value (LDAC)
			Unknown: 3c 07
	Content Protection: SCMS-T
Stream End-Point #8: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000053a
		Vendor Specific Codec ID 0x484c
	Content Protection: SCMS-T
Stream End-Point #1: Audio Sink 
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 16Khz 32Khz 44.1Khz 48Khz 
		Subbands: 4 8
		Blocks: 4 8 12 16 
		Bitpool Range: 2-53
	Content Protection: SCMS-T

EHfive, 2019-04-15:
@bwars 

Have you got ffmpeg and libldac installed?

bwars, 2019-04-15:
> Have you got ffmpeg and libldac installed?

Aha, I was missing libldac-devel. After restarting pulseaudio and the bluetooth service, it now uses using the LDAC profile by default.

However, I still cannot switch to other audio profiles - including the aptX-HD one that it was using previously!

EHfive, 2019-04-15:
@bwars 
https://github.com/EHfive/pulseaudio-modules-bt/wiki#can-i-choose-codec-

EHfive, 2019-04-15:
The "Unavailable", literally ... unavailable, at least inside those modules

bwars, 2019-04-15:
I tried enabling the -E option, but it didn't help. I'm using bluez-5.50-3.fc29 from the official Fedora repository though - perhaps the experimental features aren't available in that?

If I do "pactl list sinks | grep -e bluez -e a2dp_codec" I do indeed just get the LDAC option listed.

For the record, I'm also using pulseaudio-module-bluetooth-freeworld-1.1-3.fc29.

EHfive, 2019-04-15:
@bwars 
[image]

Help yourself, plzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzzz

bwars, 2019-04-15:
Yeah, I figured it would be that, just thought I'd give an update first as I can't try that just now. Thanks for the help!

e-minguez, 2019-04-16:
> I have a similar issue.
>
> Unfortunately, Fedora's `bluez` package comes without the `avinfo` tool included: [bug #1699680](https://bugzilla.redhat.com/show_bug.cgi?id=1699680), so I rebuilt it myself [...]

I've rebuilt the Fedora's bluez rpm package (https://packagecloud.io/eminguez/bluez) and now I have `avinfo` but I don't have my speakers to test it :) I'll update the comment to reflect my findings.

Thanks.

e-minguez, 2019-04-17:
avinfo output:

```
Connecting ... 
Stream End-Point #6: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0002
	Content Protection: SCMS-T
Stream End-Point #5: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000004f
		Vendor Specific Codec ID 0x0001
		Vendor Specific Value (aptX)
			Frequencies: 44.1kHz 48kHz 
			Channel modes: Stereo 
	Content Protection: SCMS-T
Stream End-Point #53: Audio Source 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0106
	Content Protection: SCMS-T
Stream End-Point #21: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0106
	Content Protection: SCMS-T
Stream End-Point #49: Audio Source 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0103
	Content Protection: SCMS-T
Stream End-Point #17: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0103
	Content Protection: SCMS-T
Stream End-Point #33: Audio Source 
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 16Khz 32Khz 44.1Khz 48Khz 
		Subbands: 4 8
		Blocks: 4 8 12 16 
		Bitpool Range: 2-53
	Content Protection: SCMS-T
Stream End-Point #1: Audio Sink 
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 16Khz 32Khz 44.1Khz 48Khz 
		Subbands: 4 8
		Blocks: 4 8 12 16 
		Bitpool Range: 2-53
	Content Protection: SCMS-T

```

EHfive, 2019-04-18:
@e-minguez 
> Now I can use aptX, but no LDAC (which doesn't seem to be supported by the headphones), nor aptX HD (which doesn't seem to be supported by the headphones) nor SBC (IDK why):

There is a bug in bluez; https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=9f7659a44601e043bdb6724b0ab4f3f59c46e9fe.
Then,  I commited https://github.com/EHfive/pulseaudio-modules-bt/commit/9ff89b430c67e98eb3a74e0f06a8957449e21927 after v1.1.99.

So you have to upgrade both pulseaudio-modules-bt and bluez  to newest, or downgrade bluez revision which before https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=9f7659a44601e043bdb6724b0ab4f3f59c46e9fe

closed 2019-04-24 by EHfive