EHfive/pulseaudio-modules-bt #56: Is there a config option to force AAC codec?

uentity, 2019-05-02:
I had an early preview of this module installed on Fedora 29 and noticed that my AKG-Y50BT was connecting using AAC codec instead of AptX as I expected.
After updating to Fedora 30 headphones begin constantly connecting via AptX codec. And surprisingly I found myself preferring AAC more.

Is there a config option to "force" AAC? I did changed `bluetooth.service` as WIKI describing, but don't see AAC in KDE profiles selection window.

xstsx, 2019-05-03:
If you don't need AptX at all , you can compile without Aptx 
`cmake -DCODEC_APTX_HD_FF=OFF -DCODEC_APTX_FF=OFF`

uentity, 2019-05-03:
Sound like Thanos Snap -)
Maybe there's another way?

EHfive, 2019-05-04:
https://github.com/EHfive/pulseaudio-modules-bt/issues/53#issuecomment-484336536

In short, build latest version from pulseaudio-modules-bt git or bluez git yourself.

uentity, 2019-05-05:
@EHfive build `pulseaudio-modules-bt` from current master, but situation didn't change.
Still, I only can use AptX and other codecs are listed as unavailable (including SBC). AAC profile not even listed in `pavucontrol`.

Is that because of outdated `bluez` in fedora?

EHfive, 2019-05-07:
> In short, build ***latest*** version from pulseaudio-modules-bt git or bluez git yourself.

closed 2019-05-07 by EHfive

gombosg, 2019-06-28:
Hi, now that `fdk-aac-free` is deemed to be GPL compatible I've made a Fedora build. Can you test [this build](http://koji.rpmfusion.org/koji/taskinfo?taskID=335570) with AAC? Thanks!