EHfive/pulseaudio-modules-bt #57: LDAC Not unavailable

mrdev023, 2019-05-06:
Headphone: WH-1000XM3
OS: Manjaro 18.0.4 Illyria

Package:
libldac: 2.0.2.3-2 (AUR)
pulseaudio-modules-bt-git : r133.9ff89b4-1 (AUR)
pulseaudio: 12.2-2
bluez: 5.50-6

```bash
$ pactl list cards

...
Carte #4
	Nom : bluez_card.CC_98_8B_31_21_B7
	Pilote : module-bluez5-device.c
	Module propriétaire : 29
	Propriétés :
		device.description = "LE_WH-1000XM3"
		device.string = "CC:98:8B:31:21:B7"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headset"
		bluez.path = "/org/bluez/hci0/dev_CC_98_8B_31_21_B7"
		bluez.class = "0x240404"
		bluez.alias = "LE_WH-1000XM3"
		device.icon_name = "audio-headset-bluetooth"
		device.intended_roles = "phone"
	Profils :
		headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: oui)
		a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: oui)
		a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: non)
		a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: non)
		a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (sinks: 1, sources: 0, priority: 40, available: non)
		a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (sinks: 1, sources: 0, priority: 40, available: non)
		off: Off (sinks: 0, sources: 0, priority: 0, available: oui)
	Profil actif : a2dp_sink_sbc
	Ports :
		headset-output: Headset (priority: 0, latency offset: 0 usec, available)
			Partie du(des) profil(s) : headset_head_unit, a2dp_sink_sbc, a2dp_sink_aac, a2dp_sink_aptx, a2dp_sink_aptx_hd, a2dp_sink_ldac
		headset-input: Headset (priority: 0, latency offset: 0 usec)
			Partie du(des) profil(s) : headset_head_unit
...
```

EHfive, 2019-05-07:
Disable bluez BLE (lBluetooth Low Energy): edit `/etc/bluetooth/main.conf`, set `ControllerMode` to `bredr` (the default is `dual`)

```
$ cat /etc/bluetooth/main.conf | grep LE -C2

# Restricts all controllers to the specified transport. Default value
# is "dual", i.e. both BR/EDR and LE enabled (when supported by the HW).
# Possible values: "dual", "bredr", "le"
ControllerMode = bredr

```

EHfive, 2019-05-07:
Log

mrdev023, 2019-05-07:
Log?

mrdev023, 2019-05-07:
How to get log ?

EHfive, 2019-05-08:
`journalctl --user -b -u pulseaudio > log.txt`

mrdev023, 2019-05-10:
```
-- Logs begin at Mon 2019-05-06 18:57:45 CEST, end at Fri 2019-05-10 13:48:33 CEST. --
mai 10 13:42:48 florian-pc systemd[2176]: Starting Sound Service...
mai 10 13:42:53 florian-pc systemd[2176]: Started Sound Service.
mai 10 13:42:53 florian-pc pulseaudio[2303]: E: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files
```

RlndVt, 2019-05-14:
I'm also on Manjaro with the same package versions and I'm having a similar problem. I have a AKG Y500 and it's reporting that AAC isn't available.

```
-- Logs begin at Wed 2019-02-06 10:56:28 CET, end at Tue 2019-05-14 18:53:13 CEST. --
mei 14 18:32:29 Roland-TP systemd[1167]: Starting Sound Service...
mei 14 18:32:30 Roland-TP systemd[1167]: Started Sound Service.
mei 14 18:32:30 Roland-TP pulseaudio[1355]: E: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files
mei 14 18:50:21 Roland-TP pulseaudio[1355]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_aac: Unavailable profile
```

        name: <bluez_card.F8_DF_15_6B_29_45>
        driver: <module-bluez5-device.c>
        owner module: 29
        properties:
                device.description = "AKG Y500 WIRELESS"
                device.string = "F8:DF:15:6B:29:45"
                device.api = "bluez"
                device.class = "sound"
                device.bus = "bluetooth"
                device.form_factor = "headset"
                bluez.path = "/org/bluez/hci0/dev_F8_DF_15_6B_29_45"
                bluez.class = "0x240404"
                bluez.alias = "AKG Y500 WIRELESS"
                device.icon_name = "audio-headset-bluetooth"
                device.intended_roles = "phone"
        profiles:
                a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (priority 40, available: yes)
                a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (priority 40, available: no)
                a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (priority 40, available: no)
                a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (priority 40, available: no)
                a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (priority 40, available: no)
                headset_head_unit: Headset Head Unit (HSP/HFP) (priority 30, available: no)
                off: Off (priority 0, available: yes)

dpmungo, 2019-05-18:
Same problem here with Manjaro, all codecs apart from SBC are unavailable:
Headset: Sony Wi-1000X
libldac: 2.0.2.3-2 (AUR)
pulseaudio-modules-bt-git : r135.65ffe33-1 (AUR)
bluez-git: 1:5.50.r389.g6df5c5e37 (AUR, not the repo version)
pulseaudio: 12.2-2
(just to mention that using the newer version of this package for aur and bluez-git instead of bluez doesn't solve the issue)

EHfive, 2019-05-19:
https://github.com/EHfive/pulseaudio-modules-bt/issues/24#issuecomment-450787272
@mrdev023  Maybe ` ldconfig ` would help
```
$ sudo ldconfig
```

EHfive, 2019-05-19:
@RlndVt 
Post your result of https://github.com/EHfive/pulseaudio-modules-bt/issues/31#issuecomment-462717049

https://github.com/EHfive/pulseaudio-modules-bt/issues/40

I think that more like a bluez's issue

RlndVt, 2019-05-19:
> @RlndVt
> Post your result of [#31 (comment)](https://github.com/EHfive/pulseaudio-modules-bt/issues/31#issuecomment-462717049)
> 
> #40
> 
> I think that more like a bluez's issue

```
avinfo F8:DF:15:6B:29:45     
Connecting ... 
Stream End-Point #1: Audio Sink 
        Media Codec: SBC
                Channel Modes: Stereo JointStereo
                Frequencies: 44.1Khz 48Khz 
                Subbands: 8
                Blocks: 12 16 
                Bitpool Range: 2-53
Stream End-Point #3: Audio Sink 
        Media Codec: SBC
                Channel Modes: Stereo JointStereo
                Frequencies: 44.1Khz 48Khz 
                Subbands: 8
                Blocks: 12 16 
                Bitpool Range: 2-53
```

Installing Bluez-git now to see if that helps. Fwiw, I'm having the same issue with my OP3t, but it does work using a Galaxy S9.

mrdev023, 2019-05-21:
@EHfive Currently, I'm return back to Windows. But i'm try it later

djasa, 2019-06-04:
Hi, I'm seeing the same issue. I use Fedora 30 with `pulseaudio-module-bluetooth-freeworld-1.1.99-2.fc30.x86_64` from RPMFusion and `libldac-2.0.2.3-2.fc30.x86_64` from main fedora repository.

The PA log is here and captures PA restart, headphone connection and attempt to swith to the unavailable profile:
```
$ journalctl --user -b -u pulseaudio -u bluetooth -fn0 | tee tmp/pa.log
-- Logs begin at Mon 2019-04-15 10:53:42 CEST. --
Jun 04 18:16:13 cihla.of.spice.brq.redhat.com systemd[1945]: pulseaudio.service: Succeeded.
Jun 04 18:16:13 cihla.of.spice.brq.redhat.com systemd[1945]: Starting Sound Service...
Jun 04 18:16:13 cihla.of.spice.brq.redhat.com systemd[1945]: Started Sound Service.
Jun 04 18:16:13 cihla.of.spice.brq.redhat.com pulseaudio[1697]: E: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name is not activatable
Jun 04 18:17:39 cihla.of.spice.brq.redhat.com pulseaudio[1697]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_ldac: Unavailable profile
```
System `bluetooth.service` says this:
```
# journalctl -u bluetooth --since 18:16:12 --until 18:17:40
-- Logs begin at Mon 2019-04-15 09:12:39 CEST, end at Tue 2019-06-04 18:28:13 CEST. --
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[5514]: Terminating
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com systemd[1]: Stopping Bluetooth service...
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[5514]: Endpoint unregistered: sender=:1.2307 path=/MediaEndpoint/A2DPSource/VENDOR/LD>
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[5514]: Endpoint unregistered: sender=:1.2307 path=/MediaEndpoint/A2DPSource/VENDOR/AP>
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[5514]: Endpoint unregistered: sender=:1.2307 path=/MediaEndpoint/A2DPSource/VENDOR/AP>
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[5514]: Endpoint unregistered: sender=:1.2307 path=/MediaEndpoint/A2DPSource/SBC
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[5514]: Endpoint unregistered: sender=:1.2307 path=/MediaEndpoint/A2DPSink/VENDOR/APTX>
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[5514]: Endpoint unregistered: sender=:1.2307 path=/MediaEndpoint/A2DPSink/VENDOR/APTX
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[5514]: Endpoint unregistered: sender=:1.2307 path=/MediaEndpoint/A2DPSink/SBC
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[5514]: Stopping SDP server
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[5514]: Exit
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com systemd[1]: bluetooth.service: Succeeded.
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com systemd[1]: Stopped Bluetooth service.
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com systemd[1]: Starting Bluetooth service...
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Bluetooth daemon 5.50
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com systemd[1]: Started Bluetooth service.
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Starting SDP server
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Bluetooth management interface 1.14 initialized
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint registered: sender=:1.2307 path=/MediaEndpoint/A2DPSource/VENDOR/LDAC
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint registered: sender=:1.2307 path=/MediaEndpoint/A2DPSource/VENDOR/APTX>
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint registered: sender=:1.2307 path=/MediaEndpoint/A2DPSource/VENDOR/APTX
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint registered: sender=:1.2307 path=/MediaEndpoint/A2DPSource/SBC
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint registered: sender=:1.2307 path=/MediaEndpoint/A2DPSink/VENDOR/APTXHD
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint registered: sender=:1.2307 path=/MediaEndpoint/A2DPSink/VENDOR/APTX
Jun 04 18:16:12 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint registered: sender=:1.2307 path=/MediaEndpoint/A2DPSink/SBC
Jun 04 18:16:13 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint unregistered: sender=:1.2307 path=/MediaEndpoint/A2DPSource/VENDOR/LD>
Jun 04 18:16:13 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint unregistered: sender=:1.2307 path=/MediaEndpoint/A2DPSource/VENDOR/AP>
Jun 04 18:16:13 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint unregistered: sender=:1.2307 path=/MediaEndpoint/A2DPSource/VENDOR/AP>
Jun 04 18:16:13 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint unregistered: sender=:1.2307 path=/MediaEndpoint/A2DPSource/SBC
Jun 04 18:16:13 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint unregistered: sender=:1.2307 path=/MediaEndpoint/A2DPSink/VENDOR/APTX>
Jun 04 18:16:13 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint unregistered: sender=:1.2307 path=/MediaEndpoint/A2DPSink/VENDOR/APTX
Jun 04 18:16:13 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint unregistered: sender=:1.2307 path=/MediaEndpoint/A2DPSink/SBC
Jun 04 18:16:13 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint registered: sender=:1.2359 path=/MediaEndpoint/A2DPSource/VENDOR/LDAC
Jun 04 18:16:13 cihla.of.spice.brq.redhat.com bluetoothd[1676]: Endpoint registered: sender=:1.2359 path=/MediaEndpoint/A2DPSource/VENDOR/APTX>
[... cut ...]
		a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (sinks: 1, sources: 0, priority: 40, available: no)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: a2dp_sink_sbc
	Ports:
		headset-output: Headset (priority: 0, latency offset: 0 usec)
			Part of profile(s): headset_head_unit, a2dp_sink_sbc, a2dp_sink_aptx, a2dp_sink_aptx_hd, a2dp_sink_ldac
		headset-input: Headset (priority: 0, latency offset: 0 usec)
			Part of profile(s): headset_head_unit
```
I will happily provide any other info requested.

EHfive, 2019-06-05:
@djasa 
> https://www.lidl-service.com/cps/rde/xchg/SID-3C2ABF55-2864998B/lsp/hs.xsl/product.html?id=2913054263&title=Bluetooth%C2%AE+True+Wireless+In-Ear+Headphones&count=1

I can't found out any information suggest that you headset support any other codecs except SBC which is a default standard; Also see #31.

I am closing this issue because it's misleading.

closed 2019-06-05 by EHfive

djasa, 2019-06-05:
@EHfive hi, I forgot to mention that I checked headphone codec support by pairing them to Android phone and then forcing specific codecs in Developer options. I can hear notable changes in sound quality or feel when switching between codecs (SBC, AAC, AptX, LDAC, what Android calls "Scalable Codec"). Unfortunately, when I managed to extract BT snoop from the phone (using "Generate bugreport" in the phone and ), it didn't contain anything related to codecs so I didn't bother to attach it (unlike capture by Wireshark on desktop which clearly shows just SBC). I'm attaching it to this comment in hopes that someone else's Wireshark-fu 

Anyway, after posting the previous comment, I noticed I'm [not alone](https://github.com/EHfive/pulseaudio-modules-bt/issues/20#issuecomment-470575762-permalink) so I'll ask Sesivany about his status when I see him. :)

`avinfo` also shows SBC only:
```
$ /var/lib/mock/fedora-30-x86_64/root/builddir/build/BUILD/bluez-5.50/tools/avinfo A8:75:D6:0A:78:C8
Connecting ... 
Stream End-Point #1: Audio Sink 
	Content Protection: SCMS-T
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 32Khz 44.1Khz 48Khz 
		Subbands: 4 8
		Blocks: 4 8 12 16 
		Bitpool Range: 8-35
```
[bugreport-2019-06-04-15-48-25-btsnooz.zip](https://github.com/EHfive/pulseaudio-modules-bt/files/3255940/bugreport-2019-06-04-15-48-25-btsnooz.zip)

Should I file an entirely new issue for this?

EHfive, 2019-06-05:
@djasa 
You can‘t use LDAC or Aptx or AAC if your headset don't support it (codec).
If you want to use LDAC, just buy a headset supporting LDAC.(possibly produced by Sony)

EHfive, 2019-06-05:
@djasa 
This repo just add LDAC/aptX/AAC support to pulseaudio on your PC, not your BT headset/audio transmitter.

grulja, 2019-06-25:
I have same problem, the headphones actually do support AAC, see [1]. I have Fedora 30 and pulseaudio-module-bluetooth-freeworld built with AAC support, but I see this codec as unavailable.

```
Card #1
        Name: bluez_card.00_42_79_3A_AA_D9
        Driver: module-bluez5-device.c
        Owner Module: 24
        Properties:
                device.description = "AKG Y500 WIRELESS"
                device.string = "00:42:79:3A:AA:D9"
                device.api = "bluez"
                device.class = "sound"
                device.bus = "bluetooth"
                device.form_factor = "headset"
                bluez.path = "/org/bluez/hci0/dev_00_42_79_3A_AA_D9"
                bluez.class = "0x240404"
                bluez.alias = "AKG Y500 WIRELESS"
                device.icon_name = "audio-headset-bluetooth"
                device.intended_roles = "phone"
        Profiles:
                a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: yes)
                a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: no)
                a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: no)
                a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (sinks: 1, sources: 0, priority: 40, available: no)
                a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (sinks: 1, sources: 0, priority: 40, available: no)
                headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: no)
                off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
        Active Profile: a2dp_sink_sbc
        Ports:
                headset-output: Headset (priority: 0, latency offset: 0 usec, available)
                        Part of profile(s): a2dp_sink_sbc, a2dp_sink_aac, a2dp_sink_aptx, a2dp_sink_aptx_hd, a2dp_sink_ldac, headset_head_unit
                headset-input: Headset (priority: 0, latency offset: 0 usec, not available)
                        Part of profile(s): headset_head_unit

```
[1] https://www.akg.com/on/demandware.static/-/Sites-masterCatalog_Harman/default/dw0c14c801/pdfs/AKG_Y500_Spec_Sheet_English.pdf

rajeeshknambiar, 2019-07-07:
I have similar problem with AAC and Bose AE2 Soundlink which does support AAC, according to forum posts (but see output of `avinfo`).

On Fedora 30, I've rebuilt the `pulseaudio-module-bluetooth-freeworld` version 1.1.99-6 (currently available 1.1.99-2 did not have AAC support; `a2dp_sink_acc` was missing in profiles). Now I can see `a2dp_sink_acc`, but trying to switch to that profiles fails.

```
rajeesh@athena $  avinfo 2C:41:A1:FC:60:4E
Connecting ... 
Stream End-Point #1: Audio Sink 
        Media Codec: SBC
                Channel Modes: Mono DualChannel Stereo JointStereo
                Frequencies: 16Khz 32Khz 44.1Khz 48Khz 
                Subbands: 4 8
                Blocks: 4 8 12 16 
                Bitpool Range: 2-53
        Content Protection: SCMS-T
Stream End-Point #49: Audio Source 
        Media Codec: Vendor Specific A2DP Codec
                Vendor ID 0x0000000a
                Vendor Specific Codec ID 0x0103
                Vendor Specific Data: 0x07 0x06 0x00 0x00 0xff 0xff 0x02 0x35
        Content Protection: SCMS-T
Stream End-Point #17: Audio Sink 
        Media Codec: Vendor Specific A2DP Codec
                Vendor ID 0x0000000a
                Vendor Specific Codec ID 0x0103
                Vendor Specific Data: 0x07 0x06 0x00 0x00 0xff 0xff 0x02 0x35
        Content Protection: SCMS-T
Stream End-Point #33: Audio Source 
        Media Codec: SBC
                Channel Modes: Mono DualChannel Stereo JointStereo
                Frequencies: 16Khz 32Khz 44.1Khz 48Khz 
                Subbands: 4 8
                Blocks: 4 8 12 16 
                Bitpool Range: 2-53
        Content Protection: SCMS-T

rajeesh@athena $  pactl list cards
Card #1
        Name: bluez_card.2C_41_A1_FC_60_4E
        Driver: module-bluez5-device.c
        Owner Module: 27
        Properties:
                device.description = "Bose AE2 SoundLink"
                device.string = "2C:41:A1:FC:60:4E"
                device.api = "bluez"
                device.class = "sound"
                device.bus = "bluetooth"
                device.form_factor = "headphone"
                bluez.path = "/org/bluez/hci0/dev_2C_41_A1_FC_60_4E"
                bluez.class = "0x240418"
                bluez.alias = "Bose AE2 SoundLink"
                device.icon_name = "audio-headphones-bluetooth"
        Profiles:
                headset_head_unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 30, available: yes)
                a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (sinks: 1, sources: 0, priority: 40, available: yes)
                a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (sinks: 1, sources: 0, priority: 40, available: no)
                a2dp_sink_aptx: High Fidelity Playback (A2DP Sink: aptX) (sinks: 1, sources: 0, priority: 40, available: no)
                a2dp_sink_aptx_hd: High Fidelity Playback (A2DP Sink: aptX HD) (sinks: 1, sources: 0, priority: 40, available: no)
                a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (sinks: 1, sources: 0, priority: 40, available: no)
                off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
        Active Profile: a2dp_sink_sbc
        Ports:
                headphone-output: Headphone (priority: 0, latency offset: 0 usec)
                        Part of profile(s): headset_head_unit, a2dp_sink_sbc, a2dp_sink_aac, a2dp_sink_aptx, a2dp_sink_aptx_hd, a2dp_sink_ldac
                headphone-input: Bluetooth Input (priority: 0, latency offset: 0 usec)
                        Part of profile(s): headset_head_unit
rajeesh@athena $  pactl set-card-profile 1 a2dp_sink_aac 
Failure: Invalid argument

journalctl: pulseaudio[1179]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_aac: Unavailable profile
 ```

If I check the advanced audio settings (in KDE system settings), there are only 2 profiles available for the bluetooth headset — HFP and SBC.

Any pointers to troubleshoot?

rajeeshknambiar, 2019-07-08:
> If I check the advanced audio settings (in KDE system settings), there are only 2 profiles available for the bluetooth headset — HFP and SBC.

I dug further and found another statement from Bose clearly acknowledging that the AE2 Soundlink **does not** support AAC [1]. My bad :-/

[1] https://community.bose.com/t5/Headphones-Archive/Soundlink-Around-Ear-2-wireless-AAC-support/m-p/146076/highlight/true#M28801