EHfive/pulseaudio-modules-bt #58: Please provide a way to disable native and ofono support at build time.

asavah, 2019-05-19:
Feature request:
When building pulseaudio it's possible to disable "native" and "ofono" support via configure switches ```--disable-bluez5-ofono-headset --disable-bluez5-native-headset```, this allows to avoid the hassle with switching pa profiles and allow headsets to start in a2dp mode directly.

It would be nice to have such a knob when building this project, please consider it.
Thanks

EHfive, 2019-05-19:
https://github.com/EHfive/pulseaudio-modules-bt/blob/65ffe335a419405ceef12742e72294e06d484491/config.h.in#L23-L24

Currently, you are able to disable "native" and "ofono" by deleting those lines in `config.h`.

But I would add cmake options making it configurable.

asavah, 2019-05-19:
Yep, that's exactly what I did ))
But a cmake knob would be nice to have.

EHfive, 2019-05-20:
Now you can disable both "native" and "ofono" by `cmake -DOFONO_HEADSET=OFF -DNATIVE_HEADSET=OFF <path to CMakeLists.txt's dir>`

commit 2fde9b79 (2019-05-20): Add cmake options  `OFONO_HEADSET`,  `NATIVE_HEADSET`; Close #58
commit 16b2aa4f (2019-09-11): Add cmake options  `OFONO_HEADSET`,  `NATIVE_HEADSET`; Close #58
closed 2019-05-20 by EHfive