EHfive/pulseaudio-modules-bt #61: Any plans to merge upstream?

Lastique, 2019-06-12:
Hi.

I was wondering if there are plans to merge this work (or at least some parts of it) into the upstream [PulseAudio](https://gitlab.freedesktop.org/pulseaudio/pulseaudio). I'm particularly interested in AAC support, but I'm sure other codecs are a huge improvement over SBC, which is the only choice with the vanilla PA. It would be nice to eventually have support for the more advanced codecs out of the box in the standard PA shipped by Linux distros.

Thanks.

EHfive, 2019-06-12:
Check #1.

FDK-AAC support would never be added/merged into upstream due to license issues.

closed 2019-06-12 by EHfive

EHfive, 2019-06-12:
I would continue this repo until there is a better option for extra A2DP codecs support. (May be @Pipewire)

Lastique, 2019-06-12:
Well, there is an AAC encoder in ffmpeg, although it is slightly worse quality compared to FDK-AAC. Still better than SBC.

Regarding license issues, I think this is a distro maintainer issue, and some distros consider FDK-AAC "free" and distribute it. Even if distros don't do it, users could still build PA from source with FDK-AAC.

Re PipeWire, it's still way way ahead, IMHO, and it's not clear regarding audio quality it provides.

And BTW, thank you for your effort on this.

EHfive, 2019-06-12:
> Re PipeWire, it's still way way ahead, IMHO, and it's not clear regarding audio quality it provides.

Yes, but PipeWire provides a SPA(Simple Plugin Api) so that third party can easily integrate extra A2DP codecs support into it without merge into it.

ValdikSS, 2019-06-21:
>FDK-AAC support would never be added/merged into upstream due to license issues.

https://lists.fedoraproject.org/archives/list/legal@lists.fedoraproject.org/thread/OVW25JRWOKOLVMW3XGUX7E4OXFUR2RCG/

>The Fedora Project is aware that the Free Software Foundation has stated that the Fraunhofer FDK AAC license is GPL incompatible, specifically, because of Clause 3.
>We believe that the fdk-aac software codec implementation that we wish to include in Fedora is no longer encumbered by AAC patents. This fact means that Clause 3 in the FDK AAC license is a "no op", or to put it plainly, if no patents are in play, there are no patent licenses to disclaim. For this (and only this) specific implementation of fdk-aac, we believe that the FDK AAC license is GPL compatible.