EHfive/pulseaudio-modules-bt #62: All bluetooth  stereo headsets failed to connect

patricklibert, 2019-06-13:
Ubuntu 18.04.2
Yesterday, 12 septermber 2019, i got an update through  the repository from EHfive.

Now all bluetooth stereo headsets failed to connect.
Also it doesn't work by the command bluetoothctl.
Message from bluetoothctl is: Failed to connect: org.bluez.Error.Failed

Thanks

patricklibert, 2019-06-13:
Fixed, some broken packages in ubuntu...

closed 2019-06-13 by patricklibert

sbmw, 2019-08-05:
@patricklibert What was your solution here?

patricklibert, 2019-08-05:
I don't know how exactly this happened but with an update my original ubuntu bluetooth packages where installed. I removed it and reinstalled the packages from this website.

sbmw, 2019-08-05:
Ah OK, that's a different issue than I'm seeing, but thank you very much for the update.