EHfive/pulseaudio-modules-bt #63: Higher bitrate SBC support (SBC HD/SBC UHQ)

ValdikSS, 2019-06-21:
SBC has "bitpool" configuration parameter which influence on bitrate of encoded stream. It's often 'locked' to 53 on most headphones, which corresponds to 328 kbps Joint Stereo ("High Quality" profile as defined in A2DP specification), but it could be circumvented on any headphones by using Dual Channel mode instead of Joint Stereo, which uses all available bitpool for each channel _separately_, which effectively doubles the bitrate and improves audio quality.

You can use bitrates up to 617 kbps (for 44.1 kHz) on the devices with 'locked' bitpool and up to probably 917 kbps on the devices without maximum bitpool limit.
Optimal bitrates are 452 kbps for EDR 2 Mbps devices and 551 kbps for EDR 3 Mbps devices.

SBS 451 kbps is superior to aptX and SBC 551 kbps is on par with aptX HD.

Please consider adding Dual Channel profile support into your module. @pali already added it into his patchset, see https://lists.freedesktop.org/archives/pulseaudio-discuss/2019-June/031176.html

Basically you need to add additional profile with the following parameters:
```
EDR 2 Mbps: SBC Dual Channel, bitpool 38, 16 blocks, 8 subbands, loudness
EDR 3 Mbps: SBC Dual Channel, bitpool 47, 16 blocks, 8 subbands, loudness
```

I'm not aware if you can determine EDR speed using BlueZ API, but @pali said you can determine an MTU value. Here's what I did for Android:

```
/*
 * 3DH5 minimum safe payload size for 4 audio frames of:
 * 817 bytes - (4 bytes L2CAP Header + 12 bytes AVDTP Header)
 * Computed for 551 kbps Dual Channel (200 bytes audio frame + 1 byte SBC header)
 */
#define MIN_3MBPS_AVDTP_SAFE_MTU 801

/*
 * SBC Dual Channel (SBC HD) 3DH5 bitrates.
 * 600 kbps @ 48 khz, 551.3 kbps @ 44.1 khz.
 * Up to 5 frames for 3DH5.
 */
#define A2DP_SBC_3DH5_DEFAULT_BITRATE 552
#define A2DP_SBC_3DH5_48KHZ_BITRATE 601

#define A2DP_SBC_NON_EDR_MAX_RATE 229

static uint16_t a2dp_sbc_source_rate() {
  uint16_t rate = 328;
  if (a2dp_sbc_encoder_cb.sbc_encoder_params.s16SamplingFreq == SBC_sf48000)
    rate = A2DP_SBC_48KHZ_BITRATE;

  if (a2dp_sbc_encoder_cb.peer_supports_3mbps &&
      a2dp_sbc_encoder_cb.TxAaMtuSize >= MIN_3MBPS_AVDTP_SAFE_MTU) {
    rate = A2DP_SBC_3DH5_DEFAULT_BITRATE;
    if (a2dp_sbc_encoder_cb.sbc_encoder_params.s16SamplingFreq == SBC_sf48000)
      rate = A2DP_SBC_3DH5_48KHZ_BITRATE;
  }
  /* restrict bitrate if a2dp link is non-edr */
  if (!a2dp_sbc_encoder_cb.is_peer_edr) {
    rate = A2DP_SBC_NON_EDR_MAX_RATE;
    LOG_VERBOSE(LOG_TAG, "%s: non-edr a2dp sink detected, restrict rate to %d",
                __func__, rate);
  }
  return rate;
}
```

Note that MTU here is basically a payload size including all audio headers (SBC payload + 1 byte SBC header), an 'MTU' of audio stream, not of an underlying connection.

More information here: [Bluetooth stack modifications to improve audio quality on headphones without AAC, aptX, or LDAC codecs](https://habr.com/en/post/456476/)
Bitrate calculator may be helpful: https://btcodecs.valdikss.org.ru/sbc-bitrate-calculator/

jbbr, 2019-08-20:
I'm also very interested in this. It's currently very unpleasant to use SBC-devices on Linux

soredake, 2019-09-25:
Any progress on this?

ishitatsuyuki, 2019-10-19:
FYI, LineageOS post on this (they called this SBC XQ): https://lineageos.org/engineering/Bluetooth-SBC-XQ/

mikelpr, 2019-10-30:
just bought a pair of redmi airdots for the office and they only support SBC. it would be nice if we could get the patch included

iavael, 2020-01-09:
Up

iavael, 2020-01-09:
It’s worth to mention that there’s PR for this https://github.com/EHfive/pulseaudio-modules-bt/pull/90

EHfive, 2020-03-07:
Close.

closed 2020-03-07 by EHfive

nBalykin, 2020-04-14:
> I'm currently working on this issue. For now - I've added configuration overrides for almost all SBC parameters. Stuff like SBC HD/DualChannel/etc seems to work pretty well (at least for my devices). I just need to add some user-friendly configuration options for it.

Has configration been added? Which file I should edit to configure it? I've read readme but dont understood which file i should edit.

DarkCaster, 2020-04-14:
> Has configration been added? Which file I should edit to configure it? I've read readme but dont understood which file i should edit.

Hello.You should add configuration to "default.pa" config file like that (example):

`load-module module-bluetooth-discover a2dp_config="sbc_min_bp=34 sbc_max_bp=34 sbc_freq=44k sbc_cmode=dual sbc_alloc=auto sbc_sbands=auto sbc_blen=auto"`

soredake, 2020-06-08:
`sudo sed -i 's|load-module module-bluetooth-discover|load-module module-bluetooth-discover a2dp_config="sbc_min_bp=47 sbc_max_bp=47 sbc_freq=44k sbc_cmode=dual sbc_alloc=loudness sbc_sbands=8 sbc_blen=16"|g' /etc/pulse/default.pa`