EHfive/pulseaudio-modules-bt #64: Add ffmpeg-devel as a weak dependency to the rpmfusion package

cryobry, 2019-06-21:
Is there any reason that ffmpeg-devel couldn't be added as a `Recommends:`? I think it would streamline installation for novices.

kwizart, 2019-06-25:
The FFmpeg library is linked, so it means it cannot be made optional.

The package is provided by RPM Fusion on Fedora nowadays, so the install procedure for anyone good enough to use a copr repo should be easier when using RPM Fusion.

kwizart, 2019-06-25:
Seems like there is some support to dlopen ffmpeg (libavcodec library).
Unfortunately this looks fragile and it's usually not recommended to dlpopen such library. Best would be to link if not done already (but the dlopen code would make sense if theses changes goes into the fedora side at some point).

gombosg, 2019-06-28:
Hi, package maintainer here. `ffmpeg-libs` is currently in the Recommends section of the RPM, exactly because of the dlopen part. `ffmpeg-devel` is only needed at build on the RPMFusion servers.

cryobry, 2019-06-29:
I couldn't enable the aptX HD sink until I installed ffmpeg-devel, unless this was a coincidence.

kwizart, 2019-06-29:
@cryobry which 3rd part repository are you using ?

cryobry, 2019-06-29:
RPMFusion

cryobry, 2019-07-01:
Removed ffmpeg-devel and I still have the aptX HD sink available, strange.

closed 2019-07-01 by cryobry