EHfive/pulseaudio-modules-bt #65: module-bluez5-device crashes with undefined symbol: pa_thread_make_realtime

danilovsergei, 2019-06-30:
I made pulseaudio-modules-bt [ebuild](https://github.com/danilovsergei/geonix-gentoo-overlay/blob/master/media-sound/pulseaudio-modules-bt/pulseaudio-modules-bt-9999.ebuild) for gentoo
This ebuild complies head version of pulseaudio-modules-bt for the pulseaudio currently installed.

After compiling for pulseaudio-12.2 I faced that pulseaudio crashes with:
$ pulseaudio 
W: [pulseaudio] pid.c: Stale PID file, overwriting.
E: [pulseaudio] ltdl-bind-now.c: Failed to open module /usr/lib64/pulse-12.2/modules/module-bluez5-device.so: /usr/lib64/pulse-12.2/modules/module-bluez5-device.so: undefined symbol: pa_thread_make_realtime`
`

After further research I found that pa_thread_make_realtime is not used yet in pulseaudio-12 as assumed in your [module-bluez5-device.c](https://github.com/EHfive/pulseaudio-modules-bt/blob/master/src/modules/bluetooth/module-bluez5-device.c#L1342)
For now pa_thread_make_realtime used only in head pulseaudio code. So i created small [patch](https://github.com/danilovsergei/geonix-gentoo-overlay/blob/master/media-sound/pulseaudio-modules-bt/files/01-missing_pa_thread_make_realtime.patch)

EHfive, 2019-06-30:
https://github.com/EHfive/pulseaudio-modules-bt/blob/2fde9b79449661afd162f5d6eeeba49f80eba37a/CMakeLists.txt#L141-L146

You defined a wrong pulseaudio version.

Check your `/usr/lib/pkgconfig/libpulse.pc` to see what version is defined.

danilovsergei, 2019-06-30:
Oh, I see what you mean.
My libpulse.pc has 12.2 which is correct.
But I set FORCE_LARGEST_PA_VERSION=ON , which is supposed to be used only for pulseaudio head.
I updated my [ebuild](https://github.com/danilovsergei/geonix-gentoo-overlay/blob/master/media-sound/pulseaudio-modules-bt/pulseaudio-modules-bt-9999.ebuild) and removed patch.
now everything works fine.
Thank you!

closed 2019-06-30 by EHfive