EHfive/pulseaudio-modules-bt #66: Add support for battery level reporting

BlackSoulxxx, 2019-07-17:
Can you add battery level reporting similar to Android and iOS? This may help you
https://github.com/sre/pulseaudio/commit/d66b66d20e9bc73e6d0ca89283cf2b5675304b00
In my case, Im using Audeze Cipher Bluetooth with aptX HD support. Tried bluetoothctl and found out it not using GATT.

jaccc, 2019-09-15:
it would be useful

jtojnar, 2019-12-21:
Relevant https://gitlab.freedesktop.org/pulseaudio/pulseaudio/issues/722

invidian, 2020-02-19:
I'm still interested in that feature.

invidian, 2020-04-19:
Still :smile:

userofryzen, 2020-05-20:
I am interested too, but.. may be it's not a problem of that modules(?)

jaccc, 2020-05-20:
this script works for me when correct port is found
https://github.com/TheWeirdDev/Bluetooth_Headset_Battery_Level/blob/master/bl_battery.py

bhack, 2020-05-25:
Please check https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues/722#note_503760

EHfive, 2020-06-06:
Close, I think it's not our business

closed 2020-06-06 by EHfive