EHfive/pulseaudio-modules-bt #67: Unable to fetch release from repository [404]

Miszo97, 2019-07-22:
Ubuntu: **18.04**
I tried to download package from ppa repository so tried to add this to repositories list.
```
sudo add-apt-repository ppa:eh5/pulseaudio-a2dp

```
Here is the output:
```
➜  ~ sudo add-apt-repository ppa:eh5/pulseaudio-a2dp   
 Adds LDAC, aptX, aptX HD, AAC codecs support to PulseAudio
  package: pulseaudio-modules-bt

HOMEPAGE: https://github.com/EHfive/pulseaudio-modules-bt

libldac: https://github.com/EHfive/ldacBT
 More info: https://launchpad.net/~eh5/+archive/ubuntu/pulseaudio-a2dp
Press [ENTER] to continue or Ctrl-c to cancel adding it.

Hit:1 http://pl.archive.ubuntu.com/ubuntu bionic InRelease
Get:2 http://security.ubuntu.com/ubuntu bionic-security InRelease [88,7 kB]                                                                            
Hit:3 http://pl.archive.ubuntu.com/ubuntu bionic-updates InRelease                                                                                     
Hit:4 http://pl.archive.ubuntu.com/ubuntu bionic-backports InRelease                                                                 
Ign:5 http://ppa.launchpad.net/aacebedo/fasd/ubuntu bionic InRelease                                                               
Hit:6 http://ppa.launchpad.net/eh5/pulseaudio-a2dp/ubuntu bionic InRelease                                                         
Hit:7 https://apt.enpass.io stable InRelease                                                                                       
Hit:8 http://ppa.launchpad.net/openjdk-r/ppa/ubuntu bionic InRelease                                
Err:9 http://ppa.launchpad.net/aacebedo/fasd/ubuntu bionic Release            
  404  Not Found [IP: 91.189.95.83 80]
Reading package lists... Done                                                 
E: The repository 'http://ppa.launchpad.net/aacebedo/fasd/ubuntu bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
For some reason it can't find Release file.

EHfive, 2019-07-23:
Can you read?
```
Err:9 http://ppa.launchpad.net/aacebedo/fasd/ubuntu bionic Release            
  404  Not Found [IP: 91.189.95.83 80]
Reading package lists... Done                                                 
E: The repository 'http://ppa.launchpad.net/aacebedo/fasd/ubuntu bionic Release' does not have a Release file.
```

EHfive, 2019-07-23:
Just delete `aacebedo/fasd` 
https://askubuntu.com/questions/307/how-can-ppas-be-removed?answertab=votes#tab-top

closed 2019-07-23 by EHfive