EHfive/pulseaudio-modules-bt #68: Update wiki for Fedora

gombosg, 2019-08-03:
Please update the Fedora wiki entry with some up-to-date instructions. Here is the diff:

[Packages.md.diff.txt](https://github.com/EHfive/pulseaudio-modules-bt/files/3463555/Packages.md.diff.txt)

Thanks!

EHfive, 2019-08-03:
Updated!

https://github.com/EHfive/pulseaudio-modules-bt/wiki/Packages/2b011236f5a80262af03a44c23c036868d2890f6

closed 2019-08-03 by EHfive

JaneSmith, 2020-03-25:
The given command for Fedora Silverblue doesn't seem to work. It results in the error "Unknown option --install".

gombosg, 2020-03-25:
Sorry, I don't use Fedora Silverblue, probably the CLI commands have changed since then. Let us know if you find the correct install command and we'll update the wiki.

JaneSmith, 2020-03-26:
As far as I can tell it's just currently not possible to install this directly from the repository with Fedora Silverblue: https://github.com/coreos/rpm-ostree/issues/1265

I think I managed to install it directly from the .rpm file using:
```
wget https://download1.rpmfusion.org/free/fedora/releases/31/Everything/x86_64/os/Packages/p/pulseaudio-module-bluetooth-freeworld-1.3-1.fc31.x86_64.rpm
rpm-ostree override replace ./pulseaudio-module-bluetooth-freeworld-1.3-1.fc31.x86_64.rpm
```

This isn't the ideal way to install it and I assume that it won't be updated automatically.

JaneSmith, 2020-03-28:
Okay, sorry for the confusion. It turns out that the command given on the wiki _is_ correct - and also _isn't_.

This is what the wiki says to enter (which doesn't work):

```
rpm-ostree override remove pulseaudio-module-bluetooth -–install pulseaudio-module-bluetooth-freeworld
```

This is the correct command:

```
rpm-ostree override remove pulseaudio-module-bluetooth --install pulseaudio-module-bluetooth-freeworld
```

Depending on the font used, these two lines might look identical, but they are not. The first line has some other symbol instead of a hyphen, giving `-–install` instead of `--install`, which is why it results in an error about it being an unknown option.

gombosg, 2020-03-28:
OMG. Thanks for figuring this out, @JaneSmith  yeah, one of those is a regular hyphen, and the other is an en dash. I have no clue how that happened since I presumably copied from the terminal and there's no en dash key on the keyboard...

@EHfive 
Can you please update the Wiki Fedora Silverblue command to the correct 
```
rpm-ostree override remove pulseaudio-module-bluetooth --install pulseaudio-module-bluetooth-freeworld
```
(note: only the dashes are different in `--install`)? thanks!

EHfive, 2020-03-29:
Updated.

https://github.com/EHfive/pulseaudio-modules-bt/wiki/Packages#fedora

carwyn, 2020-03-30:
When it says:
> Due to licensing restrictions, it doesn't support fdk-aac found in the nonfree RPMFusion repos - you have to create your own build for that.
Does that mean rebuild pulseaudio-modules-bt or is there more that needs rebuilding to get fdk-aac to work?

gombosg, 2020-03-30:
No, you just have to roll your own pulseaudio-modules-bt with fdk-aac-devel etc.

ktdreyer, 2020-12-14:
What changes do we need specifically for [pulseaudio-module-bluetooth-freeworld.spec](https://github.com/rpmfusion/pulseaudio-module-bluetooth-freeworld/blob/master/pulseaudio-module-bluetooth-freeworld.spec)? I see it already has `BuildRequires:  pkgconfig(fdk-aac)`.

Is this related to a difference between `fdk-aac` and `fdk-aac-free`?