EHfive/pulseaudio-modules-bt #69: Ubuntu installation issue

sbmw, 2019-08-05:
Hi, I'm sorry if I'm missing something obvious, but the package installation isn't working for me on Ubuntu

I follow the basic package installation instructions

sudo add-apt-repository ppa:eh5/pulseaudio-a2dp
sudo apt-get update
sudo apt-get install libavcodec58 libldac pulseaudio-modules-bt

This process removes pulseaudio-module-bluetooth (and in turn module-bluetooth-discover) which then means I cannot connect my bluetooth headphones.  I get the following error in blueman "Connection Failed: blueman.bluez.errors.DBusFailedError: Protocol not available"

Many thanks

EHfive, 2019-08-06:
Require pulseaudio >= 12.0

EHfive, 2019-08-06:
U can simply run `apt upgrade` because eh5/pulseaudio-a2dp already inculdes pulseaudio 12.2 for Ubuntu 18.04.

sbmw, 2019-08-06:
Ah, amazing!  Thanks!

closed 2019-08-06 by sbmw