EHfive/pulseaudio-modules-bt #70: Build fails with pulseaudio 12.99.1

gombosg, 2019-08-06:
```
/usr/bin/cc -fPIC -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -O2 -Wall -Wno-builtin-macro-redefined -Wno-unused -fno-common -DFAST
/usr/bin/ld: cannot find -lpulsecommon-12.99.1
/usr/bin/ld: cannot find -lpulsecore-12.99.1
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/bluez5-util.dir/build.make:222: libbluez5-util.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:150: CMakeFiles/bluez5-util.dir/all] Error 2
make: *** [Makefile:133: all] Error 2
```

But in the Fedora [specfile](https://src.fedoraproject.org/rpms/pulseaudio/blob/master/f/pulseaudio.spec#_437) it's only 12.99 - not containing the patch version number.

Maybe [this](https://github.com/EHfive/pulseaudio-modules-bt/blob/99aa1fe3d0b90a9ad5682d8cb3092e6e10f6d5cb/CMakeLists.txt#L161) could be fixed so that it's actually looking for `libpulsecore-12.99`?

Which one is the proper versioning?
Thanks!

asavah, 2019-08-06:
This was fixed almost a month ago https://github.com/EHfive/pulseaudio-modules-bt/commit/99aa1fe3d0b90a9ad5682d8cb3092e6e10f6d5cb 
git pull or redownload ....

gombosg, 2019-08-06:
Ouch!

gombosg, 2019-08-06:
That's why you should make a new release... I can't make a new package downstream for every commit. :(

closed 2019-08-06 by gombosg

EHfive, 2019-08-07:
Well, seems it's time to release v1.2.