EHfive/pulseaudio-modules-bt #72: Packages for openSUSE

aaccioly, 2019-08-11:
Hi guys,

I'm just wondering if there are plans to release an RPM for openSUSE. I've tried the RPM for Fedora and it unfortunately didn't work on my System (openSUSE Leap 15.1).

Kind regards,

aaccioly, 2019-10-17:
Can I appeal against the decision to close this issue somehow? :D

EHfive, 2019-10-18:
Hope someone can help.

jonasdemoor, 2019-12-26:
@aaccioly @EHfive I created an RPM for openSUSE Tumbleweed. It might also work for Leap, haven't tested it on Leap yet.

You can find the package here: https://build.opensuse.org/package/show/home:sonaj96/pulseaudio-modules-bt

Hope it helps!

aaccioly, 2019-12-26:
Thanks Jonas :)

---
Anthony Accioly

jonasdemoor, 2019-12-26:
I tried building a package for Leap 15.1, but seems like the PulseAudio version on Leap is too old (11.1, we need >=11.59.1) :/

aginies, 2020-01-14:
i package this for LEAP15.1: see: https://build.opensuse.org/project/show/home:aginies:bluetoothHD

pedro-hro, 2020-04-21:
I've been using the @jonasdemoor RPM for a while but recently when tried in a new Tumbleweed with gnome installation it just doesn't worked, the same hardware with a new SSD. I'm don't know too much of Linux, sorry, I'm studying yet, in Blueman I have this output: "Failed to change profile to a2dp_sink_aptx". Don't know where to find any other log about this error, please help me.

jonasdemoor, 2020-04-21:
> I've been using the @jonasdemoor RPM for a while but recently when tried in a new Tumbleweed with gnome installation it just doesn't worked, the same hardware with a new SSD. I'm don't know too much [...]

@pedro-hro  I've done a new Tumbleweed installation yesterday, I'm checking if I have the same problem.

jonasdemoor, 2020-04-21:
@pedro-hro No issues here, can you try switching switching the audio profile from GNOME's sound settings menu? I'm using GNOME's bluetooth and PulseAudio's settings menus.

If that doesn't help, you can see if the PulseAudio and bluetooth services are logging some errors:
`journalctl --user -u pulseaudio.service -l`
and
`sudo journalctl -u bluetooth.service -l`

This should give you some indication what's going wrong.

pedro-hro, 2020-04-21:
Tried using GNOME's settings, it gave me no output but doesn't worked. 

The output of:`pactl list | grep a2dp_codec`:
bluetooth.a2dp_codec = "SBC"

The output of `journalctl --user -u pulseaudio.service -l`:
Apr 21 19:35:54 install systemd[2052]: Starting Sound Service...
Apr 21 19:35:55 install systemd[2052]: Started Sound Service.
Apr 21 19:35:55 install pulseaudio[2125]: E: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceU>
Apr 21 19:37:17 install pulseaudio[2125]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_aptx: Unavailable profile
Apr 21 19:47:31 install pulseaudio[2125]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_aptx: Unavailable profile
Apr 21 19:47:37 install pulseaudio[2125]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_aptx: Unavailable profile`

The output of `sudo journalctl -u bluetooth.service -l`:
Apr 21 19:34:36 install systemd[1]: Starting Bluetooth service...
Apr 21 19:34:36 install bluetoothd[1402]: Bluetooth daemon 5.54
Apr 21 19:34:37 install bluetoothd[1402]: Starting SDP server
Apr 21 19:34:37 install bluetoothd[1402]: Bluetooth management interface 1.15 initialized
Apr 21 19:34:39 install systemd[1]: Started Bluetooth service.
Apr 21 19:35:55 install bluetoothd[1402]: Endpoint registered: sender=:1.53 path=/MediaEndpoint/A2DPSource/VENDOR/LDAC
Apr 21 19:35:55 install bluetoothd[1402]: Endpoint registered: sender=:1.53 path=/MediaEndpoint/A2DPSource/SBC
Apr 21 19:35:55 install bluetoothd[1402]: Endpoint registered: sender=:1.53 path=/MediaEndpoint/A2DPSink/SBC
Apr 21 19:37:10 install bluetoothd[1402]: /org/bluez/hci0/dev_06_A2_00_0E_4F_00/sep1/fd0: fd(51) ready

Don know what to try next.

jonasdemoor, 2020-04-24:
@pedro-hro Could you try running `avinfo` to see which codecs are supported by your headset? see: https://github.com/EHfive/pulseaudio-modules-bt/issues/31#issuecomment-462717049

pedro-hro, 2020-04-25:
Here it is:
`Stream End-Point #5: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000004f
		Vendor Specific Codec ID 0x0001
		Vendor Specific Data: 0xf2
		Vendor Specific Value (aptX)
			Frequencies: 16kHz 32kHz 44.1kHz 48kHz 
			Channel modes: Stereo 
	Content Protection: SCMS-T
Stream End-Point #3: Audio Sink 
	Media Codec: MPEG24
		Object Types: MPEG-2 AAC LC MPEG-4 AAC LC 
		Frequencies: 8kHz 11.025kHz 12kHz 16kHz 22.05kHz 24kHz 32kHz 44.1kHz 48kHz 
		Channels: 1 2 
		Bitrate: 320000
		VBR: Yes
	Content Protection: SCMS-T
Stream End-Point #4: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0001
		Vendor Specific Data: 0x03 0x23
		Vendor Specific Value (FastStream)
			Directions: sink source 
			Sink Frequencies: 44.1kHz 48kHz 
			Source Frequencies: 16kHz 
Stream End-Point #1: Audio Sink 
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 16Khz 32Khz 44.1Khz 48Khz 
		Subbands: 4 8
		Blocks: 4 8 12 16 
		Bitpool Range: 2-53
	Content Protection: SCMS-T
`

pedro-hro, 2020-04-25:
Here it is:
```
 Stream End-Point #5: Audio Sink 

	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000004f
		Vendor Specific Codec ID 0x0001
		Vendor Specific Data: 0xf2
		Vendor Specific Value (aptX)
			Frequencies: 16kHz 32kHz 44.1kHz 48kHz 
			Channel modes: Stereo 
	Content Protection: SCMS-T
Stream End-Point #3: Audio Sink 
	Media Codec: MPEG24
		Object Types: MPEG-2 AAC LC MPEG-4 AAC LC 
		Frequencies: 8kHz 11.025kHz 12kHz 16kHz 22.05kHz 24kHz 32kHz 44.1kHz 48kHz 
		Channels: 1 2 
		Bitrate: 320000
		VBR: Yes
	Content Protection: SCMS-T
Stream End-Point #4: Audio Sink 
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0001
		Vendor Specific Data: 0x03 0x23
		Vendor Specific Value (FastStream)
			Directions: sink source 
			Sink Frequencies: 44.1kHz 48kHz 
			Source Frequencies: 16kHz 
Stream End-Point #1: Audio Sink 
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 16Khz 32Khz 44.1Khz 48Khz 
		Subbands: 4 8
		Blocks: 4 8 12 16 
		Bitpool Range: 2-53
	Content Protection: SCMS-T

```

adianas, 2020-07-22:
I have the same problem,  it was working fine for few months but now it can only recognized profile is SBC (OnePLus Bullets Wireless 2)
before:
```
linux-pld8:~/Desktop> pactl list | grep a2dp_codec
                bluetooth.a2dp_codec = "aptX_HD"
```

now:
```
linux-pld8:~> pactl list | grep a2dp_codec
                bluetooth.a2dp_codec = "SBC"
```

I noticed that pulse audio module bet is not loaded, I reinstalled everything including pulse but all to avail

```
Distributor ID: openSUSE
Description:    openSUSE Tumbleweed
Release:        20200717
```

```
linux-pld8:~> uname -a
Linux linux-pld8 5.7.7-1-default #1 SMP Wed Jul 1 19:03:27 UTC 2020 (cba119b) x86_64 x86_64 x86_64 GNU/Linux
```

adianas, 2020-07-27:
@pedro-hro 

I managed to fix the problem

```
$ systemctl --user stop pulseaudio.socket
$ systemctl --user stop pulseaudio.service
$ pulseaudio --verbose
```
pulseaudio logs showed the following warning
```
W: [pulseaudio] ffmpeg_libs.c: Cannot open libavcodec library: libavcodec.so.58. libavcodec.so.58: cannot open shared object file: No such file or directory
W: [pulseaudio] ffmpeg_libs.c: Cannot open libavcodec library: libavcodec.so. libavcodec.so: cannot open shared object file: No such file or directory
```

Reinstalling  FFmpeg codec library `(libavcodec58_91) ` didn't help but it worked after I installed `ffmpeg-4-libavcodec-devel:`

```
linux-pld8:~> pactl list | grep a2dp_codec
                bluetooth.a2dp_codec = "aptX_HD"
```
I hope that helpes.

LizardWithHat, 2020-08-19:
Had the same problem with pulseaudio not finding the codec library, however installing the devel package or reinstalling libavcodecs didn't work.
The right codecs are there if libavcodecs58/libavutils56 (in my case) are installed. They just aren't named right for the module to find.
Symbolic linking them (libavcodec.so.58.XX -> libavcodec.so.58, same for util) worked just fine. That's just a workaround, though.

closed 2021-01-18 by EHfive