EHfive/pulseaudio-modules-bt #73: Selecting other profiles with pacmd

olewolf, 2019-08-20:
I apologize if this is not a PulseAudio issue. I've managed to successfully compile and install pulseaudio-modules-bt, and I'm now happily playing AAC-encoded sound over my Bluetooth loudspeaker.

However, for testing purposes I want to be able to also select the SBC codec and the LDAC codec. The pacmd command reports that the following profiles are supported:
```
a2dp_sink_sbc: High Fidelity Playback (A2DP Sink: SBC) (priority 40, available: no)
a2dp_sink_aac: High Fidelity Playback (A2DP Sink: AAC) (priority 40, available: unknown)
a2dp_sink_ldac: High Fidelity Playback (A2DP Sink: LDAC) (priority 40, available: no)
off: Off (priority 0, available: yes)
```
I can switch from "Off" to the AAC codec by issuing:
```
pacmd set-card-profile 1 a2dp_sink_aac
```
but switching to the SBC codec (using "a2dp_sink_sbc" instead of "a2dp_sink_aac") results in the following error:
```
Failed to set card profile to 'a2dp_sink_sbc'.
```
The pulseaudio debug output reports that:
```
(1187.496|   0.000) W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink_sbc: Unavailable profile
```
I can see in the profiles list above that the profile appears to not be available but before I installed pulseaudio-modules-bt the product did play back SBC-encoded sound.

What am I doing wrong?

EHfive, 2019-08-20:
https://github.com/EHfive/pulseaudio-modules-bt/wiki#can-i-choose-codec-

>No, A2DP codec is chosen by bluez. Though the order of codecs is declared in this module, it is arranged by quality of codec(LDAC > APTX HD > APTX > SBC). It means if your device support APTX,LDAC,SBC, bluez will choose LDAC codec as primary codec.

EHfive, 2019-08-20:
You need to build lateset bluez from bluez git master branch, and then enable its experimental features.

EHfive, 2019-08-21:
Close.

closed 2019-08-21 by EHfive