EHfive/pulseaudio-modules-bt #77: Codec switching not working with PulseAudio 12.99

ilyats, 2019-08-28:
Support for codec switching does not seem to work with PA 12.99. An attempt to set profile to sbc
produces error message
Refused to switch profile to a2dp_sink_sbc: Unavailable profile
(The headset supports SBC and AAC and bluetoothd does run with switch -E) I am investigating it.

ilyats, 2019-08-28:
No, bluez is from Debian package 5.50-1+b1 . I did some debugging, it seems that set_configuration_cb is only called for AAC during device connection. At request of profile change, set_profile_cb is called and ends with error message.

ilyats, 2019-08-28:
It worked with this version of bluez with PA 12.2-4 and pulseaudio-modules-bt 1.99-3 from Ubuntu packages.

EHfive, 2019-08-28:
Are you in bluez git master ?

EHfive, 2019-08-28:
Well, codec switching requires bluez git master.
Bluez adds codec switching interfaces after bluez@25d3c9da6254d6dcbba3dae49e9e8d7fc637b7ad aheading bluez 5.50.

> https://github.com/EHfive/pulseaudio-modules-bt/commit/6a4121c54b2636f1f6cb10acd816c02d03bb424d

> https://github.com/EHfive/pulseaudio-modules-bt/wiki#can-i-choose-codec-

EHfive, 2019-08-28:
pulseaudio-modules-bt 1.1.99 from PPA is for PA 12.2 only

EHfive, 2019-08-28:
Maybe your bluez is patched with bluez@25d3c9da6254d6dcbba3dae49e9e8d7fc637b7ad to bluez@02877c5e98b81655c0a369c5cffbf975b431ec7a.
But there is no possible that codec switching works on pure bluez official release v5.50.

ilyats, 2019-08-28:
It works with bluez from git.

closed 2019-08-28 by ilyats